OE4T/meta-tegra #432: some gstreamer plugins are missed in docker

cg3dland, 2020-09-30:
Dear Sir,

I build a image for Nano dev board (jetson-nano-qspi-sd) with Cuda, cudnn, tensorrt, gstreamer, nvidia docker, nvidia container runtime and toolkit, I add following lines into my local.conf:

DISTRO_FEATURES_append = " virtualization"
IMAGE_INSTALL_append += " cuda-toolkit cuda-compiler cuda-nvcc cuda-libraries cuda-driver cuda-cudart cuda-command-line-tools"
IMAGE_INSTALL_append += " cudnn tensorrt gstreamer1.0-plugins-nvvideo4linux2"
IMAGE_INSTALL_append += " l4t-graphics-demos-egldevice tegra-udrm-probeconf tegra-libraries libdrm"
IMAGE_INSTALL_append += " gstreamer1.0 gstreamer1.0-vaapi gstreamer1.0-omx-tegra v4l-utils"
IMAGE_INSTALL_append += " gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad"
IMAGE_INSTALL_append += " nvidia-docker"
CORE_IMAGE_EXTRA_INSTALL += " tegra-libraries-argus"

on Nano host, I run command gst-inspect-1.0 and get following results:
# gst-inspect-1.0 | grep omx
omx:  nvoverlaysink: OpenMax Video Sink
omx:  omxvp9enc: OpenMAX VP9 Video Encoder
omx:  omxvp8enc: OpenMAX VP8 Video Encoder
omx:  omxh265enc: OpenMAX H.265 Video Encoder
omx:  omxh264enc: OpenMAX H.264 Video Encoder
omx:  omxwmvdec: OpenMAX WMV Video Decoder
omx:  omxmpeg2videodec: OpenMAX MPEG2 Video Decoder
omx:  omxvp9dec: OpenMAX VP9 Video Decoder
omx:  omxvp8dec: OpenMAX VP8 Video Decoder
omx:  omxh265dec: OpenMAX H.265 Video Decoder
omx:  omxh264dec: OpenMAX H.264 Video Decoder
omx:  omxmpeg4videodec: OpenMAX MPEG4 Video Decoder

# gst-inspect-1.0 | grep nvv
nvvideo4linux2:  nvv4l2vp9enc: V4L2 VP9 Encoder
nvvideo4linux2:  nvv4l2vp8enc: V4L2 VP8 Encoder
nvvideo4linux2:  nvv4l2h265enc: V4L2 H.265 Encoder
nvvideo4linux2:  nvv4l2h264enc: V4L2 H.264 Encoder
nvvideo4linux2:  nvv4l2decoder: NVIDIA v4l2 video decoder
nvvideocuda:  videocuda: CUDA Post processor
nvv4l2camerasrc:  nvv4l2camerasrc: NvV4l2CameraSrc
nvvidconv:  nvvidconv: NvVidConv Plugin
nvvideosink:  nvvideosink: nVidia Video Sink

Then, I start dokcer nvcr.io/nvidia/l4t-base:r32.4.3 with following command:
docker run -it --rm --runtime nvidia --net=host --ipc=host --gpus all nvcr.io/nvidia/l4t-base:r32.4.3

and then, run same gst-inspect-1.0 commands, but I get following results:
# gst-inspect-1.0 | grep omx
libav:  avenc_h264_omx: libav OpenMAX IL H.264 video encoder encoder

# gst-inspect-1.0 | grep nvv
nvvideocuda:  videocuda: CUDA Post processor
nvvidconv:  nvvidconv: NvVidConv Plugin
nvvideosink:  nvvideosink: nVidia Video Sink

some gst plugins are missed in docker, e.g., "omx:  nvoverlaysink: OpenMax Video Sink", "nvvideo4linux2:  nvv4l2h264enc: V4L2 H.264 Encoder".

What wrong do I make in the docker command line or local.conf for Yocto?

Thanks,

madisongh, 2020-09-30:
Hard to say for sure.  Which image are you building? Make sure it's one that includes the drivers you need.

The other gstreamer plugin libraries may actually be present but are getting blacklisted due to loading issues.  You can check this by running `gst-inspect-1.0 -b` in the container. For blacklisted plugins, you can debug the loading issues by clearing the plugins cache and running gst-inspect with additional logging:
```
$ rm -rf ~/.cache/gstreamer-1.0
$ GST_DEBUG=4 gst-inspect-1.0 2>gst-inspect.log
```
then look at the log file to see what errors are being reported.

You can debug issues with missing files for the container passthrough mechanism by editing the file `/etc/nvidia-container-runtime/config.toml` and un-commenting the `debug =` lines, then reviewing the generated log files after starting up the container.

Hopefully one of the above will point you at any missing files.

cg3dland, 2020-10-09:
The image I building is core-image-base.

With your guide, I get following messages:

1. the command ‘gst-inspect-1.0 -b’ outputs:
Blacklisted files:
  libgstnvvideo4linux2.so
  libgstnveglglessink.so
  libgstnvarguscamerasrc.so
  libgstomx.so
  libgstnvv4l2camerasrc.so

2. with commands 'rm -rf ~/.cache/gstreamer-1.0; GST_DEBUG=4 gst-inspect-1.0 2>gst-inspect.log', the following messages are found from log file:
0:00:00.976203561    23   0x55b32f7800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvv4l2camerasrc.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.987559613    23   0x55b32f7800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:01.009220151    23   0x55b32f7800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvarguscamerasrc.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:01.011020733    23   0x55b32f7800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglglessink.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:01.013112308    23   0x55b32f7800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideo4linux2.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading

It seems the problem reason is the version of gstreamer on the host (v1.16) mismatches with the verison in container (v1.14).

I learn from one issue "Missing gstreamer plugins from nVidia #254" for downgrading the gstreamer version to v1.14:

You should be able to copy over all the gstreamer recipes from the openembedded-core or poky warrior branch into your own layer. You'll need to set PREFERRED_VERSION_* for the gstreamer packages to have the build choose the older version.

but I am sorry I do not fully understand this comment, may I have an example to set PREFERRED_VERSION_* for a gstreamer package, e.g,  gstreamer1.0-omx-tegra?

Thanks,

madisongh, 2020-10-09:
If you're using a recent-vintage branch, the simplest way to downgrade gstreamer to 1.14 is to add the `meta-tegra/contrib` to your bblayers configuration (if you haven't already), then follow the instructions in [the README file for that layer](https://github.com/OE4T/meta-tegra/blob/master/contrib/README.md).

cg3dland, 2020-10-12:
Following your guide, I made an image with gstreamer v1.14, the problem goes away on this image.

closed 2020-10-12 by cg3dland