OE4T/meta-tegra #440: Cross-compiling jetson_inference application that use CMake+CUDA

mozcelikors, 2020-10-14:
A while back, I've asked a question regarding yocto recipes or guides for running common neural networks [Here.](https://github.com/OE4T/meta-tegra/issues/379) I found that [jetson-inference](https://github.com/dusty-nv/jetson-inference) runs well on Ubuntu and I started trying to write it a Yocto recipe. The repo has a CMake build and it asks for Cuda installation. This is what I've got so far:

```
SUMMARY = "jetson-inference"
DESCRIPTION = "jetson-interence by dusy-nv"
LICENSE = "CLOSED"

RDEPENDS_${PN}_append = " libpython3 v4l-utils python3 python3-dev python3-numpy cuda-nvcc cuda-cudart cuda-toolkit cuda-target-environment cuda-libraries cuda-driver qtbase"
DEPENDS_append = " python3-numpy v4l-utils cuda-nvcc cuda-cudart cuda-toolkit cuda-target-environment cuda-libraries cuda-driver nativesdk-cuda-environment qtbase"

FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI += "git://www.github.com/dusty-nv/jetson-inference"

S = "${WORKDIR}/git"
SRCREV = "022739abc7631fa853ae63550ec091cd838b2105"

inherit pkgconfig cmake

# Defines to make cuda installation visible to CMake
EXTRA_OECMAKE += " -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-10.2"
EXTRA_OECMAKE += " -DCUDA_INCLUDE_DIRS=/usr/local/cuda-10.2/include"
EXTRA_OECMAKE += " -DCUDA_CUDART_LIBRARY=/usr/local/cuda-10.2/lib64/libcudart.so"

do_configure_prepend() {
  cd ${WORKDIR}/git
  git submodule update --init --recursive
}

do_install_append(){

}

FILES_${PN} += ""

```

I know that dependencies are a bit excessive now but I will be tailoring/optimizing them later.

Here, I got 2 questions:

1) I don't really know how cross compiling with cuda works.
First error I get during do_configure is:
`build/tmp/work/aarch64-poky-linux/jetson-inference/1.0-r0/recipe-sysroot/usr/local/cuda-10.2/bin/nvcc: cannot execute binary file`

2) Secondly, CUDA is not recognized:
```
| -- Could NOT find CUDA (missing: CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found version ".")
| -- CUDA version: .
```

3) Third, numpy is not found:
```
| -- To find NumPy Python interpretator is required to be found.
| -- NumPy not found
```

4) This weird error:
```
| CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
|   add_library cannot create target "jetson-utils-python-35" because another
|   target with the same name already exists.  The existing target is a shared
|   library created in source directory

| CMake Error at utils/python/bindings/CMakeLists.txt:59 (target_link_libraries):
|   Attempt to add link library "jetson-utils" to target
|   "jetson-utils-python-35" which is not built in this directory.
| 
|   This is allowed only when policy CMP0079 is set to NEW.
```

I am looking for guidance to continue porting this example to Yocto. What am I missing? It just feels like I should use nativesdk for some packages and I'm using target packages.
Any guidance is well appreciated.

madisongh, 2020-10-14:
Try adding `cuda` to your `inherit` line -  `cuda.bbclass` sets things up for CUDA cross-compilation, including some CMake integration.  And then you should be able to drop most of your explicit cuda DEPENDs and the EXTRA_OECMAKE settings.

Cross-building python modules can be a bit tricky; you at least need to add `python3native` and possibly `python3-dir` to your `inherit` line so python3 is available during the build.  I'm not sure how much that will help with the specific errors you're seeing, though.

You should also consider using the `gitsm://` fetcher in your SRC_URI to handle the submodule initialization for you, rather than doing it yourself in the do_configure prepend.

Hope this helps.

mozcelikors, 2020-10-14:
This helped in many aspects. I also added cmake_qt5 to make it so that Qt5Widgets.cmake also recognizes. Current status is as follows:

```
SUMMARY = "jetson-inference"
DESCRIPTION = "jetson-interence by dusy-nv"
LICENSE = "CLOSED"

RDEPENDS_${PN}_append = " libpython3 v4l-utils python3 python3-dev python3-numpy cuda-nvcc cuda-cudart cuda-toolkit cuda-target-environment cuda-libraries cuda-driver qtbase"
DEPENDS_append = " python3-native python3-numpy-native v4l-utils qtbase"

FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI += "git://www.github.com/dusty-nv/jetson-inference"

S = "${WORKDIR}/git"
SRCREV = "022739abc7631fa853ae63550ec091cd838b2105"

inherit pkgconfig cmake cmake_qt5 cuda python3native python3-dir

do_configure_prepend() {
  cd ${WORKDIR}/git
  git submodule update --init --recursive
}
```

Regarding `gitsm://`, I've read its buggy in many Yocto releases so that's why I didn't consider using it.
Anyhow, I am still having this last issue:

```
| CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
|   add_library cannot create target "jetson-inference-python-38" because
|   another target with the same name already exists.  The existing target is a
|   shared library created in source directory
|   "/media/mozcelikors/ext4/Yocto_Builds/jetson_nano_dunfell/build/tmp/work/armv8a_tegra210-poky-linux/jetson-inference/1.0-r0/git/python/bindings".
|   See documentation for policy CMP0002 for more details.
| Call Stack (most recent call first):
|   python/bindings/CMakeLists.txt:57 (cuda_add_library)
| 
| 
| CMake Error at python/bindings/CMakeLists.txt:59 (target_link_libraries):
|   Attempt to add link library "jetson-inference" to target
|   "jetson-inference-python-38" which is not built in this directory.
| 
|   This is allowed only when policy CMP0079 is set to NEW.
| 
| 
| CMake Error at python/bindings/CMakeLists.txt:62 (target_link_libraries):
|   Attempt to add link library "npymath" to target
|   "jetson-inference-python-38" which is not built in this directory.
| 
|   This is allowed only when policy CMP0079 is set to NEW.
```
I think this might be related to CMake version.

mozcelikors, 2020-10-16:
Hi again @madisongh,

I'm having trouble making CMake recognize cuda host binaries. Latest state of the recipe file is as follows:

```
SUMMARY = "jetson-inference"
DESCRIPTION = "jetson-interence by dusty-nv"
LICENSE = "CLOSED"

RDEPENDS_${PN}_append = " libpython3 v4l-utils python3 python3-dev python3-numpy cuda-nvcc cuda-cudart cuda-toolkit cuda-target-environment cuda-libraries cuda-driver qtbase libpthread-stubs"
DEPENDS_append = " libpthread-stubs-native python3-native python3-numpy-native v4l-utils qtbase"

FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI += " git://www.github.com/dusty-nv/jetson-inference"
SRC_URI += " file://src_override/utils/cuda/FindCUDA.cmake"
SRC_URI += " file://src_override/utils/python/bindings/CMakeLists.txt"

S = "${WORKDIR}/git"
SRCREV = "022739abc7631fa853ae63550ec091cd838b2105"

inherit pkgconfig cmake cmake_qt5 cuda python3native python3-dir

# Setting a CMake policy
EXTRA_OECMAKE += " -DCMAKE_POLICY_DEFAULT_CMP0079=NEW"

do_configure_prepend() {
  # Initialize all submodules
  cd ${WORKDIR}/git
  git submodule update --init --recursive

  # Update our modified FindCUDA.cmake with original after submodules are initialized,
  # the update is required to supress duplicate add_library target_link_libraries errors.
  cp ${WORKDIR}/src_override/utils/cuda/FindCUDA.cmake ${WORKDIR}/git/utils/cuda

  # Update our modified utils/python/bindings/CMakeLists.txt with original one.
  cp ${WORKDIR}/src_override/utils/python/bindings/CMakeLists.txt ${WORKDIR}/git/utils/python/bindings/CMakeLists.txt
}
```
I'm getting:

`: /media/mozcelikors/ext4/Yocto_Builds/jetson_nano_dunfell/build/tmp/work/armv8a_tegra210-poky-linux/jetson-inference/1.0-r0/recipe-sysroot/usr/local/cuda-10.2/bin/nvcc: cannot execute binary file`

When I inherit cuda, "nvcc" executable is used from the recipe-sysroot (target). I don't understand why.
I am on branch dunfell, cuda version is 10.2, CMake version is 3.16. I read in cuda.bbclass that for CMake >3.8 toolchain file is used, so I also tried with `EXTRA_OECMAKE += " -DCMAKE_TOOLCHAIN_FILE=${WORKDIR}/toolchain.cmake"`. But no use. 
My source overrides are for suppressing some errors, But the root issue is the one I mentioned above.

Any help is greately appreciated.

mozcelikors, 2020-10-16:
What's weird is:
During do_configure, I get native executable:
`| -- Found CUDA: /media/mozcelikors/ext4/Yocto_Builds/jetson_nano_dunfell/build/tmp/work/armv8a_tegra210-poky-linux/jetson-inference/1.0-r0/recipe-sysroot-native/usr/local/cuda-10.2 (found version "10.2")`
But during do_compile, target executable is being used:
`: /media/mozcelikors/ext4/Yocto_Builds/jetson_nano_dunfell/build/tmp/work/armv8a_tegra210-poky-linux/jetson-inference/1.0-r0/recipe-sysroot/usr/local/cuda-10.2/bin/nvcc: cannot execute binary file`

madisongh, 2020-10-16:
@mozcelikors take a look at [this commit](https://github.com/madisongh/tegra-test-distro/commit/c46c049a9a99e8436d046831535edf39d2c9e077) for something to work with.  The cmake files really needed changes to work in the cross-build environment, as I alluded to earlier.  Getting rid of Dusty's Find modules clears up the CUDA issue; for locating numpy there was no easy way to do it in CMake, so I added a setting that gets passed down from the recipe.

I couldn't build it with Qt5, but the rest of it builds OK.  Hopefully this will give you a better start.

mozcelikors, 2020-10-16:
Nicely done, Thank you very much I will definitely give this one a try soon.

madisongh, 2020-10-17:
> As far as I understood, punch line was removing
`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/utils/cuda" )`
for the issue that I was getting.

For the CUDA issue, yes.  It was also causing the problem with locating the native NumPy package rather than the target package.

>I will test this and try to fix any possible runtime problems as well.

Good luck.  As I mentioned, I don't have Qt in my layers, so you may run into some further build-related issues around that.

mozcelikors, 2020-10-18:
@madisongh I fixed some more stuff (network/image integration, qt, ttf-dejavu fonts) and updated the recipe. Binaries work fine now but the python gives a certain error:
https://github.com/dusty-nv/jetson-utils/issues/44
If you have any ideas, please do share. Thanks.

madisongh, 2020-10-20:
I don't know, but you could try patching the PyNumpy.cpp file as you suggested.  That file does trigger a missing return value warning in the function that calls `init_array()`, and I have run into cases in other C++ programs that had such warnings where the calling function was simply optimized away.  Since the numpy code retrieves configuration information from that array, including the endianness info, I could see you getting that runtime error due to the init function never getting called.

mozcelikors, 2020-10-20:
I tried with:
```
do_configure_prepend(){
    #sed -i -r "s/import_array[(][)][;]/import_array(); return NULL;/" ${S}/utils/python/bindings/PyNumpy.cpp
}
```
I can see this added `return NULL;` in tmp/work  but the error unfortunately remains the same.

madisongh, 2020-10-20:
Well, it was worth a shot.  I'm not sure what else to try at this point, sorry.

mozcelikors, 2020-10-21:
Fyi:
Well, I solved the issue, but in a weird fashion.
Either I made some mistake earlier adding `return NULL;`, or I somehow luckily solved a possible [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug) of some sort by adding a print statement :)
Flashed multiple times. The issue is resolved now with:

```
do_configure_prepend(){
    # Fix Python Runtime Error: "RuntimeError: FATAL: module compiled as little endian, but detected different endianness at runtime"
    sed -i -r "s/import_array[(][)][;]/import_array(); return NULL;/" ${S}/utils/python/bindings/PyNumpy.cpp
    sed -i -r "s/PyNumpy_ImportNumPy[(][)][;]/fprintf(stderr, \"ImportNumPy 1\"); fflush(stderr); PyNumpy_ImportNumPy();/" ${S}/utils/python/bindings/PyNumpy.cpp
}
```
There are still minor run-time problems, continuing the debug.

mozcelikors, 2020-10-24:
@madisongh One final question, if you would:
I'm trying to launch this on Xwayland,
Even with DISPLAY=:0 I'm getting:
```
[OpenGL] glDisplay -- X screen 0 resolution: 1920x1080
[OpenGL] glDisplay -- X window resolution: 1920x1080
[OpenGL] failed to create X11 Window
jetson-utils -- no output streams, creating fake null output
```
This means DISPLAY is recognized, but I still get "failed to create X11 window".
`xrandr` gives: 
```
root@jetson-nano-qspi-sd:~# DISPLAY=:0 xrandr                                                                                                                            
Screen 0: minimum 16 x 16, current 1920 x 1080, maximum 32767 x 32767                                                                                                    
XWAYLAND0 connected 1920x1080+0+0 (normal left inverted right x axis y axis) 0mm x 0mm                                                                                   
   1920x1080     59.96*+   
```
Is this something you know about? I feel like this is not related to tegra<->weston port. Is that true?
Where it fails in the code is: https://github.com/dusty-nv/jetson-utils/blob/fb828ef1c549d7bc5a342eb92b248c3b4e68a07f/display/glDisplay.cpp somewhere between lines 269 and 342 fail.

madisongh, 2020-10-25:
Yeah, I think there are issues with trying to connect NVIDIA's OpenGL X support up with an XWayland window.  I don't remember the details, though.

ichergui, 2021-09-07:
Hey @mozcelikors 

Any update concerning this issue ?  
Did you manage to resolve your issue ?

ichergui, 2021-10-04:
Will close this issue because no updates 
For any issue similar we can re-open new issue

closed 2021-10-04 by ichergui

vigneshoff, 2022-03-03:
> Hi again @madisongh,
> 
> I'm having trouble making CMake recognize cuda host binaries. Latest state of the recipe file is as follows:
> 
> ```
> SUMMARY = "jetson-inference"
> DESCRIPTION = "jetson-interence by dusty-nv"
> LICENSE = "CLOSED"
> 
> RDEPENDS_${PN}_append = " libpython3 v4l-utils python3 python3-dev python3-numpy cuda-nvcc cuda-cudart cuda-toolkit cuda-target-environment cuda-libraries cuda-driver qtbase libpthread-stubs"
> DEPENDS_append = " libpthread-stubs-native python3-native python3-numpy-native v4l-utils qtbase"
> 
> FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
> 
> SRC_URI += " git://www.github.com/dusty-nv/jetson-inference"
> SRC_URI += " file://src_override/utils/cuda/FindCUDA.cmake"
> SRC_URI += " file://src_override/utils/python/bindings/CMakeLists.txt"
> 
> S = "${WORKDIR}/git"
> SRCREV = "022739abc7631fa853ae63550ec091cd838b2105"
> 
> inherit pkgconfig cmake cmake_qt5 cuda python3native python3-dir
> 
> # Setting a CMake policy
> EXTRA_OECMAKE += " -DCMAKE_POLICY_DEFAULT_CMP0079=NEW"
> 
> do_configure_prepend() {
>   # Initialize all submodules
>   cd ${WORKDIR}/git
>   git submodule update --init --recursive
> 
>   # Update our modified FindCUDA.cmake with original after submodules are initialized,
>   # the update is required to supress duplicate add_library target_link_libraries errors.
>   cp ${WORKDIR}/src_override/utils/cuda/FindCUDA.cmake ${WORKDIR}/git/utils/cuda
> 
>   # Update our modified utils/python/bindings/CMakeLists.txt with original one.
>   cp ${WORKDIR}/src_override/utils/python/bindings/CMakeLists.txt ${WORKDIR}/git/utils/python/bindings/CMakeLists.txt
> }
> ```
> 
> I'm getting:
> 
> `: /media/mozcelikors/ext4/Yocto_Builds/jetson_nano_dunfell/build/tmp/work/armv8a_tegra210-poky-linux/jetson-inference/1.0-r0/recipe-sysroot/usr/local/cuda-10.2/bin/nvcc: cannot execute binary file`
> 
> in do_compile step now.
> 
> When I inherit cuda, "nvcc" executable is used from the recipe-sysroot (target). I don't understand why. I am on branch dunfell, cuda version is 10.2, CMake version is 3.16. I read in cuda.bbclass that for CMake >3.8 toolchain file is used, so I also tried with `EXTRA_OECMAKE += " -DCMAKE_TOOLCHAIN_FILE=${WORKDIR}/toolchain.cmake"`. But no use. My source overrides are for suppressing some do_configure errors (related to pthread inclusion + CUDA_VERSION calculation), But the root issue is the one I mentioned above.
> 
> Any help is greately appreciated.

Can you please share the modification which you have made in FindCUDA.cmake and CMakeLists.txt files.

I am also getting the same kind of issue.

And also does this package dont require jetson-utils as a dependency

Thanks
Vignesh

Sid3y1, 2023-09-25:
> Can you please share the modification which you have made in FindCUDA.cmake and CMakeLists.txt files.
>
> I am also getting the same kind of issue.
>
> And also does this package dont require jetson-utils [...]

I'm interessted too !