OE4T/meta-tegra #446: Is meta-tegra registered  as Yocto Project® Compatible ?

fischer, 2020-10-20:
Apologies in advance for wasting time on a non-technical topic.

Has anyone filled out the Yocto compatible form for meta-tegra at https://www.yoctoproject.org/ecosystem/branding/compatible-registration ?

I don't feel comfortable representing meta-tegra, or I would fill it out.  As far as I can tell, using the word *Yocto* in the README.md file requires compatibility registration.  From https://www.yoctoproject.org/logos-and-guidelines :

> Any use of {Y,y}-o-{C,c}-{T,t}-o Project or any subset of those letters or sounds in that order, alone or in combination with other letters, words or marks that would tend to cause someone to make a reasonable connection with the Yocto Project has to be in reference to code that is in conformance with the Yocto Project Compliance Program.

koansoftware, 2020-10-21:
Hi Todd,
why you don't feel comfortable representing meta-tegra?
You (or your company) are the creator and mantainer of the project so probably you would be the main responsible for it.
The registration process is very easy and pain free.
Feel free to ask me anything you consider not clear enough.
Best,
--
_Marco Cavallini
{ Yocto Project Ambassador }
{ KOAN  C.E.O. }
Bergamo - Italia
embedded software engineering
https://KoanSoftware.com_

madisongh, 2020-10-21:
> Has anyone filled out the Yocto compatible form for meta-tegra at https://www.yoctoproject.org/ecosystem/branding/compatible-registration ?

Not yet... branding hasn't really been on my radar so far, but I'll try submitting an entry to see what they say.

> You (or your company) are the creator and mantainer of the project so probably you would be the main responsible for it.

Ahem, but no, that is incorrect.

fischer, 2020-10-21:
Matt,

The form looks to take around 5 minutes to fill out.  They at least made it easy.

Thanks for looking into it.

Todd

madisongh, 2020-10-21:
OK, I've submitted both a 'Participant' form for the project, and a 'Compatible' form for the layer itself.  I'll post here if/when I get a response.

koansoftware, 2020-10-21:
@madisongh very good!

/marco

closed 2020-12-11 by madisongh