OE4T/meta-tegra #447: Question about linux-tegra for Xavier

danielwangksu, 2020-10-21:
I'm using Warrior branch with the latest backport to bundle initramfs. I noticed that in `build/tmp/work/jetson_xavier-poky-linux/linux-tegra/4.9+gitAUTOINC+40031e9439-r0/deploy-linux-tegra/deploy-linux-tegra` there are 2 bins:
* Image--4.9+git0+40031e9439-r0-jetson-xavier-20201013220733.bin - 34.3MB
* Image-initramfs--4.9+git0+40031e9439-r0-jetson-xavier-20201013220733.bin - 44.6MB
What is the different between those two? It seems rootfs.img actually using the 34.3MB one. Is that support to be that way?

madisongh, 2020-10-21:
* `Image...bin` is just the kernel without the initramfs bundled.
* `Image-initramfs....bin` is the kernel with the bundled initramfs.  You should also see an `Image-initramfs...cboot`, which is the Android-style boot image that cboot uses to load the kernel+initramfs (it also has a slot for a separate initrd, but that will be empty).

The `.cboot` file is the one that gets loaded into the `kernel` partition when flashing, since that's the format that cboot understands.

As for the BUP payloads, the names give a clue:
* `bl_only.bup_payload` = bootloader only (i.e., no kernel)
* `kernel_only.bup_payload` = kernel/kernel device tree only (i.e., no bootloaders)
* Just `bup_payload` = full BUP payload with both bootloader and kernel

danielwangksu, 2020-10-21:
Thank you @madisongh for your detailed explanation. It seems that Xavier requires requires kernel partition to be signed (using `oe_make_payload` if I understand it correctly) the kernel file md5sum matches `bup_payload/signed/boot_signheader.img.encrypt`. So the `boot_signheader.img.encrypt` is actually signed version of `Image-initramfs...cboot`?

madisongh, 2020-10-21:
Yes, and yes.  When you aren't using secure boot (with your own signing and encryption keys), the kernel/initramfs boot image is "signed" using an all-zero signing key. I believe this is done as a way to verify the integrity of the partition contents during boot.

danielwangksu, 2020-10-21:
Thank you so much for your help! I was confused between TX2i and Xavier. Your respond help clarified all my confusions (as always!)

closed 2020-10-21 by danielwangksu