OE4T/meta-tegra #449: Issues with Xavier switch_root

danielwangksu, 2020-10-22:
I have a question on bundled initramfs switching to real rootfs.img. I use a slightly different version of `meta-tegra/recipes-core/initrdscripts/tegra-minimal-init/init-boot.sh` I modified to fit my scenario.

Instead of mount emmc and switch to that emmc partition, my script create a new ramfs then load a cpio formatted rootfs.img into that ramfs and then switch into it. This works fine with Xavier dev kit but when adding additional initscripts the kernel panic and reboot (I will attach the log in the end) 

Here is my init-boot.sh:
```
echo "Mounting ${rootdev}..."
mkdir /mnt/rootfs
mount -t ext4 ${rootdev} /mnt/rootfs
mkdir /mnt/switch-root
mount -o size=2G -t tmpfs none /mnt/switch-root
cp /mnt/rootfs/boot/rootfs.img /mnt/switch-root/rootfs.img.gz
cd /mnt/switch-root
gzip -d rootfs.img.gz
cpio -i < rootfs.img
rm rootfs.img

echo "Switching to rootfs on /mnt/rootfs/boot/rootfs.img..."
mount --move /sys  /mnt/switch-root/sys
mount --move /proc /mnt/switch-root/proc
mount --move /dev  /mnt/switch-root/dev
exec switch_root /mnt/switch-root /sbin/init
```

Did I do something wrong? Any potential issue that I have not considered before? Thank you very much!

```
Dump for CPU0:
pid: 3587  comm: udevd
  x0 0000000000000040   x1 000000006f3e6f38
  x2 0000000000000006   x3 0000000000006f38
  x4 ffffffbf0fa417d0   x5 ffffff8008245f00
  x6 ffffffc3b69d3e20   x7 afb504000afb5041
  x8 0000000000000000   x9 ffffffc31ca53780
 x10 0000000000000a20  x11 0000000000001c18
 x12 000000000000a167  x13 00000000008b712a
 x14 0000000000000001  x15 ffffffc3ececec28
 x16 0000000000000003  x17 0000000000000002
 x18 0000000000000f43  x19 ffffff800aa57ac0
 x20 ffffff800aa560c0  x21 ffffffc3ffdafe20
 x22 0000000000000000  x23 ffffff80081d6b98
 x24 0000000000000000  x25 ffffff800aa57ac0
 x26 0000000000001a00  x27 ffffffbf0daf2780
 x28 ffffffc3bbcc7000  x29 ffffffc31ca536f0
 x30 ffffff80081d7ed4   sp ffffffc31ca536f0
  pc ffffff8008f54374 cpsr 004000c5 (EL1h)

_raw_spin_lock_irqsave+0x54/0x70:
  pc ffffff8008f54374   sp ffffffc31ca536f0   fp ffffffc31ca536f0
pagevec_lru_move_fn+0x8c/0x120:
  pc ffffff80081d7ed4   sp ffffffc31ca53700   fp ffffffc31ca53710
lru_add_drain_cpu+0x114/0x120:
  pc ffffff80081d8adc   sp ffffffc31ca53720   fp ffffffc31ca53770
lru_add_drain+0x38/0x68:
  pc ffffff80081d8db0   sp ffffffc31ca53780   fp ffffffc31ca537a0
exit_mmap+0x6c/0x118:
  pc ffffff8008209ffc   sp ffffffc31ca537b0   fp ffffffc31ca537c0
mmput+0x60/0x130:
  pc ffffff80080b0528   sp ffffffc31ca537d0   fp ffffffc31ca53880
do_exit+0x26c/0xa08:
  pc ffffff80080b95fc   sp ffffffc31ca53890   fp ffffffc31ca538a0
bug_handler.part.2+0x0/0x88:
  pc ffffff800808c528   sp ffffffc31ca538b0   fp ffffffc31ca53910
bad_mode+0x88/0x98:
  pc ffffff800808cc20   sp ffffffc31ca53920   fp ffffffc31ca53950
handle_serr+0x124/0x128:
  pc ffffff800808cdf4   sp ffffffc31ca53960   fp ffffffc31ca53980
el1_serr+0xb8/0x148:
  pc ffffff8008082d58   sp ffffffc31ca53990   fp ffffffc31ca53b10
pagevec_lru_move_fn+0x8c/0x120:
[... 202 lines cut ...]
  pc ffffff8008265844   sp ffffffc31ca5be20   fp ffffffc31ca5be90
el0_svc_naked+0x34/0x38:
  pc ffffff80080838c0   sp ffffffc31ca5bea0   fp 0000000000000000
debug>   x0 0000000000000040   x1 000000006f3d6f38
  x2 0000000000000005   x3 0000000000006f38
  x4 0000000000000039   x5 ffffffc3e999bfc0
  x6 ffffffc3ee2aa000   x7 0000000000000000
  x8 0000000000000010   x9 000000046474e552
 x10 000000000001bc18  x11 000000000002bc18
 x12 000000000002bc18  x13 00000000000003e8
 x14 00000000000003e8  x15 ffffffffffffffff
 x16 30636d6d2f74736f  x17 30303a30636d6d2f
 x18 0000000000000001  x19 ffffff800aa57ac0
 x20 ffffff800aa560c0  x21 ffffffc3ffddde20
 x22 0000000000000000  x23 ffffff80081d6b98
 x24 0000000000000000  x25 ffffff800aa57ac0
 x26 0000000000001a00  x27 ffffffbf0db0fbc0
 x28 ffffffc3b74b9790  x29 ffffffc31ca5bae0
 x30 ffffff80081d7ed4   sp ffffffc31ca5bae0
  pc ffffff8008f54374 cpsr 004000c5 (EL1h)
 sp_el0   ffffffc3c8f28e00
 elr_el1  ffffff8008f54838
 spsr_el1 00400045
debug> ��
```

madisongh, 2020-10-23:
From the stack trace on CPU0, looks like the processor is reporting a data abort exception.  Maybe a misbehaving driver?

danielwangksu, 2020-10-23:
> From the stack trace on CPU0, looks like the processor is reporting a data abort exception. Maybe a misbehaving driver?

Thank you! I could be I will try to debug it.

ichergui, 2021-03-03:
Hey @danielwangksu 

Could you please give us an update about your issue ? Did you manage to fix it ?

danielwangksu, 2021-03-03:
Hi @ichergui yes I fixed this issue. The problem was in kernel variables and drivers in my specific setup. Thank you.

closed 2021-03-03 by danielwangksu

ichergui, 2021-03-03:
> Hi @ichergui yes I fixed this issue. The problem was in kernel variables and drivers in my specific setup. Thank you.

Perfect