OE4T/meta-tegra #451: Troubles with cuda binaries on dunfell-l4t-r32.4.3

chhg82, 2020-10-26:
Hi,
I cloned the dunfell branche from git://git.yoctoproject.org/poky.git and the dunfell-l4t-r32.4.3 branche from here on ubuntu 16.04 and 18.04 and also on openSUSE 15.1. With the nvidia sdk I downloaded the cuda target drivers 16.04 and 18.04 and tried to bitbake with following build configuration:

local.conf:
> MACHINE = "jetson-xavier"
INHERIT += "rm_work"
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash tar.gz"
EXTRA_IMAGE_FEATURES += "package-management"
NVIDIA_DEVNET_MIRROR = "/home/jupidu/Downloads/nvidia/sdkm_downloads_ubuntu1804"
#CUDA_VERSION="10.2"
CUDA_BINARIES = "cuda-binaries-ubuntu1804"
GCCVERSION_aarch64 = "8.%"

bblayer.conf:
> POKY_BBLAYERS_CONF_VERSION = "2"
BBPATH = "${TOPDIR}"
BBFILES ?= ""
BBLAYERS ?= "\\
  /home/jupidu/yocto/poky/meta \\
  /home/jupidu/yocto/poky/meta-poky \\
  /home/jupidu/yocto/poky/meta-yocto-bsp \\
  \\
  /home/jupidu/yocto/poky/meta-tegra \\
  /home/jupidu/yocto/poky/meta-tegra/contrib \\
  "

I have following errors in all my operation systems:
> ERROR: /home/jupidu/yocto/poky/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804_10.2.89-1.bb: Error executing a python function in \<code\>:
The stack trace of python calls that resulted in this exception/failure was:
File: '\<code\>', lineno: 3, function: \<module\>
     0001:__anon_20__home_jupidu_yocto_poky_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_32__home_jupidu_yocto_poky_meta_classes_patch_bbclass(d)
 *** 0003:__anon_707__home_jupidu_yocto_poky_meta_classes_base_bbclass(d)
     0004:__anon_202__home_jupidu_yocto_poky_meta_classes_reproducible_build_bbclass(d)
     0005:__anon_1130__home_jupidu_yocto_poky_meta_classes_insane_bbclass(d)
     0006:__anon_1370__home_jupidu_yocto_poky_meta_classes_insane_bbclass(d)
     0007:__anon_252__home_jupidu_yocto_poky_meta_classes_package_bbclass(d)
File: '/home/jupidu/yocto/poky/meta/classes/base.bbclass', lineno: 590, function: __anon_707__home_jupidu_yocto_poky_meta_classes_base_bbclass
     0586:
     0587:    needsrcrev = False
     0588:    srcuri = d.getVar('SRC_URI')
     0589:    for uri in srcuri.split():
 *** 0590:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
     0591:
     0592:        # HTTP/FTP use the wget fetcher
     0593:        if scheme in ("http", "https", "ftp"):
     0594:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
File: '/home/jupidu/yocto/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 357, function: decodeurl
     0353:    """
     0354:
     0355:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0356:    if not m:
 *** 0357:        raise MalformedUrl(url)
     0358:
     0359:    type = m.group('type')
     0360:    location = m.group('location')
     0361:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: '/home/jupidu/Downloads/nvidia/sdkm_downloads_ubuntu1604/cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb' is invalid and cannot be interpreted
ERROR: Failed to parse recipe: /home/jupidu/yocto/poky/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804_10.2.89-1.bb

This error stays even with a wrong NVIDIA_DEVNET_MIRROR, but everything works fine without cuda drivers.

Any ideas? Is it a python problem?
Thanks!

madisongh, 2020-10-26:
Your NVIDIA_DEVNET_MIRROR is missing the `file://` URL prefix.  Try adding that, as in:
```
NVIDIA_DEVNET_MIRROR = "file:///home/jupidu/Downloads/nvidia/sdkm_downloads_ubuntu1804"
```
and the fetcher should stop complaining.

chhg82, 2020-10-26:
Great, thank you!!! :-)

closed 2020-10-26 by chhg82