OE4T/meta-tegra #458: Real time kernel patches recently broken

jlucke86, 2020-10-28:
gist of failure <script src="https://gist.github.com/jlucke86/6df3fe717716415a6a1d401e6392e4c6.js"></script>

PREMPT RT was working properly with meta-tegra on branch dunfell-l4t-r32.4.3 at commit https://github.com/OE4T/meta-tegra/commit/34402290f8de86e2ad59b17a6d05be39bfdf2597

I've updated to https://github.com/OE4T/meta-tegra/commit/799b14568d70de9a868d5cd8abe9eabfa8a8bd95

Now I'm getting errors compiling the kernel modules. I can find old forum posts for other board bsp's (nxp etc) asking for help with failures to build the real time kernel and the support provided was it is the incorrect defconfig being used.

madisongh, 2020-10-28:
Are you using a custom `defconfig` by any chance?  I just tried building a kernel after following the instructions [on the wiki](https://github.com/OE4T/meta-tegra/wiki/Applying-PREEMPT-RT-Patches-dunfell-l4t-r32.4.3) and it worked OK for me, but from the error you show in the log, it looks like your kernel config is missing CONFIG_PREEMPT_RT_FULL.  If you're using a custom `defconfig` from a non-PREEMPT_RT build, make sure you modify your config to match the changes applied by the `scripts/rt-patch.sh` script to the default configs in the kernel tree.

jlucke86, 2020-10-28:
At this stage the kernel sources have 2 patches in the bbappend. One for the nano pinmux settings in the device tree. And another for the RT patches. I will manually patch the source and patch it by commit and hash and make sure it's not my process.

I'll also check the variable you mentioned

jlucke86, 2020-11-11:
@madisongh OK agreed it is still working.

What I cannot get working is making a patch by collecting the changes from applying the real time patch script and then appending it in rather than forking the repo and patching the kernel source directly.

meta-intel has a preferred virtual kernel provider option where the user can specify they want the RT Linux. If we were to do that we would need to add another branch to linux-tegra-4.9 or alternatively get the RT patches working from a single patch file

madisongh, 2020-11-11:
@jlucke86 

I just tried generating a patch by going through the steps on the wiki up to the `git commit` and then using `git format-patch HEAD~` to generate a patch file that I added to the SRC_URI in my bbappend, and that worked OK.  What didn't work for you?

As an alternative, I also tried adding this to my bbappend, and it looks like it's doing the right thing:
```
do_patch_prepend() {
    oldwd=$PWD
    cd ${S}/scripts
    ./rt-patch.sh apply-patches
    cd ${S}
    git add . && git commit -m 'Apply PREEMPT_RT patches'
    cd $oldwd
}
```
It's maybe not great if you have other patches you need to apply on top of the RT patches, though.

jlucke86, 2020-11-12:
I tested your do_patch_prepend(). It makes the same error as me patching the sources using the git format-patch process.

Manually forking and patching still works but it's not very maintainable. I really don't understand where it's going wrong

jlucke86, 2020-11-12:
Could my issues be because I also have appends on recipes-bsp/tegra-binaries/tegra-flashvars/ for my xavier nx platform?

madisongh, 2020-11-12:
> Could my issues be because I also have appends on recipes-bsp/tegra-binaries/tegra-flashvars/ for my xavier nx platform?

That wouldn't have any effect on building the kernel.  Do you have anything else besides the RT patch in your kernel bbappend?  Adding any configuration changes, other patches, etc.?

You mentioned that you were seeing errors when compiling the kernel modules... can you attach the log file, or at least parts of it with the errors you're seeing?

jlucke86, 2020-11-12:
I will take some time to throw a complete minimal example together

jlucke86, 2021-01-14:
I spent another day or so looking for the issue. I can definitely see that the problem is the rt patches to the tegra_defconfig are not propagating into the stage where it compiles the kernel modules.

Anyway here is the link to the complete minimal example umbrella project setup with the same failure as the gist above.

https://github.com/jlucke86/jetson-yocto-example

I will help out with documentation / instructions when I get this patch working properly. I have multiple patches and custom device tree work I need to get working on after these issues are understood.

madisongh, 2021-01-14:
Thanks, @jlucke86, that helps.  The problem is that the `defconfig` file gets generated before the `do_patch`, so the application of the RT patch set needs to be inserted before that.  Replacing the `do_patch_append()` I mentioned earlier with this:
```
do_rt_preempt_patch() {
    ./rt-patch.sh apply-patches
}
do_rt_preempt_patch[dirs] = "${S}/scripts"
addtask rt_preempt_patch after do_validate_branches before do_kernel_metadata
```
works better.  It ensures that the patch gets applied after the kernel sources are fixed at the `${SRCREV}` hash, but before the `defconfig` gets generated.

jlucke86, 2021-01-15:
Thanks for your reply. I tested that and still get the same error.

madisongh, 2021-01-15:
Well, I forked your [jetson-yocto-example](https://github.com/madisongh/jetson-yocto-example) and [meta-example](https://github.com/madisongh/meta-example) repos, made the above [change](https://github.com/madisongh/meta-example/commit/a4009178f8a2d59226855b28983ad5be5b0aa94d), tried building the kernel again, and got no errors. The kernel config that was generated correctly includes the PREEMPT_RT-related additions (the lack of which was causing the compilation errors before).

If you're still getting the error with your minimal example environment, check against the changes I made in my forks.  If you're getting the error in your real environment, check to see if there's something else that's preventing the kernel config from including the needed additions for PREEMPT_RT.

closed 2021-09-01 by madisongh