OE4T/meta-tegra #459: HTTP 404 errors for CUDA packages on dunfell-l4t-r32.4.3 branch

Leo-PL, 2020-10-28:
Hi,

I'm trying to build CUDA 10.2 for Jetson TX2 board, but I'm quite surprised to see HTTP 404 errors during fetching many of packages, namely:
```
recipes-devtools/cuda/cuda-npp_10.2.89-1.bb
recipes-devtools/cuda/cuda-driver_10.2.89-1.bb
recipes-devtools/cuda/cuda-nvrtc_10.2.89-1.bb
recipes-devtools/cuda/libcublas_10.2.2.89-1.bb
recipes-devtools/cuda/cuda-misc-headers_10.2.89-1.bb
recipes-devtools/cuda/cuda-cusolver_10.2.89-1.bb
recipes-devtools/cuda/cuda-curand_10.2.89-1.bb
recipes-devtools/cuda/cuda-nvcc-headers_10.2.89-1.bb
recipes-devtools/cuda/cuda-nvgraph_10.2.89-1.bb
recipes-devtools/cuda/cuda-cusparse_10.2.89-1.bb
recipes-devtools/cuda/cuda-cufft_10.2.89-1.bb
recipes-devtools/cuda/cuda-cudart_10.2.89-1.bb
recipes-devtools/cuda/cuda-samples_10.2.89-1.bb
recipes-devtools/cuda/cuda-nvml_10.2.89-1.bb
```
At first glance it looks like nVidia just nuked (or moved) their repository with arm64 .deb packages. Do you know, by any chance any alternative source for those packages? Devnet possibly, like for native part?

I also could not find `cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb` anywhere on the 'net, even in official amd64 repository, the closest one being `cuda-repo-ubuntu1804-10-2-local-10.2.89-440.33_1.0-1_amd64.deb`

madisongh, 2020-10-28:
NV has had some issues with their CDN recently.  I just tried a download of one of the CUDA packages and it worked OK, but I did see this myself for a while last week.

You can use the NVIDIA SDK Manager to download the CUDA packages for the target, then extract the contents of the `cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb` and copy all of the `.deb` files it provides into the DL_DIR (typically $BUILDDIR/downloads) for your build workspace.  For example:
```
$ cd $HOME/Downloads/nvidia/sdkm_downloads
$ dpkg-deb --fsys-tarfile cuda-repo-l4t-10-2-local-10.2.89_1.0-1_arm64.deb | tar -C /path/to/your/builddir/downloads -xvf- --strip-components=3 --wildcards "./var/cuda-repo-10-2-local-10.2.89/*.deb"
```

To download the host-side tools, use the NVIDIA SDK Manager and set the NVIDIA_DEVNET_MIRROR variable to point to its downloads directory, as described in the [README](https://github.com/OE4T/meta-tegra/blob/master/README.md).

Leo-PL, 2020-10-28:
Thanks for the idea - once I get my hands on the cuda-repo .deb, I'll try this, once I sign up for SDK manager, to which I'm pretty reluctant, but I see no alternative means.

FYI, I tried downloading the smaller packages from a few hosts in different locations I know, and 3 Tor endpoints - all of them gave me 404, same for a few friends.

Leo-PL, 2020-10-29:
Huh, it seems I had wrong CUDA_VERSION set in machine config causing all of this - I was upgrading my BSP from L4T 28.2 .
Thanks for your help.

closed 2020-10-29 by Leo-PL