OE4T/meta-tegra #463: Zeus: nvv4l2h265enc: Error getting capabilities for device '/dev/nvhost-msenc'

quaresmajose, 2020-10-29:
The `nvv4l2h265enc` doesn't  work in zeus-l4t-r32.3.1
```
gst-launch-1.0 videotestsrc ! nvv4l2h265enc ! fakesink
Setting pipeline to PAUSED ...
libv4l2: error getting capabilities: Inappropriate ioctl for device
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/nvv4l2h265enc:nvv4l2h265enc0: Error getting capabilities for device '/dev/nvhost-msenc': It isn't a v4l2 driver. Check if it is a v4l1 driver.
Additional debug info:
v4l2_calls.c(98): gst_v4l2_get_capabilities (): /GstPipeline:pipeline0/nvv4l2h265enc:nvv4l2h265enc0:
system error: Inappropriate ioctl for device
Setting pipeline to NULL ...
Freeing pipeline ...
```

The kernel send me this:
```
[  225.718522] (NULL device *): nvhost_channelctl: invalid cmd 0x80685600
```

madisongh, 2020-10-29:
Does `/dev/nvhost-msenc` exist? If it doesn't, check `lsmod` to see if the `nvgpu` driver is loaded.  Depending on what image you're building, the driver may not have been included automatically in your build (e.g., core-image-minimal).

Also, I don't think you can connect videotestsrc directly to nvv4l2h265enc, since the encoder requires NVMM buffers.  This is mentioned (although not necessarily abundantly clear) in the [Accelerated GStreamer](https://docs.nvidia.com/jetson/l4t/Tegra%20Linux%20Driver%20Package%20Development%20Guide/accelerated_gstreamer.html#) section of the L4T documentation.  When I try your pipeline on my builds, gstreamer reports an 'erroneous pipeline'.

Your options are either to use the (still present, but deprecated) `omxh265enc` element, or to use, for example, `nvvidconv` to convert the normal raw buffer to NVMM for you:
```
$ gst-launch-1.0 videotestsrc ! nvvidconv ! nvv4l2h265enc ! fakesink
```

The `nvhost_channelctl:` message you see from the kernel is due to one of the V4L2 plugins (libv4l-mplane, IIRC) sending a V4L2 ioctl to one of the NVIDIA drivers that doesn't support that ioctl.  It's safe to ignore.

quaresmajose, 2020-10-29:
I write this sort pipeline as an example. This error comes from a big one that work in previous versions of zeus
```
gst-launch-1.0 videotestsrc ! nvv4l2h265enc ! fakesink
```
I can add `nvvidconv` but the result will be the same, I will check this.

I need to check the presence of `nvgpu` kernel module.

Thanks for your quick answer.

quaresmajose, 2020-10-29:
This doesn't work as you said
```
gst-launch-1.0 videotestsrc ! nvv4l2h265enc ! fakesink
WARNING: erroneous pipeline: could not link videotestsrc0 to nvv4l2h265enc0
```

To simplify
```diff
--videotestsrc
++fakesrc
```

```
gst-launch-1.0 fakesrc ! nvv4l2h265enc ! fakesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/nvv4l2h265enc:nvv4l2h265enc0: Error getting capabilities for device '/dev/nvhost-msenc': It isn't a v4l2 driver. Check if it is a v4l1 driver.
Additional debug info:
v4l2_calls.c(98): gst_v4l2_get_capabilities (): /GstPipeline:pipeline0/nvv4l2h265enc:nvv4l2h265enc0:
system error: Inappropriate ioctl for device
Setting pipeline to NULL ...
Freeing pipeline ...
```

And the last I have the `nvgpu` kernel module loaded
```
lsmod|grep nvgpu
nvgpu                1720663  0
```

madisongh, 2020-10-29:
Do you have the NVIDIA libv4l2 plugins at `/usr/lib/libv4l/plugins`?  There should be `libv4l2_nvvidconv.so` and `libv4l2_nvvideocodec.so` there.  If not, make sure you add `tegra-libraries-libv4l-plugins` to your image.

quaresmajose, 2020-10-29:
I have it place:
```
ls -l /usr/lib/libv4l/plugins
total 44
-rw-r--r-- 1 root root 32000 Oct 27  2020 libv4l2_nvvidconv.so
-rw-r--r-- 1 root root 10112 Oct 27  2020 libv4l2_nvvideocodec.so
```

I think  the problems comes in:

meta: update L4T BSP from R32.2.3 to R32.3.1 …
https://github.com/OE4T/meta-tegra/commit/9c0da22392f34e0b2c43ceba5a548051e14bf099

or

meta: update L4T from R32.2.1 to R32.2.3
https://github.com/OE4T/meta-tegra/commit/9e6e40d5d8a1bacc4691b52ff8315f080bda041d

I need to revert this commits because i have a previews `zeus` version that work as expected with `nvv4l2h265enc`

madisongh, 2020-10-29:
Hmm. I running an image on my Nano from the latest off the `zeus-l4t-r32.3.1` branch, and everything is working as expected.

quaresmajose, 2020-11-01:
I said that the problem comes in 9c0da22 or 9e6e40d but it's not true.
I have cherry-pick 262783e and 847049f but it not solves.
I have try to find it with different kernels version but I can't find where the problem comes.

I forgot to mention that this problem is on the TX2 and I only use OE core layer

madisongh, 2020-11-02:
OK, I was able to reproduce this.  The other change you need to cherry-pick is 0e0119b5ef92f482fca57accf811b9f9b2d8e57e. This omits some NVIDIA-provided binary-only libraries that are interfering with normal libv4l2 and are trying to open the v4l2 plugins on a non-existent path (from their Ubuntu file system layout).

With that change in place, I get a more normal-looking failure.  By adding `gstreamer1.0-plugins-tegra` to my image and adding `nvvidconv` to the pipeline, I can now see the `gst_buffer_resize_range` errors you mention on #471 .

quaresmajose, 2020-11-03:
Thank you for the clarification.
The solution for this are on https://github.com/OE4T/meta-tegra/commit/0e0119b5ef92f482fca57accf811b9f9b2d8e57e as you said. I will try it and came back with the result.

quaresmajose, 2020-11-07:
The commit 0e0119b fix the issue and now the nvvideo4linux2 works as expected.
I tested it in diferente setups: encoding, decoding and transcoding.
Thank you very match.

PR #483 (2020-11-07): zeus-l4t-r32.3.1: Fix nvv4l2h265enc getting capabilities for device
Fix #463 [Zeus: nvv4l2h265enc: Error getting capabilities for device '/dev/nvhost-msenc']

Backport:
```
    tegra-libraries: don't install libnvv4l2, libnvv4lconvert
    
    These are not directly linked against by anything, and don't
    appear to provide any functionality that isn't provided by
    the open-source libv4l2 libraries, so just remove them.

    Signed-off-by: Matt Madison <matt@madison.systems>
```

This is needed on zeus to work properly with the gstreamer nvv4l2 plugin.
closed 2020-11-07 by quaresmajose