OE4T/meta-tegra #464: qtwayland on jetson-xavier

chhg82, 2020-10-29:
Hi,

I try to bitbake `IMAGE_INSTALL_append += "qtwayland"` (finally goal is a running qt5-creator) from meta-qt5 (dunfell branch) for jetson-xavier with dunfell-l4t-r32.4.3 branch and get following error:

> ERROR: qtwayland-5.14.2+gitAUTOINC+af3976e5e7-r0 do_configure: Error calling /home/jupidu/yocto-nvidia-dunfell/poky/build/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/recipe-sysroot-native/usr/bin/qmake -makefile -o Makefile    QT_BUILD_PARTS-=examples QT_BUILD_PARTS-=tests   /home/jupidu/yocto-nvidia-dunfell/poky/build/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/git/qtwayland.pro  --   -no-feature-wayland-brcm -feature-wayland-client -no-feature-wayland-drm-egl-server-buffer -feature-wayland-egl -no-feature-wayland-libhybris-egl-server-buffer -feature-wayland-server -feature-xcomposite-egl -feature-xcomposite-glx
ERROR: qtwayland-5.14.2+gitAUTOINC+af3976e5e7-r0 do_configure: Execution of '/home/jupidu/yocto-nvidia-dunfell/poky/build/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/temp/run.do_configure.20760' failed with exit code 1:
Running configuration tests...
Project ERROR: Test config.qtwayland_client.tests.dmabuf-server-buffer tries to use undeclared library 'drm'
WARNING: /home/jupidu/yocto-nvidia-dunfell/poky/build/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/temp/run.do_configure.20760:1 exit 1 from 'exit 1'
ERROR: Logfile of failure stored in: /home/jupidu/yocto-nvidia-dunfell/poky/build/tmp/work/aarch64-poky-linux/qtwayland/5.14.2+gitAUTOINC+af3976e5e7-r0/temp/log.do_configure.20760

Why is libdrm not available, any idea?

kekiefer, 2020-10-29:
I am using qtwayland and it works great. It has been a long time since I set it up, so the details of what exactly I needed to do are lost, but I'll try to help. Not using qtcreator on target, so I won't be able to help there.

For my setup, I don't have any non-default configuration options set for qtwayland itself, but do have a number of non-default configuration options for qtbase. One critical thing you'll need to enable so that the eglfs-kms-egldevice backend will be built:

qtbase_%.bbappend
```
PACKAGECONFIG_append_tegra = " kms"
```

This would also happen to pull in the drm dependency.

Another thing to try is making sure your distro features are set the way you intend to use the system. You might try disabling x11, and make sure wayland is enabled.

```
DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES_append = " gles2 wayland"
```

chhg82, 2020-10-30:
Thank you kekiefer!

With `PACKAGECONFIG_append_tegra += "kms"` in my `./build/conf/local.conf` it compiles without errors but I get a lot of warnings:
> …
WARNING: libgudev-233-r0 do_configure: QA Issue: libgudev: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: librsvg-2.40.21-r0 do_configure: QA Issue: librsvg: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: vulkan-headers-1.1.126.0-r0 do_configure: QA Issue: vulkan-headers: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: mobile-broadband-provider-info-1_20190618-r0 do_configure: QA Issue: mobile-broadband-provider-info: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: vulkan-loader-1.1.126.0-r0 do_configure: QA Issue: vulkan-loader: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: python3-pycairo-1.19.0-r0 do_configure: QA Issue: python3-pycairo: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: avahi-0.7-r0 do_configure: QA Issue: avahi: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: dbus-glib-0.110-r0 do_configure: QA Issue: dbus-glib: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: ofono-1.31-r0 do_configure: QA Issue: ofono: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtquickcontrols-5.14.2+gitAUTOINC+8c3882d939-r0 do_configure: QA Issue: qtquickcontrols: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtserialport-5.14.2+gitAUTOINC+30807cadfc-r0 do_configure: QA Issue: qtserialport: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtwebsockets-5.14.2+gitAUTOINC+13c3b3edc5-r0 do_configure: QA Issue: qtwebsockets: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtsvg-5.14.2+gitAUTOINC+63961dd0c7-r0 do_configure: QA Issue: qtsvg: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qttools-5.14.2+gitAUTOINC+9d5d92a492-r0 do_configure: QA Issue: qttools: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtx11extras-5.14.2+gitAUTOINC+fbed3bc300-r0 do_configure: QA Issue: qtx11extras: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: gstreamer1.0-plugins-base-1.16.2-r0 do_configure: QA Issue: gstreamer1.0-plugins-base: invalid PACKAGECONFIG: kms [invalid-packageconfig]
WARNING: qtscript-5.14.2+gitAUTOINC+bdf42faf19-r0 do_configure: QA Issue: qtscript: invalid PACKAGECONFIG: kms [invalid-packageconfig
…

Where exactly should I put the PACKAGECONFIG? I could not find the file qtbase_%.bbappend.
Thx!

kekiefer, 2020-10-30:
That override should be in a bbappend file in your own layer, e.g. `./recipes-qt/qt5/qtbase_%.bbappend`.

If you don't have your own layer, and just want to do this for testing, you can use it in the local.conf like this:

```
PACKAGECONFIG_append_pn-qtbase = " kms"
```

chhg82, 2020-11-05:
Thx a lot!

closed 2020-11-05 by chhg82