OE4T/meta-tegra #465: Problem using tegra-bup-payload/generate_bup_payload with user partitions

klakaiser, 2020-10-30:
Hello,
i have an issue with using tegra-bup-payload and generate_bup_payload script. I have added some user partitions to the flash.xml
and therefore had to override tegraflash_create_flash_config_tegra186(). Now when trying to generate the bup payload the process complains that it is not able to open the files (user partition images).
tegra-bup-payload is trying to access the non sed'ed filename of the partition.
generate_bup_payload only works when removing the filenames for the user partitions.

BTW, can you tell me what the correct way of patching the smd_info.cfg used by tegra-binaries recipe to enable boot redundancy from beginning is?

Thank you

I am using TX2 with dunfell-32.3.4

lfdmn, 2020-10-30:
Hi,

For stripping the names set the ```IMG_STRIP_NAMES``` with the image name you want to exclude in your distro config. 
Ex:
```
IMG_STRIP_NAMES = "persistent.img,overlay.img"
```

For smd_info.cfg, bbappend tegrabinaries. I have added smd_info_ab_enabled.cfg file and copied it to the correct location ```do_preconfigure_append```

smd_info_ab_enabled.cfg :
```
 # Copyright (c) 2017-2018, NVIDIA CORPORATION. All rights reserved.
 #
 # Permission is hereby granted, free of charge, to any person obtaining a
 # copy of this software and associated documentation files (the "Software"),
 # to deal in the Software without restriction, including without limitation
 # the rights to use, copy, modify, merge, publish, distribute, sublicense,
 # and/or sell copies of the Software, and to permit persons to whom the
 # Software is furnished to do so, subject to the following conditions:
 #
 # The above copyright notice and this permission notice shall be included in
 # all copies or substantial portions of the Software.
 #
 # THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 # IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 # FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
 # THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 # LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 # FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
 # DEALINGS IN THE SOFTWARE.

# SMD metadata information
< VERSION 3 >

#
# Config 1: Disable A/B support (Default)
#

# slot info order is important!
# <priority>    <suffix>     <retry_count>  <boot_successful>
#15                  _a          7               1

#
# Config 2: Enable redundancy support (by removing comments ##)
#
< REDUNDANCY_USER 1 >

# slot info order is important!
# <priority>    <suffix>     <retry_count>  <boot_successful>
15                  _a          7               1
14                  _b          7               1
```

tegra-binaries_32.%.bbappend:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SRC_URI += " \
    file://flash_l4t_custom.xml \
    file://smd_info_ab_enabled.cfg \
"

do_preconfigure_append() {
    mkdir -p ${S}/bootloader/${NVIDIA_BOARD}/cfg/
    cp ${WORKDIR}/flash_l4t_custom.xml ${S}/bootloader/${NVIDIA_BOARD}/cfg/

    cp ${WORKDIR}/smd_info_ab_enabled.cfg ${S}/bootloader/
}

```

Then bbappend tegra-bootfiles and set
```
SMD_CFG = "${S}/bootloader/smd_info_ab_enabled.cfg"
```

klakaiser, 2020-10-30:
Thanks for that info.
Can you tell me how that IMG_STRIP_NAMES is getting used (what uses it)? I would have expected some kind of manipulation of the flash.xml.in in case of generating the bup payloads or a parameter passed into the tegratools used for the generation.

lfdmn, 2020-10-30:
Oh yeah, there's piece missing =)

I've made a own image type which has

my_distro/classes/image_types_custom.bbclass
```
inherit image_types image_types_tegra pythonnative perlnative

# Strip filename tags for the img names
tegraflash_generate_bupgen_script_append() {
    if [ ! -z "${IMG_STRIP_NAMES}" ]; then
        strip_sed="" 
        for i in $(echo ${IMG_STRIP_NAMES} | tr "," "\n")
        do
            echo ${strip_sed}
        	strip_sed="${strip_sed} -e'/${i}/d' " 
        done
        sed -i -e"1a sed -i ${strip_sed} ./flash.xml.in" $outfile
    fi 
}
...
```
Then in your distro add ```IMAGE_CLASSES += "image_types_custom"```

It removes the image names just like you mentioned before, so these partitions won't belong to the payload.

If I recall right, to be able to append a bbclass function, you have to make a own class which inherits the original one. It's been a long while so I might be wrong.

Here's the original class ```meta-tegra/classes/image_types_tegra.bbclass``` generating the ```generate_bup_payload.sh``` script.

klakaiser, 2020-11-02:
Thank you very much , that makes it absolute clear.

closed 2020-11-02 by klakaiser