OE4T/meta-tegra #474: Compatibility issue with openembedded-core and ccache.

schvarcz, 2020-11-03:
Good evening everyone.

First, I would like to say thank you very much for this amazing meta layer you have been providing to the community. It is an essencial stone in our stack and this is why I really appreciate the work you put on this repo. 

I have been learning a lot reading the shared code you made available to us and this is why I am coming here today. I guess I found an issue that probably you should be aware in order to ensure compatibility with the main meta layers available. 

I have an issue but also a possible fix I would like to share and debate here. :)

## Context
We have a project where we are using basically `meta-openembedded`, `meta-tegra` and our homemade layer.

At our project, we are using `ccache` by using the `ccache.bbclass` provided by the [openembedded-core](http://cgit.openembedded.org/openembedded-core/tree/meta/classes/ccache.bbclass?h=dunfell) repo.

## Problem

There are a couple of ways that someone could use `ccache`. The `ccache.bbclass` chose to use `ccache` by appending its name in front of the compilation command. This way, they avoid creating symbolic links to every compiler everywhere.

Therefore the compilations:
```sh
g++ source.cpp
```
Will become:
```sh
ccache g++ source.cpp
```

Therefore, making use of the cache.

They do that by changing the `CXX` environment variable to something like:
```sh
export CXX="ccache aarch64-g++ [flags]"
```

Changing the position where the actual compiler name would be placed.

Coming back to `meta-tegra`, you have a very useful class named `cuda.bbclass` where at [line 16](https://github.com/OE4T/meta-tegra/blob/cd149cd5bfee77fbf0209130f0a7f02c3b520449/classes/cuda.bbclass#L16) the code tries to access the compiler name by splitting the `CXX` variable and taking the first argument.

If the project is using `ccache`, **that code would think that the compiler name is `ccache`**. Failing, of course, at the first code it would try to compile with a very not informative message.

## Possible solutions

#### 1. Being simplistic (but would not work):

The most simple and direction solution would be changing the [line 16](https://github.com/OE4T/meta-tegra/blob/cd149cd5bfee77fbf0209130f0a7f02c3b520449/classes/cuda.bbclass#L16):
```sh
export CUDAHOSTCXX = "${@d.getVar('CXX').split()[0]}"
```
to something like:
```sh
export CUDAHOSTCXX = "${@" ".join(d.getVar('CXX').split()[0,2]) if d.getVar('CXX').split()[0] == 'ccache' else d.getVar('CXX').split()[0]"
```

Bringing, this way, the `ccache` name together with the right compiler name.

Unfortunately, it would not work because at some step of its process, `nvcc` is doing a similar check as `meta-tegra` and do not recognize the correct compiler name in the end. (at least, this was the case when I tried to compile `opencv` with `cuda`. I can share the error message if needed)

#### 2. Ignore ccache (and the user's request to use a cache):

```sh
export CUDAHOSTCXX = "${@d.getVar('CXX').split()[1] if d.getVar('CXX').split()[0] == 'ccache' else d.getVar('CXX').split()[0]}"
```

Even though this approach works, it doesn't seem very integrated with the other layers (and with what the user is requesting to do). But it is as a possible solution to avoid an unnecessary error.

#### 3. Creating your local symbolic link:

Since the `ccache.bbclass` method does not work in this case, we are forced to use the `ccache` by a local symbolic link.

I would rather make it local to the current recipe in order to not generate useless links in the system and/or to other recipes.

```sh
CCACHE_CUDAHOSTCXX_DIR="${STAGING_LIBDIR}/ccache"
CUDAHOSTCXX_CCACHE="${CCACHE_CUDAHOSTCXX_DIR}/${@d.getVar('CXX').split()[1]}"
export CUDAHOSTCXX = "${@d.getVar('CUDAHOSTCXX_CCACHE') if d.getVar('CXX').split()[0] == 'ccache' else d.getVar('CXX').split()[0]}"

do_prepare_ccache() {
    if [ "${@d.getVar('CXX').split()[0]}" = "ccache" ]; then
    	mkdir -p ${CCACHE_CUDAHOSTCXX_DIR}
    	if [ ! -e $CUDAHOSTCXX ]; then
            ln -s $(which ccache) $CUDAHOSTCXX
    	fi
    fi
}

addtask prepare_ccache after do_patch before do_configure
```

In the code above, I decided to create the actual symbolic link just before the `do_configure` step, when it would be referenced by the first time. If you can think of a better "time-space" position to this linking step, please let me know!

In the code above, I broke the `CUDAHOSTCXX` variable in several steps to make it easier to read, but be aware that the `CUDAHOSTCXX_CCACHE` line could break the building if the variable `CXX` has only one part.

#### 4. Live and let die:

If, by any chance, you decide that none of these solutions is suitable to `meta-tegra` project, could you at least add a warning clarifying that we should not use `ccache` with `meta-tegra` somewhere? Thanks :)

## To Finish

I implemented the third solution as a new `bbclass` that inherits `cuda.bbclass`. So, my particular issue is solved. However, you might like to be aware of this compatibility issue and these possible solutions. 

Also, maybe someone here could spot a potential issue with my current solution that I am not aware. If this is the case, I would be very grateful to hear what you are thinking. :)

As I commented at the start, I am new to the yocto project and I have been learning with your code. Thank you for that. Thank you once again for this amazing code. And thank for being this extraordinary person I know you are!!

Have a nice day.

madisongh, 2020-11-04:
Thanks for the detailed report.  It looks like there are two somewhat related issues here.  One is to get `-ccbin` set correctly on the `nvcc` commands, which means detecting if `ccache` is there and adjusting things accordingly if so.

The second is to get CUDA compilations to support caching via ccache.  For that, it looks like we should be setting `CUDACXX` to `"${CCACHE}${CUDA_TOOLKIT_ROOT}/bin/nvcc"` to pass the entire nvcc command line to ccache, rather than trying to use ccache under the hood for nvcc's internal invocation(s) of the C++ compiler.

Getting that to play nicely with CMake, too, might mean a bit of additional work.  Looks like both CMake and ccache were updated in recent years to be able to handle using ccache for nvcc compilations, so in theory, at least, it should be doable.

As you can probably tell, I don't use ccache myself, so it may take some time for me to get around to trying this out.  I'm happy to improve compatibility with ccache, I just want to make sure that doing so doesn't break existing use cases, and that it does work for different recipe-level build systems - at least regular makefiles and CMake, and hopefully meson as well.

schvarcz, 2020-11-04:
Hello @madisongh.

Thank you for the quick answer. 

Yes, I am also concerned that the final solution should work for any case. For this reason, I am glad to discuss this with you when you need. If you need any test or log report from my side, let me know and I will be glad to provide.

About your concern with `nvcc`. You are right. I was thinking that `ccache.bbclass` would take care of it by adding `ccache` in front of any compiler name.  But `nvcc` is defined in another env variable. And indeed:

```sh
export CUDACXX='/home/[LONG-PATH-HERE]/recipe-sysroot-native/usr/local/cuda-${CUDA_VERSION}/bin/nvcc'
```
it is missing the ccache.

My natural intuition says that I should add this to my current `bbclass`, but if you can think on a better design or draft anything where I should investigate further and help you in the process, I would be happy to hear.

Kind regards.

madisongh, 2020-11-05:
Have a look at #481, let me know what you think.

schvarcz, 2020-11-06:
Thank you so much for this patch @madisongh !

I will try this branch as soon as I can and I will come back here to inform you if it worked.

madisongh, 2020-11-25:
@schvarcz Have you had a chance to try out the changes?

schvarcz, 2020-11-26:
Hello @madisongh 

Yes, I did and it seems to work perfectly.

However, in order to test it with our pipeline, I needed these modifications on `dunfell` branch. In order to make it available so you could follow what I tested, I forked this project and created this PR https://github.com/schvarcz/meta-tegra/pull/1 to let you see the exact modifications I applied.

They are basically the same as your commits, this is why I left you as the author, a part from this commit here https://github.com/schvarcz/meta-tegra/commit/49c6d4ad0d066b4bb55d162d182bf354ff74a6ce related to your commit here https://github.com/OE4T/meta-tegra/commit/930662004750ea98293e7f194fccbd59526256cc, where I had to change it a little bit since the file was not exactly the same. I think that the difference is only related to `meson`, not sure if we should add it to `dunfell`.

If you find it useful, I can create the same PR on this repository. If you prefer do these commits by yourself, when you will be ready to merge them to `dunfell`, make yourself at home. =)

----------

A part from that, I am wondering if we are not missing a `ccache` at `-ccbin` option. To be more precise, I am referring to this https://github.com/schvarcz/meta-tegra/commit/47026a18e0dbcb5ff2c90b8db8fdcefada812289.

When I read the `nvcc` documentation [here](https://docs.nvidia.com/cuda/cuda-compiler-driver-nvcc/index.html#file-and-path-specifications-compiler-bindir), I understand that if it is not a `cuda` file (ending by `cu`, for instance), `nvcc` will call whatever is written on `-ccbin` to do a host compilation.

If this is the case, I believe we should add `${CCACHE}` there too, right?

I tested this modification with my building system and it works fine, even though, until now, I didn't notice any difference in terms of building time.

----------

That is the feedback I have for now. Sorry for the time taken to answer this issue. I wanted to have a better understanding about the `-ccbin` option before coming back to here. However, maybe you could have a better understanding of the `-ccbin` option and decide if it worths adding the `ccache` there too or not.

Please, let me know if you are planning to merge this to `dunfell` and/or I could help on this issue in another way.

Kind regards

madisongh, 2020-11-26:
>When I read the nvcc documentation here, I understand that if it is not a cuda file (ending by cu, for instance), nvcc will call whatever is written on -ccbin to do a host compilation.

>If this is the case, I believe we should add ${CCACHE} there too, right?

I don't think so; nvcc expects a single word as the argument for --ccbin, so we can't really wedge in the ${CCACHE} prefix. I don't think it matters, though...  looking at the ccache test cases, they always pass the real C/C++ compiler as the argument to --ccbin.  The output will get cached by ccache, so the next time the same nvcc command comes along, ccache should find it.

As far as back-porting to dunfell... I usually take care of back-porting once changes go into master, but since you've already done the work for dunfell, I'd be happy to take a PR from you for them.

Thanks!

closed 2020-11-27 by madisongh

schvarcz, 2020-11-28:
Hello again.

I read this message just now and I can see that you merged them already.

Nice work! Thank you.