OE4T/meta-tegra #475: Build Errors for Jetson TX2 Yocto

gnaneshdev, 2020-11-04:
Hi,

I have followed below steps to build yocto image for jetson tx2 but getting build errors.

Pre-requisites: 1. Installed gcc-8, g++-8 as mentioned at https://github.com/OE4T/tegra-demo-distro in my ubuntu 18.04 machine.
2. Created Nvidia developer account and downloaded all HOST COMPONENTS packages in ~/Download/nvidia/sdkm_downloads
Downloaded host Components are (Cuda, Computer Vision, Developer Tools).

Performed below steps.
1. git clone https://github.com/OE4T/tegra-demo-distro.git
2. cd tegra-demo-distro
3. git submodule update --init
4. source ./setup-env --distro tegrademo --machine jetson-tx2
5. Set the NVIDIA_DEVNET_MIRROR to ~/Download/nvidia/sdkm_downloads in ~/tegra-demo-distro/build/conf/local.conf file.
6. bitbake demo-image-base

On running these steps, I get the following error.

ERROR: cuda-binaries-ubuntu1804-native-10.2.89-1-r0 do_preconfigure: Execution of '/home/YoctoLinux/mytx2build/tegra-demo-distro/build/tmp/work-shared/x86_64/cuda-binaries-10.2.89-1-r0/temp/run.do_preconfigure.16570' failed with exit code 2:
dpkg-deb: error: failed to read archive '/home/YoctoLinux/mytx2build/tegra-demo-distro/build/tmp/work-shared/x86_64/cuda-binaries-10.2.89-1-r0/var/cuda-repo-10-2-local-10.2.89-440.40/cuda-license-10-2_10.2.89-1_amd64.deb': No such file or directory
WARNING: exit code 2 from a shell command.

ERROR: Logfile of failure stored in: /home/YoctoLinux/mytx2build/tegra-demo-distro/build/tmp/work-shared/x86_64/cuda-binaries-10.2.89-1-r0/temp/log.do_preconfigure.16570

As I see my sdkm_downloads folder has following packages w.r.t cuda but missing those native packages mentioned in above error.

cuda-license-10-2_10.2.89-1_amd64.deb
cuda-repo-cross-aarch64-10-0-local-10.0.326_1.0-1_all.deb
cuda-repo-l4t-10-0-local-10.0.326_1.0-1_arm64.deb
cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb
cuda-repo-ubuntu1804_10.2.89-1_amd64.deb

Please let me know if I am missing something here.

Thanks!

madisongh, 2020-11-04:
Double-check your NVIDIA_DEVNET_MIRROR setting... for the demo distro, the `setup-env` script should set it automatically for you in your local.conf.  Did you change what it set for you?

gnaneshdev, 2020-11-04:
I believe it was not set. Had to set it explicitly in local.conf. 

I have also set in ~/tegra-demo-distro/repos/meta-tegra/classes/nvidia_devnet_downloads.bbclass file as well. Not sure if this is required.

madisongh, 2020-11-04:
Make sure it's in URL format, then, i.e. `"file:///path/to/the/directory"`, and make sure that `cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb` is present and correct - should be around 1.8GB in size.  I don't see that particular `deb` file mentioned in your earlier post.

dwalkes, 2020-11-04:
> I believe it was not set. Had to set it explicitly in local.conf.

It looks like we've got it commented out currently at https://github.com/OE4T/tegra-demo-distro/blob/dunfell-l4t-r32.4.3/layers/meta-tegrademo/conf/template-tegrademo-mender/local.conf.sample#L35 in the wrong format.  I think it should be 
```
#NVIDIA_DEVNET_MIRROR ?= "file:///${HOME}/Downloads/nvidia/sdkm_downloads"
```
something happened with the magic [here](https://github.com/OE4T/tegra-demo-distro/blob/dunfell-l4t-r32.4.3/setup-env#L100) that's supposed to fill in based on `@TD_SETUP_DEVNET_MIRROR@ ` at https://github.com/OE4T/tegra-demo-distro/blob/dunfell-l4t-r32.4.3/layers/meta-tegrademo/conf/template-tegrademo-mender/local.conf.sample#L34

@gnaneshdev can you share what you see in your local.conf generated by `setup-env` on [these lines](https://github.com/OE4T/tegra-demo-distro/blob/dunfell-l4t-r32.4.3/layers/meta-tegrademo/conf/template-tegrademo-mender/local.conf.sample#L35)?  Could you try `cp build/conf ~/conf.old` followed by `rm -rf build/conf` and then re-run setup-env to see if you can reproduce whatever caused this logic to fail?

madisongh, 2020-11-04:
> It looks like we've got it commented out ... in the wrong format.

Yep, we should fix that.  I'm just surprised that the magic didn't work like it should have.

gnaneshdev, 2020-11-05:
Yes, it is as specified below
NVIDIA_DEVNET_MIRROR ?= "file:///home/YoctoLinux/Downloads/nvidia/sdkm_downloads".

I don't see this specific package is downloaded by sdkmanager. 
I was looking at this place https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/ 
but this package is missing. Is it available somewhere

gnaneshdev, 2020-11-05:
@dwalkes  I see that line is not being enabled even after running "source ./setup-env --distro tegrademo --machine jetson-tx2"
#NVIDIA_DEVNET_MIRROR ?= "${HOME}/Downloads/nvidia/sdkm_downloads"

so had to add this line
NVIDIA_DEVNET_MIRROR ?= "file:///home/YoctoLinux/Downloads/nvidia/sdkm_downloads"

I have also done following steps as suggested.

cp -af build/conf ~/conf.old
rm -rf build/conf
source ./setup-env --distro tegrademo --machine jetson-tx2

Surprisingly this has same old following two lines.

NVIDIA_DEVNET_MIRROR ?= "file:///home/YoctoLinux/Downloads/nvidia/sdkm_downloads"
#NVIDIA_DEVNET_MIRROR ?= "${HOME}/Downloads/nvidia/sdkm_downloads"
Is it caching or picking it up from somewhere?

And one more thing, I added this NVIDIA_DEVNET_MIRROR variable in following file(~/mytxt2build/tegra-demo-distro/repos/meta-tegra/classes/nvidia_devnet_downloads.bbclass) as well

#NVIDIA_DEVNET_MIRROR ??= "file://NOTDEFINED" 
NVIDIA_DEVNET_MIRROR ??= "file:///home/YoctoLinux/Downloads/nvidia/sdkm_downloads"

Is this step necessary?

dwalkes, 2020-11-05:
```
Surprisingly this has same old following two lines.

NVIDIA_DEVNET_MIRROR ?= "file:///home/YoctoLinux/Downloads/nvidia/sdkm_downloads"
#NVIDIA_DEVNET_MIRROR ?= "${HOME}/Downloads/nvidia/sdkm_downloads"
Is it caching or picking it up from somewhere?
```

That's actually exactly what we'd expect, see link above to the magic I referenced [here](https://github.com/OE4T/tegra-demo-distro/blob/dunfell-l4t-r32.4.3/setup-env#L99-L106) which should be filling this in for you based on logic [here](https://github.com/OE4T/tegra-demo-distro/blob/9b7ea42d43641b028a19680a12c79a9be7d78967/scripts-setup/setup-env-internal#L146-L161) and the value of `TD_DEVNET_MIRROR`.

So you shouldn't need to setup `NVIDIA_DEVNET_MIRROR` yourself with the step you referenced above because the `setup-env` script should be (and looks like it is?) doing this for you.

> Set the NVIDIA_DEVNET_MIRROR to ~/Download/nvidia/sdkm_downloads in ~/tegra-demo-distro/build/conf/local.conf file.

This shouldn't be necessary.

So it looks like the problem though is that you still aren't finding the cuda repo there.  Here's what I see:

```
yocto@yocto-Latitude-E6540:~$ cd ~/Downloads/nvidia/sdkm_downloads/
yocto@yocto-Latitude-E6540:~/Downloads/nvidia/sdkm_downloads$ find . -name "cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb"
./cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
yocto@yocto-Latitude-E6540:~/Downloads/nvidia/sdkm_downloads$ ls -la cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
-rw-rw-r-- 1 yocto yocto 1904275712 Nov  4 09:12 cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
yocto@yocto-Latitude-E6540:~/Downloads/nvidia/sdkm_downloads$ md5sum cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
286beb1a09da17efcef278930384f86b  cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
```
If the file isn't there, I suspect you didn't get it downloaded from the NVIDIA SDK manager somehow.  Check the logs there to try to figure out why.

gnaneshdev, 2020-11-10:
@dwalkes  I have looked at log ~/.nvsdkm/sdkm.log and could see "cuda-repo-ubuntu1804-10-0-local-10.0.326-410.108_1.0-1_amd64.deb" installed successfully but it is missed to download "cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb".

If this package available somewhere on web to download, I can copy manually to sdkm_downloads folder and try rebuilding.

madisongh, 2020-11-10:
@gnaneshdev Looks like you downloaded JetPack 4.3 (which had CUDA 10.0), rather than JetPack 4.4 or 4.4.1 (both of which have CUDA 10.2).  Is your SDK Manager up to date?  You may need to install the latest version of SDK Manager to download the current JetPack components.

gnaneshdev, 2020-11-11:
@madisongh  I have sdkmanager version (1.3.1.7110) installed. Is there a new version to this?

dwalkes, 2020-11-11:
@gnaneshdev 1.3.1 is the right version, so the next question is whether you've successfully downloaded Jetpack 4.4 or 4.4.1.  What do you see here?
[image]

vaishnavils97, 2020-11-27:
@dwalkes , I have installed jetPack 4.4 Version and have provided the NVIDIA_DEVNET_MIRROR path to sdkm_downloads. But When I tried to build the demo-full-image. I was getting the below error.

ERROR: /home/TegraVirtual/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604_10.2.89-1.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 3, function: <module>
     0001:__anon_20__home_TegraVirtual_tegra_demo_distro_layers_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_32__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_patch_bbclass(d)
 *** 0003:__anon_707__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_base_bbclass(d)
     0004:__anon_202__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_reproducible_build_bbclass(d)
     0005:__anon_1130__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_insane_bbclass(d)
     0006:__anon_1370__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_insane_bbclass(d)
     0007:__anon_252__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_package_bbclass(d)
File: '/home/TegraVirtual/tegra-demo-distro/layers/meta/classes/base.bbclass', lineno: 590, function: __anon_707__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_base_bbclass
     0586:
     0587:    needsrcrev = False
     0588:    srcuri = d.getVar('SRC_URI')
     0589:    for uri in srcuri.split():
 *** 0590:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
     0591:
     0592:        # HTTP/FTP use the wget fetcher
     0593:        if scheme in ("http", "https", "ftp"):
     0594:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
File: '/home/TegraVirtual/tegra-demo-distro/repos/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 357, function: decodeurl
     0353:    """
     0354:
     0355:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0356:    if not m:
 *** 0357:        raise MalformedUrl(url)
     0358:
     0359:    type = m.group('type')
     0360:    location = m.group('location')
     0361:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: '/home/TegraVirtual/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1604-10-2-local-10.2.89-440.40_1.0-1_amd64.deb' is invalid and cannot be interpreted

ERROR: Failed to parse recipe: /home/TegraVirtual/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1604_10.2.89-1.bb
 

So I have changed the file tegra-demo-distro/repos/meta-tegra/classes/nvidia_devnet_downloads.bbclass: NVIDIA_DEVNET_MIRROR ??= "file:///home/TegraVirtual/Downloads/nvidia/sdkm_downloads"

Now I am getting the below error
ERROR: /home/TegraVirtual/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804_10.2.89-1.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 3, function: <module>
     0001:__anon_20__home_TegraVirtual_tegra_demo_distro_layers_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_32__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_patch_bbclass(d)
 *** 0003:__anon_707__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_base_bbclass(d)
     0004:__anon_202__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_reproducible_build_bbclass(d)
     0005:__anon_1130__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_insane_bbclass(d)
     0006:__anon_1370__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_insane_bbclass(d)
     0007:__anon_252__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_package_bbclass(d)
File: '/home/TegraVirtual/tegra-demo-distro/layers/meta/classes/base.bbclass', lineno: 590, function: __anon_707__home_TegraVirtual_tegra_demo_distro_layers_meta_classes_base_bbclass
     0586:
     0587:    needsrcrev = False
     0588:    srcuri = d.getVar('SRC_URI')
     0589:    for uri in srcuri.split():
 *** 0590:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
     0591:
     0592:        # HTTP/FTP use the wget fetcher
     0593:        if scheme in ("http", "https", "ftp"):
     0594:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
File: '/home/TegraVirtual/tegra-demo-distro/repos/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 357, function: decodeurl
     0353:    """
     0354:
     0355:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0356:    if not m:
 *** 0357:        raise MalformedUrl(url)
     0358:
     0359:    type = m.group('type')
     0360:    location = m.group('location')
     0361:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: '/home/TegraVirtual/Downloads/nvidia/sdkm_downloads/cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb' is invalid and cannot be interpreted

ERROR: Failed to parse recipe: /home/TegraVirtual/tegra-demo-distro/layers/meta-tegra/recipes-devtools/cuda/cuda-binaries-ubuntu1804_10.2.89-1.bb

@dwalkes I have cross checked the ubuntu package in downloads with .bb file , it was same. Please give a solution for this issue.

dwalkes, 2020-11-27:
It looks like your `SRC_URI` is malformed, look really closely at what you have in your `build/conf/local.conf` compared to https://github.com/OE4T/tegra-demo-distro/blob/master/layers/meta-tegrademo/conf/template-tegrademo-mender/local.conf.sample#L36.  I suspect you are missing the `file://` prefix.  I just noticed we have the same error in the template-tegrademo comment for `local.conf.sample` so this might be confusing things as well.

vaishnavils97, 2020-12-04:
@dwalkes , Thanks for the right input. I am able to build the image without any errors. I have added meta-iotedge and meta-rust layers to this build. But once the image is flashed the Azure Iotedge is not available in the yocto. Below are the changes I have made inorder to include the meta-iotedge layer.
1. Added the meta-iotedge and meta-rust on this path:~/home/tegra-demo-distro/repos/meta-iotedge (dunfell branch)
                                                                                        :~/home/tegra-demo-distro/repos/meta-rust (master branch)
2. Added the meta-iotedge and meta-rust on this path:- /home/tegra-demo-distro/layers/meta-iotedge(dunfell)
                                                                                          /home/TegraVirtual/tegra-demo-distro/layers/meta-rust(master)
3. Added the path on build/conf/bblayers.conf    
                                                                                  /home/tegra-demo-distro/layers/meta-iotedge \
                                                                                  /home/tegra-demo-distro/layers/meta-rust \

Is there any provision to see whether meta-iotedge layer have added to the yocto image.

vaishnavils97, 2020-12-10:
I have flashed the Yocto image to Jetson tx2. But we are not able to find meta-iotedge layer on Jetson after flashing. So please provide the input to check whether the layer meta-iotedge has added properly.

madisongh, 2020-12-10:
@vaishnavils97 Sorry, but I have no idea what meta-iotedge is.  Is it [this](https://github.com/Azure/meta-iotedge)?  Do you see the layer `$BUILDDIR/conf/bblayers.conf` file?  If not, follow the instructions in that layer's README file to add it.

Having the layer in your build setup is the first step... the next step would be to add a package built from that layer to whatever image you are building.  Hopefully that layer provides documentation on what you need to do for that.

dwalkes, 2020-12-10:
>  the next step would be to add a package built from that layer to whatever image you are building. Hopefully that layer provides documentation on what you need to do for that.

You can also reference https://wiki.yoctoproject.org/wiki/Cookbook:Example:Adding_packages_to_your_OS_image 
https://hub.mender.io/t/how-to-create-custom-images-using-yocto-project/902
https://hub.mender.io/t/how-to-install-packages-in-yocto-project/1004
and https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#openembedded-build-system-build-concepts
 for some more hints/tips and a better understanding of the overall process.

ichergui, 2021-03-01:
Hey @gnaneshdev 

Could you please give us an update ? Are you able to build for `Jetson TX2` without issues ?
Please let us know if you need help.

ichergui, 2021-03-28:
No updates here, I will close this issue.

closed 2021-03-28 by ichergui