OE4T/meta-tegra #480: Issue with flashing the image on Jetson TX2

chandshiv, 2020-11-05:
Hello,

Have followed the steps mentioned in the below link to build the minimalist image and to flash it on to the Jetson TX2 device,
https://hub.mender.io/t/nvidia-tegra-jetson-tx2/123

We have encountered an error mentioned below while running the ./doflash.sh script inside to flash the build to the device.
[image]

We followed below steps to enter the recovery mode of the device,

- Power cycle
- Hold down the RST button and REC button
- Release the RST button while continuing to hold the REC button
- Wait two seconds, then release the REC BUTTON

And could see the NVIDIA device listed in the lsusb device output of my Ubuntu Virtual box VM.

Also, have run this lsusb -d 0955:7c18 to make sure the device is in recovery mode and could get the output.

Please do let us know if there are any steps are missed.

dwalkes, 2020-11-05:
Hi @chandshiv 

> And could see the NVIDIA device listed in the lsusb device output of my Ubuntu Virtual box VM.

In my experience it's not reliable to use a VM for tegraflashing, suggest using a physical installation for this step.  Not sure if others have had other observations.

ichergui, 2020-11-05:
Hi @chandshiv 

The same here, I didn't use VM for tegra flashing.

Could you please try the commands line below after switching your device to RCM ? Please share the result.

```
$ sudo ./tegraflash.py --chip 0x18 --applet mb1_recovery_prod.bin --cmd "dump eeprom boardinfo cvm.bin"
```
 if the execution of command above is Okay, try the next one:

```
$ sudo ./chkbdinfo -p cvm.bin
$ sudo ./chkbdinfo -g cvm.bin
```
 
**Note:** These commands are valid if secure boot is not enabled.

chandshiv, 2020-11-06:
> witching your device to RCM

chandshiv, 2020-11-06:
> Hi @chandshiv
>
> The same here, I didn't use VM for tegra flashing.
>
> Could you please try the commands line below after switching your device to RCM ? Please share the result.
>
> ```
> $ sudo [...]

Unfortunately we cant login to the device as we don't have the password and also default username and password (nvidia and ubuntu) did not work.
Please let me know if we could reset the password otherwise

dwalkes, 2020-11-06:
Assuming you haven't changed [EXTRA_IMAGE_FEATURES settings](https://github.com/OE4T/tegra-demo-distro/blob/9b7ea42d43641b028a19680a12c79a9be7d78967/layers/meta-tegrademo/conf/template-tegrademo-mender/local.conf.sample#L136) and are using [debug-tweaks](https://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#ref-features-image) it should be an empty root password.

chandshiv, 2020-11-06:
> Assuming you haven't changed [EXTRA_IMAGE_FEATURES [...]

Yes its correct, we have not changed. We followed below link for building the minimalist image 
https://github.com/OE4T/tegra-demo-distro

radek-uavia, 2020-11-06:
I once had similar errors. My mistake was related to the following ([source](https://forums.developer.nvidia.com/t/jetpack-4-2-flashing-issues-and-how-to-resolve/73387)):

> lsusb alone is not sufficient. Need put device to recovery mode again. Specifically, lsusb simply shows that the Jetson device is attached to your PC and in recovery mode. However, there is a one-off setup communication that occurs between the PC and Jetson each time flashing is attempted, and once this has happened, any flash attempt will fail unless Jetson is forced back into recovery mode, even if it already appears to be in recovery mode.

So if you try to flash once, and it fails, it will keep failing because the device is not in recovery mode anymore. In this situation I had the same errors as you. Powering off the device, and rebooting it in recovery again to attempt a new flash fixed it for me.

It might not be your issue maybe, but it's worth keeping in mind when you try to flash.

ichergui, 2020-11-06:
> Unfortunately we cant login to the device as we don't have the password and also default username and password (nvidia and ubuntu) did not work.
> Please let me know if we could reset the password [...]

No need to the login and password here. power your device, and force it into the recovery mode and try the command line.

chandshiv, 2020-11-09:
> > > Hi @chandshiv
> > > The same here, I didn't use VM for tegra flashing.
> > > Could you please try the commands line below after switching your device to RCM ? Please share the result.
> > > ```
> > > $ sudo ./tegraflash.py --chip 0x18 --applet mb1_recovery_prod.bin --cmd "dump eeprom boardinfo cvm.bin"
> > > ```
 
Please find the execution result of the given command 
[   0.0119 ] Generating RCM messages
[   0.0189 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm mb1_recovery_prod.bin 0 0
[   0.0221 ] RCM 0 is saved as rcm_0.rcm
[   0.0347 ] RCM 1 is saved as rcm_1.rcm
[   0.0365 ] List of rcm files are saved in rcm_list.xml
[   0.0472 ] 
[   0.0473 ] Signing RCM messages
[   0.0559 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0588 ] Assuming zero filled SBK key
[   0.0899 ] 
[   0.0901 ] Copying signature to RCM mesages
[   0.0940 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
[   0.1110 ] 
[   0.1113 ] Boot Rom communication
[   0.1199 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml
[   0.1257 ] BR_CID: 0x81801001645102820800000013fc0380
[   0.1474 ] RCM version 0X180001
[   0.1576 ] Boot Rom communication completed
[   1.2098 ] 
[   2.2149 ] tegrarcm_v2 --isapplet
[   2.2321 ] USB communication failed.Check if device is in recovery
[   3.4237 ] 
[   3.4410 ] tegradevflash_v2 --iscpubl
[   3.4610 ] CPU Bootloader is not running on device.
[   3.6700 ] 
[   4.6847 ] tegrarcm_v2 --isapplet
[   4.6918 ] Applet version 01.00.0000
[   4.9059 ] 
[   4.9183 ] Retrieving EEPROM data
[   4.9186 ] tegrarcm_v2 --oem platformdetails eeprom cvm /home/chandrika/Downloads/cvm.bin
[   4.9211 ] Applet version 01.00.0000
[   5.1133 ] Saved platform info in /home/chandrika/Downloads/cvm.bin
[   5.4257 ] 

> No need to the login and password here. power your device, and force it into the recovery mode and try the command line.

ichergui, 2020-11-09:
> > > > Hi @chandshiv
> > > > The same here, I didn't use VM for tegra flashing.
> > > > Could you please try the commands line below after switching your device to RCM ? Please share the result.
> > > > ```
> > > > $ sudo ./tegraflash.py --chip 0x18 --applet mb1_recovery_prod.bin --cmd "dump eeprom boardinfo cvm.bin"
> > > > ```
> 
> Hi @ichergui
> Please find the execution result of the given command
> [ 0.0119 ] Generating RCM messages
> [ 0.0189 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm mb1_recovery_prod.bin 0 0
> [ 0.0221 ] RCM 0 is saved as rcm_0.rcm
> [ 0.0347 ] RCM 1 is saved as rcm_1.rcm
> [ 0.0365 ] List of rcm files are saved in rcm_list.xml
> [ 0.0472 ]
> [ 0.0473 ] Signing RCM messages
> [ 0.0559 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
> [ 0.0588 ] Assuming zero filled SBK key
> [ 0.0899 ]
> [ 0.0901 ] Copying signature to RCM mesages
> [ 0.0940 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
> [ 0.1110 ]
> [ 0.1113 ] Boot Rom communication
> [ 0.1199 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml
> [ 0.1257 ] BR_CID: 0x81801001645102820800000013fc0380
> [ 0.1474 ] RCM version 0X180001
> [ 0.1576 ] Boot Rom communication completed
> [ 1.2098 ]
> [ 2.2149 ] tegrarcm_v2 --isapplet
> [ 2.2321 ] USB communication failed.Check if device is in recovery
> [ 3.4237 ]
> [ 3.4410 ] tegradevflash_v2 --iscpubl
> [ 3.4610 ] CPU Bootloader is not running on device.
> [ 3.6700 ]
> [ 4.6847 ] tegrarcm_v2 --isapplet
> [ 4.6918 ] Applet version 01.00.0000
> [ 4.9059 ]
> [ 4.9183 ] Retrieving EEPROM data
> [ 4.9186 ] tegrarcm_v2 --oem platformdetails eeprom cvm /home/chandrika/Downloads/cvm.bin
> [ 4.9211 ] Applet version 01.00.0000
> [ 5.1133 ] Saved platform info in /home/chandrika/Downloads/cvm.bin
> [ 5.4257 ]
> 
> > > > if the execution of command above is Okay, try the next one:
> > > > ```
> > > > $ sudo ./chkbdinfo -p cvm.bin
> > > > $ sudo ./chkbdinfo -g cvm.bin
> > > > ```
> > > > 
> > > > 
> > > > **Note:** These commands are valid if secure boot is not enabled.
> > > 
> > > 
> > > Unfortunately we cant login to the device as we don't have the password and also default username and password (nvidia and ubuntu) did not work.
> > > Please let me know if we could reset the password otherwise
> > 
> > 
> > No need to the login and password here. power your device, and force it into the recovery mode and try the command line.

Please try the following commands and share the result:
```
$ sudo ./chkbdinfo -p cvm.bin
$ sudo ./chkbdinfo -g cvm.bin
```

chandshiv, 2020-11-09:
> sudo ./chkbdinfo -g cvm.bin

> > > > > Hi @chandshiv
> > > > > The same here, I didn't use VM for tegra flashing.
> > > > > Could you please try the commands line below after switching your device to RCM ? Please share the result.
> > > > > ```
> > > > > $ sudo ./tegraflash.py --chip 0x18 --applet mb1_recovery_prod.bin --cmd "dump eeprom boardinfo cvm.bin"
> > > > > ```
> > 
> > 
> > Hi @ichergui
> > Please find the execution result of the given command
> > [ 0.0119 ] Generating RCM messages
> > [ 0.0189 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm mb1_recovery_prod.bin 0 0
> > [ 0.0221 ] RCM 0 is saved as rcm_0.rcm
> > [ 0.0347 ] RCM 1 is saved as rcm_1.rcm
> > [ 0.0365 ] List of rcm files are saved in rcm_list.xml
> > [ 0.0472 ]
> > [ 0.0473 ] Signing RCM messages
> > [ 0.0559 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
> > [ 0.0588 ] Assuming zero filled SBK key
> > [ 0.0899 ]
> > [ 0.0901 ] Copying signature to RCM mesages
> > [ 0.0940 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
> > [ 0.1110 ]
> > [ 0.1113 ] Boot Rom communication
> > [ 0.1199 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml
> > [ 0.1257 ] BR_CID: 0x81801001645102820800000013fc0380
> > [ 0.1474 ] RCM version 0X180001
> > [ 0.1576 ] Boot Rom communication completed
> > [ 1.2098 ]
> > [ 2.2149 ] tegrarcm_v2 --isapplet
> > [ 2.2321 ] USB communication failed.Check if device is in recovery
> > [ 3.4237 ]
> > [ 3.4410 ] tegradevflash_v2 --iscpubl
> > [ 3.4610 ] CPU Bootloader is not running on device.
> > [ 3.6700 ]
> > [ 4.6847 ] tegrarcm_v2 --isapplet
> > [ 4.6918 ] Applet version 01.00.0000
> > [ 4.9059 ]
> > [ 4.9183 ] Retrieving EEPROM data
> > [ 4.9186 ] tegrarcm_v2 --oem platformdetails eeprom cvm /home/chandrika/Downloads/cvm.bin
> > [ 4.9211 ] Applet version 01.00.0000
> > [ 5.1133 ] Saved platform info in /home/chandrika/Downloads/cvm.bin
> > [ 5.4257 ]
> > > > > if the execution of command above is Okay, try the next one:
> > > > > ```
> > > > > $ sudo ./chkbdinfo -p cvm.bin
> > > > > $ sudo ./chkbdinfo -g cvm.bin
> > > > > ```
> > > > > 
> > > > > 
> > > > > **Note:** These commands are valid if secure boot is not enabled.
> > > > 
> > > > 
> > > > Unfortunately we cant login to the device as we don't have the password and also default username and password (nvidia and ubuntu) did not work.
> > > > Please let me know if we could reset the password otherwise
> > > 
> > > 
> > > No need to the login and password here. power your device, and force it into the recovery mode and try the command line.
> 
> Well, this is mean that you can read data from EEPROM.
> 
> Please try the following commands and share the result:
> 
> ```
> $ sudo ./chkbdinfo -p cvm.bin
> $ sudo ./chkbdinfo -g cvm.bin
> ```
> 
> Try again to flash you device and let us know.

Find the execution results below for these commands, also I tried flashing it again it fails with same issue as reported:

$ sudo ./chkbdinfo -p cvm.bin
-- 699-83310-1000-D01 B.0 
$ sudo ./chkbdinfo -g cvm.bin
-- 00:04:4B:C5:E9:30

chandshiv, 2020-11-11:
Hello,

I could not succeed flashing the Jetson TX2 from Ubuntu VM.
So tried with Raspberry Pi 3 ( armv7l ) as host with Ubuntu 20.04.1 LTS .
I could see that the Jetson TX2 device going to recovery mode,
[image]

Then I tried flashing the device by ./doflash.sh as per below doc,
 https://hub.mender.io/t/nvidia-tegra-jetson-tx2/123
Its giving the  error as below,
[image]

Please let me know if I am missing any steps here. 

-Thank you

madisongh, 2020-11-11:
The tools, such as `tegrarcm_v2`, that are used during flashing are compiled for x86-64 and are supplied by NVIDIA in binary form only. It might be technically possible to get them to run on your ARM-based RPi, but your really should be using an x86-64 host to do your flashing.

If you don't have an x86-64 system that runs a recent Ubuntu or other typical desktop Linux release and you have to use a VM, I'd suggest searching the NVIDIA Developer Forum for tips on how to make that work. Getting the USB access to work properly can be tricky, and the solutions will vary based on host OS, which VM manager you're using, and possibly other factors.  A Google search on "flash jetson from virtual machine" will turn up several hits on forum posts and other sites that may help.

But as @dwalkes mentioned earlier, it's really best if you can run Linux natively.  And I'll add that it really should be an Intel or AMD x86-64 system. NVIDIA only supports Ubuntu 16.04 or 18.04, but other modern desktop Linux OSes should work OK as well.

dwalkes, 2020-11-11:
> It might be technically possible to get them to run on your ARM-based RPi

We've been able to get this to work with QEMU and were planning to share a project that helps turn an RPI into a tegraflasher, if anyone is interested.  But I do highly suggest having at least one physical native Linux x86-64 system in the mix for your development environment.

Sankalppund, 2020-11-11:
> Hello,
>
> I could not succeed flashing the Jetson TX2 from Ubuntu VM.
> So tried with Raspberry Pi 3 ( armv7l ) as host with Ubuntu 20.04.1 LTS .
> I could see that the Jetson TX2 device going to recovery [...]

Please try the below-mentioned steps to tegraflash the device using the raspberry PI.

**1.** Install the Raspberry Pi OS image using these steps: [Install the Raspberry Pi OS](https://www.raspberrypi.org/documentation/installation/installing-images/)
**2.** You need to install qemu to cross-compile the x86-64 tegraflash binaries on the raspberry PI's ARM platform.
      **a.** Install qemu, binfmt-support, and qemu-user-static using command : `sudo apt install qemu binfmt- 
          support qemu-user-static `
      **b.** Check whether the binfmt entries were successfully registered using command:`sudo update-binfmts -- 
          display` This command should print entries for each supported target user emulator, except for the host 
          system.
       **c.** Depending on your kernel settings, you may need to set 'vm.mmap_min_addr=0' sysctl option to allow a program being run under a regular user, not root using the command `sudo sysctl vm.mmap_min_addr=0 `
**3.** All the above installation makes raspberry pi compatible to run x86 binaries which are used in the Tegraflash. To read more about qemu emulation, please visit [QemuUserEmulation](https://wiki.debian.org/QemuUserEmulation)
**4.** Connect the Nvidia TX2 board to the raspberry pi through USB in recovery mode and you should be able to successfully tegraflash your Nvidia device.

chandshiv, 2020-11-13:
> Please try the below-mentioned steps to tegraflash the device using the raspberry PI.
>
> **1.** Install the Raspberry Pi OS image using these steps: [Install the Raspberry Pi [...]

Thank you for your detailed steps @Sankalppund.
I tried this and now again I am getting the same issue as I got from VM (which I mentioned in this post)
[  33.7938 ] tegrarcm_v2 --isapplet
[  33.7999 ] USB communication failed.Check if device is in recovery
[  33.8425 ]
[  33.8481 ] tegradevflash_v2 --iscpubl
[  33.8537 ] Cannot Open USB
[  33.8972 ]
Error: None of the bootloaders are running on device. Check the UART log.
ERR: could not retrieve EEPROM board information

madisongh, 2020-11-13:
@chandshiv It's possible that this is a cabling/communication problem.  Make sure you're using a good quality cable (the one that comes with the TX2 dev kit is a good one), and that you're connecting the A (host) side of the cable directly into a USB port on your host, rather than through an external hub.  The flashing tools can be sensitive to drops and latency in the USB path, so a directly-connected, high-quality cable can help.

Also, make sure that the TX2 really is in recovery mode by holding the recovery button down, pressing the reset button, then keeping the recovery button held down four 4-5 seconds.  It can be in a state where it shows up in **lsusb** but isn't in the right state to accept the RCM commands from the flashing tools.

If you still have flashing issues after the above, it would help to see what the TX2 is printing on the serial console (if anything).  Do you have that connected?  If not, [this JetsonHacks article](https://www.jetsonhacks.com/2017/03/24/serial-console-nvidia-jetson-tx2/) can help.

dwalkes, 2020-11-14:
@chandshiv you should also try `sudo` with `doflash.sh` if you haven't already.

ichergui, 2021-02-15:
Hey @chandshiv 
Could you please give us updates concerning your issue ? otherwise we can close it.

ichergui, 2021-02-23:
No feedback from @chandshiv, I will close this issue

closed 2021-02-23 by ichergui