OE4T/meta-tegra #482: Partition order and sizes on SD card images for Xavier NX

radek-uavia, 2020-11-06:
Hi,

I am very appreciative of all the work done here. I am using this repository on the `dunfell-l4t-r32.4.3` branch to work with a Xavier NX. It works, but I have a minor issue that I am trying to work around. Any help is greatly appreciated.

Ultimately, it boils down to having an image size that fits to the SD card size. By default, a 16GB image is created, so if a 128GB SD card is used, most of the space will be left unused with no filesystem.

There are two conceptual approaches to bypass this:
1. Set `ROOTFSPART_SIZE` and `TEGRAFLASH_SDCARD_SIZE` to match de SD card. There is also a `num_sectors` value in the `flash.xml` file that should match. This is explored in the comments of #453 although with the goal of making the image smaller.
2. Resize the `APP` partition after the image is flashed to the SD card.

The issue that I have with approach (1) is that creating the SD card with `dd` takes a lot longer if the SD card is very large (I have not tried `bmaptool`).

It seems that NVidia favors approach (2) themselves. For example, they have this script called [nvresizefs.sh](https://github.com/NVIDIA-AI-IOT/jetcard/blob/master/scripts/archive/nvresizefs.sh) that resizes the `APP` partition to take all the space for a Nano (not sure why not for Xavier NX, it explicitly checks against it). The script seems to have been executed automatically at first boot. However, for this to work, the `APP` partition must be the last one, and currently it is configured to be the first one so it cannot be resized. [On this forum post](https://forums.developer.nvidia.com/t/sd-card-partition-after-flask-by-jetpack/124610/3), NVidia's JerryChang recommends the following:
> you may try to put APP partition at end offset, then it can be resized according to the SD card capacity.
please generate the SD card image using create sdcard image script; then write that image into the SD card for using.
such script file makes sure to create APP partition at the end of the disk and latest resizing service will take care of resizing to occupy the unallocated memory of the SD card.

First, I would like to ask if my understanding of the situation is correct. If not, I would be grateful for any clarification.

Then, I wonder what is the cleanest way to change the order of the partitions. Is it to edit the `flash.xml` file manually?

It contains the following description for the `APP` partitiong:
```
            <description> **Required.** Contains the rootfs. This partition must be defined
              after `primary_GPT` so that it can be accessed as the fixed known special device
              `/dev/mmcblk0p1`. </description>
```
Does it mean that it must be just after `primary_GPT`, or that it can be at any point after it, including last?

I notice that there is a `secondary_gpt` at the very end, but it seems unused as far as I can tell. What is its purpose? Would it be intended to host any additional partitions?

It seems like a solution is within my grasp, but I would like to do things properly. If anyone has a workflow that is similar, please let me know how you deal with large SD cards. I am also interested to know how you deal with it with the Nano, which I will hopefully get to use myself pretty soon. NVidia's partition resizing script was specifically for the Nano, so I am unsure why the particular attention and what the differences are.

Thank you!

madisongh, 2020-11-06:
There are two things to consider with flash or SDcard layout - the physical location of the partitions (start and end LBAs), and the indexes of the entries in the partition table (which gives you the number after the `p` in the device name).

With the flash layout XML files (and tools) in L4T R32.4.3, entries are processed in order for both location and table entry purposes.  In meta-tegra, however, we have the `dosdcard.sh` script that gets bundled into the tegraflash package, which will automatically locate the APP partition at the end of the SDcard and have it fill out the remaining available space, if there is no partition with the attribute `0x808` (which indicates fill-to-end). It will still get assigned index 1, since it's the first entry after the primary GPT, but the starting LBA for that entry will be after all of the other partitions.

As for the secondary GPT, it's a backup copy that's there for redundancy should the primary copy get corrupted.  See the [GUID Partition Table page on Wikipedia](https://en.wikipedia.org/wiki/GUID_Partition_Table) for a description.

Now as to the reason why you can't relocate and resize the APP partition with the stock flash layout file for the NX dev kit: the layout that NVIDIA provides includes a UDA partition at the end with the `0x808` attribute, so that's the one that will be filled to the end of the card.  You can provide a customized layout file that eliminates the UDA partition; if you do that and then use the `dosdcard.sh` script to program you SDcard, APP should get relocated to the end of the card and fill out all available space.

Note that if you do change the partition layout, you should also reprogram the bootloader entries in the SPI flash (via USB), just to be sure that the BCT stored there is in sync with the GPT on the SDcard.

> The issue that I have with approach (1) is that creating the SD card with dd takes a lot longer if the SD card is very large (I have not tried bmaptool).

You should definitely use bmaptool if you can - since your SDcard image is going to be mostly empty space, the optimizations that bmaptool uses will considerably speed up the write time.  The `dosdcard.sh` script will automatically use bmaptool if you have it installed on your host system.

(I also see that with L4T R32.4.4, NVIDIA has modified their flashing tools so they can now separate the partition table index assignment from the physical location on the SDcard/flash.  The layout XML file for Xavier NX devkit in 32.4.4 takes advantage of this new feature.)

radek-uavia, 2020-11-06:
I see that there are a few aspects that I missed (especially about the LBA). I have one further question about the UDA partition. The description says that it is required. I gather from your message that it isn't really required since you're suggesting that it can be removed, right? Since its purpose seems to be storage space in the first place, maybe the cleanest solution is to rely on it and mount it rather than resizing APP, and it would achieve a somewhat similar outcome with minor constraints.

Thank you for the comprehensive answer.

madisongh, 2020-11-06:
> I have one further question about the UDA partition. The description says that it is required.

Yeah, I don't know why.  AFAIK, L4T/JetPack doesn't do anything with it.  And if you're doing your own builds with OE/Yocto, you can customize the flash layout however you like, as long as you don't break anything that the bootloaders use.

> maybe the cleanest solution is to rely on it and mount it rather than resizing APP

That's definitely easier, if you just need to be able to store your own files somewhere.

radek-uavia, 2020-11-17:
I've been using the UDA partition just fine since the last time, but I'd still like to try to use a large APP partition.

However, I'm not sure how to actually do this step:
> Note that if you do change the partition layout, you should also reprogram the bootloader entries in the SPI flash (via USB), just to be sure that the BCT stored there is in sync with the GPT on the SDcard.

Things I've tried:
1. Remove the UDA partition from `flash.xml.in`, then use `doflash.sh` with the Xavier NX in recovery mode plugged through USB. After this, there is no UDA partition on the sd card, but APP is still only 14G.
2. Remove the UDA partition from `flash.xml.in`, then use `dosdcard.sh`. Then there is no UDA partition on the sd card, APP is expanded to take all the available space. However, once the Xavier NX boots from the card, `df -h` shows
```
Filesystem                Size      Used Available Use% Mounted on
/dev/mmcblk0p1           13.1G      2.5G      9.8G  20% /
```
(while `lsblk` does show 116G for mmcblk0p1)

So I think neither of these steps is sufficient for updating the BCT? I am probably missing a step, maybe another file to edit besides `flash.xml.in` before running `doflash.sh`?

madisongh, 2020-11-17:
@radek-uavia What you're seeing isn't related to the BCT, but is due to the rootfs filesystem having a fixed maximum size which is set during the build, and is much smaller than the available space in the partition.

The simplest way to deal with this would be to use `resize2fs` on the rootfs partition after you run `dosdcard.sh`.  (You probably need to run `fsck` on it first.)  That will update the filesystem to occupy the full size of the partition.

radek-uavia, 2020-11-24:
Thank you very much for your help. The issue is solved as far as I am concerned, if you want to close it.

closed 2020-11-24 by madisongh