OE4T/meta-tegra #486: dunfell-l4t-r32.4.3: the gstreamer nvvideo4linux2 plugin are failing in some scenarios

quaresmajose, 2020-11-07:
I have found this with the test of the PR #471
I test this with gstreamer **1.16.3** and **1.16.2** using OE core  `dunfell`  and  meta-tegra `dunfell-l4t-r32.4.3` branchs

This pipeline don't work:
```
gst-launch-1.0 videotestsrc ! nvvidconv ! nvv4l2h264enc ! h264parse ! nvv4l2decoder ! fakesink dump=1
```

This one work:
```
gst-launch-1.0 videotestsrc ! nvvidconv ! nvv4l2h264enc ! nvv4l2decoder ! fakesink dump=1
```

I try to find more information about this issue and come back
but apparently the `nvvideo4linux2` is broken on `dunfell-l4t-r32.4.3`

madisongh, 2020-11-07:
Yes, I see a hang with the first pipeline, and no hang with the second.  If I remove `nvv4l2decoder` from the first pipeline, I get no hang, so it may be that element (or one of the binary-blob libraries) that's the problem.  I tried a couple of the patches from the NVIDIA dev forum for gstreamer issues (see [this page](https://elinux.org/Jetson/L4T/r32.4.x_patches) for links), specifically the replacements for `libnvtvmr.so`, since they appeared to be related to encode/decode issues, but they didn't help.

quaresmajose, 2020-11-08:
I already discovered the problem, there's a dead lock on `nvv4l2decoder` and I am trying to address it.

I think this problem affect the Ubuntu as well and the gstreamer 1.14 on the contrib layer.
Can anyone test the pipelines on the issue description on Ubuntu to check the results?

The dead lock is because the `gstv4l2bufferpool` only set  `capture_plane_stopped` if the object that use the poll are `V4L2_TYPE_IS_OUTPUT` and in this pipeline it have an encoder in front of the decoder. Because of this the `gstv4l2bufferpool` don't set the `capture_plane_stopped` and `nvv4l2decoder`  will waiting forever for this.

This only happens because the `h264parse` change the caps when the pipeline goes to running that will trigger a restart of the `nvv4l2decoder` in the `gst_v4l2_video_dec_set_format`

`gstv4l2bufferpool.c` code that don't set `capture_plane_stopped`:
```c
#ifdef USE_V4L2_TARGET_NV
  /* Make sure, there are no outstanding buffers to be unreffed by
   * downstream component. In case of DRC, wait before reseting decoder
   * until last buffer on capture plane is unreffed.
   */
  if (!V4L2_TYPE_IS_OUTPUT(pool->obj->type))
  {
      g_mutex_lock (&pool->obj->cplane_stopped_lock);
      pool->obj->capture_plane_stopped = TRUE;
      g_cond_signal (&pool->obj->cplane_stopped_cond);
      g_mutex_unlock (&pool->obj->cplane_stopped_lock);
  }
#endif
```

This fix the problem for now but a better solution is needed:

```diff
diff --git a/gstv4l2videodec.c b/gstv4l2videodec.c
index 753f8e2..2818eae 100644
--- a/gstv4l2videodec.c
+++ b/gstv4l2videodec.c
@@ -680,26 +680,6 @@ gst_v4l2_video_dec_set_format (GstVideoDecoder * decoder,
 
     gst_v4l2_object_stop (self->v4l2capture);
     self->output_flow = GST_FLOW_OK;
-
-#ifdef USE_V4L2_TARGET_NV
-    g_mutex_lock (&self->v4l2capture->cplane_stopped_lock);
-    while (self->v4l2capture->capture_plane_stopped != TRUE)
-    {
-        g_cond_wait (&self->v4l2capture->cplane_stopped_cond,
-                &self->v4l2capture->cplane_stopped_lock);
-    }
-    self->v4l2capture->capture_plane_stopped = FALSE;
-    g_mutex_unlock (&self->v4l2capture->cplane_stopped_lock);
-    gst_v4l2_object_close (self->v4l2output);
-    gst_v4l2_object_close (self->v4l2capture);
-    gst_v4l2_object_open (self->v4l2output);
-    if (!gst_v4l2_object_open_shared (self->v4l2capture, self->v4l2output)) {
-        g_print ("gstv4l2object open shared failed\n");
-        if (GST_V4L2_IS_OPEN (self->v4l2output))
-            gst_v4l2_object_close (self->v4l2output);
-        return FALSE;
-    }
-#endif
   }
 
   ret = gst_v4l2_object_set_format (self->v4l2output, state->caps, &error);
```

quaresmajose, 2020-11-08:
I will put some more time on this issue to address and fix this deadlock.
Any help that may exist is welcome.

madisongh, 2020-11-08:
> I think this problem affect the Ubuntu as well and the gstreamer 1.14 on the contrib layer.
Can anyone test the pipelines on the issue description on Ubuntu to check the results?

I just fired up my TX2 with the stock L4T R32.4.4 Ubuntu image and got the same results.

quaresmajose, 2020-11-08:
I have fix it !!!
We need to check if the pool is active in the `nvv4l2decoder` before the locking.
I will send a patch with the fix

madisongh, 2020-11-08:
> This fix the problem for now but a better solution is needed:

That code path is a little different in the latest (R32.4.4) version of the nvvideo4linux2 plugin source, but the changes don't affect the result with this test. 

From [this dev forum post](https://forums.developer.nvidia.com/t/extremely-unreliable-hw-encoder/75929) it sounds like their v4l2 codec plugins are still a bit of a work in progress.

quaresmajose, 2020-11-08:
> > I think this problem affect the Ubuntu as well and the gstreamer 1.14 on the contrib layer.
> > Can anyone test the pipelines on the issue description on Ubuntu to check the results?
> 
> I just fired up my TX2 with the stock L4T R32.4.4 Ubuntu image and got the same results.

I suspected that, thanks for confirming.

> > This fix the problem for now but a better solution is needed:
> 
> That code path is a little different in the latest (R32.4.4) version of the nvvideo4linux2 plugin source, but the changes don't affect the result with this test.

I have work only in the R32.4.3 in this issue. I will need to create a repo with `nvvideo4linux2` plugin to track what Nvidia do
I don't understand this type of fork! if they provide a git repo for this it will much easier for the community to integrate it upstream in gst-plugins-good for example if anyone can work on this.

> 
> From [this dev forum post](https://forums.developer.nvidia.com/t/extremely-unreliable-hw-encoder/75929) it sounds like their v4l2 codec plugins are still a bit of a work in progress (although, to be fair, that post is about a year old).

I think so as well, there are many workarounds on gst v4l2 plugins and it as around of several month, it is announced to replace the omx plugin but I don't know when this will happen.

madisongh, 2020-11-08:
> I don't understand this type of fork! 

Yeah, I think the problem here is that the `nvbuf` APIs they're using are proprietary.  Unless NVIDIA switches over to an open-source license for those, we're stuck with these out-of-tree plugins.

closed 2020-11-08 by madisongh