OE4T/meta-tegra #490: root of /dev/mmcblk0p1 in gatesgarth is read only?

arrow53, 2020-11-11:
I'm using `tegra-demo-distro` and it seems in gatesgarth that the root filesystem is read only. I need to write some files to `/boot`

Is this a meta-tegra thing or something that NVIDIA has done with JetPack 4.4.1 in the devicetree?

Sorry, I'm clueless but I have no idea where to start to make it read/write. Can someone give me a hint?

dwalkes, 2020-11-11:
Are you using `tegrademo-mender`?  If so it's read only, just search for `read-only` and you'll see this commit: https://github.com/OE4T/tegra-demo-distro/commit/ed540aa823de59976ef3ed8848123ebb7f0fde76

> Sorry, I'm clueless but I have no idea where to start to make it read/write.

`mount -oremount,rw /` should do it

arrow53, 2020-11-11:
yep, using tegrademo-mender but moving from dunfell to gatesgarth. This explanation/solution is perfect. thanks!

closed 2020-11-11 by arrow53