OE4T/meta-tegra #492: [jetson-nano-b01] `eeprom_init: EEPROM read failed` during Nano booting

przemyslaw-chwiala, 2020-11-12:
Hey, I recently changed my A02 Nano to B01 with SD card and tried to run yocto on it (same that works great on A02).
I noticed that booting procedure takes a way longer that used to.

I connected UART to see full logs from booting and this is what I saw.

```
[0049.274] I2C write failed
[0049.277] Writing offset failed
[0049.280] eeprom_init: EEPROM read failed
[0049.284] pm_ids_update: eeprom init failed
[0049.288] pm_ids_update: Updating 3,a0, size 256, type 0
[0057.284] pm_ids_update: The pm board id is 3448-0000-400
[0057.291] Adding plugin-manager/ids/3448-0000-400=/i2c@7000c500:module@0x50
[0057.299] pm_ids_update: pm id update successful
[0057.303] pm_ids_update: Updating 3,ae, size 256, type 0
[0057.309] I2C slave not started
[0057.312] I2C write failed
[0057.314] Writing offset failed
[0057.317] eeprom_init: EEPROM read failed
[0057.321] pm_ids_update: eeprom init failed
```
It is only the part of this log. This `eeprom_init: EEPROM read failed` occurs 6 times, every 8 seconds and then system start properly.

ichergui, 2020-11-12:
Hi @przemyslaw-chwiala 

I have the Jetson Nano B01 (SDcard), I'm using this repo [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) with `dunfell-l4t-r32.4.3` branch. It works fine for me.

I had had the same message as you in the Cboot stage, It happen three time but this is not a blocker for me. the device continue to boot successfully.

Could you please provide more detail on what you have done (build, flash, ...) ? Are you sure that you are using the Nano B01 ?

attached the complete logs from the serial UART.
[jetson_nano_sdcard_serial_uart.log](https://github.com/OE4T/meta-tegra/files/5530054/jetson_nano_sdcard_serial_uart.log)

przemyslaw-chwiala, 2020-11-12:
Hi @ichergui, 
thank you for your answer.
I'm sure I'm using B01, same soft on A02 works fine. I'm attaching full log from my booting.
I'm working on zeus branch. Could update to dunfell help? There are any additional steps required to switch from A02 to B01?

[nano_emmc_error.log](https://github.com/OE4T/meta-tegra/files/5530283/nano_emmc_error.log)

ichergui, 2020-11-12:
Could you please try with same the repo and the branch that I shared ?

Here is the steps: 
```
$ git clone https://github.com/OE4T/tegra-demo-distro.git
$ git checkout origin/dunfell-l4t-r32.4.3 -b dunfell-l4t-r32.4.3
$ git submodule update --init
$ unset NVIDIA_DEVNET_MIRROR
$ export NVIDIA_DEVNET_MIRROR="/path/to/jetpack_4.4/nvidia/sdkm_downloads"
$ . ./setup-env --machine jetson-nano-qspi-sd --distro tegrademo-mender build-distro
$ bitbake demo-image-base
```
**Note:** Please take a look [here](https://github.com/OE4T/meta-tegra/wiki/Tegraflash-enhancements) for SDcard flashing

Please let me know if you need help.

madisongh, 2020-11-12:
@przemyslaw-chwiala From the log you attached, it looks like you're using a branch based off L4T R32.2.2, is that right? If so, try @ichergui's suggestion, or at least try a build using a branch based off L4T R32.2.3 or later.  IIRC, Nano rev B01 support was introduced in 32.2.3.

ichergui, 2020-12-09:
Hi @przemyslaw-chwiala 

Did you try with the release `L4T R32.4.3` ?
Could you please give us updates about your issue ?

ichergui, 2021-02-15:
I assume you resolved this issue, since it's been a few months. Closing.

closed 2021-02-15 by madisongh