OE4T/meta-tegra #497: Xavier NX: Flashing to eMMC stuck

om26er, 2020-11-19:
We had our custom yocto distro, today I added support for Xavier NX, I was able to produce the image successfully, however I am not able to flash the image to the eMMC, it "hangs". Attached are the logs.
[tegra-flash.log](https://github.com/OE4T/meta-tegra/files/5569260/tegra-flash.log)

madisongh, 2020-11-19:
A couple of questions:
* Is the device really a Xaver NX production (eMMC) module installed in a Xavier NX dev kit or Nano carrier? If it's actually a standard dev kit, then there is no eMMC, and you should use the `jetson-xavier-nx-devkit` machine configuration.
* Do you have a serial console cable connected?  Seeing what the device is doing during the flashing process could help with troubleshooting.

om26er, 2020-11-19:
You are right, I had to use the jetson-xavier-nx-devkit for the flashing to "proceed" a bit. But now I am faced with another issue. Attached the logs
[tegra2.log](https://github.com/OE4T/meta-tegra/files/5569460/tegra2.log)

"End sector for RECROOTFS, expected at: 30253022, actual: 30369575"

om26er, 2020-11-19:
The above was an issue with my sdcard, I tried another one and it seems to be flashing fine now.

Is there a way to flash the image on eMMC instead ?

madisongh, 2020-11-19:
The NX module with on-board eMMC is a different part.  The module in the dev kit only has the SDcard slot.

closed 2020-11-19 by om26er