OE4T/meta-tegra #499: Redundant Rootfs on Xavier NX

om26er, 2020-11-22:
On a Xavier NX, we want to have redundant Rootfs and a separate data partitioning. Are there some parameters that we could set in our config ?

ichergui, 2020-11-22:
Hi @om26er 
Could you please give us more detail on what we want to do ?

om26er, 2020-11-22:
> Could you please give us more detail on what we want to do ?

Hey, we are using RAUC (https://rauc.io/) as our update mechanism (mender alternative), things work fine on the RaspberryPi since we were able to get away with using wic. However for tegra, currently with the default partition scheme, it seems to create a single rootfs parition (expected). We are looking to change the schema to have two rootfs (for redundancy) and a separate data partition.

The we'll adapt RAUC to support CBoot mostly using `nvbootctrl` commands.

ichergui, 2020-11-22:
All right, to be honest I didn't use RAUC yet. I'm using mender. 
You can update the XML file `the partitions layout` for tegra, you could base how we are doing that via mender.
Let's see with @madisongh or @dwalkes for a better advise.

dwalkes, 2020-11-22:
> We are looking to change the schema to have two rootfs (for redundancy) and a separate data partition.

This is exactly what the mender community layer does, so you should be able to leverage this.  See https://github.com/mendersoftware/meta-mender-community/tree/dunfell/meta-mender-tegra/recipes-bsp/tegra-binaries/mender-custom-flash-layout for instance where the flash files are customized to include extra partitions for data and application B partition.

Unfortunately there's no easy way to do this today other than defining your own flash layout files, and even then it's a pain to keep up with [incompatible layout changes across L4T upgrades](https://forums.developer.nvidia.com/t/mmcblk0boot1-usage-at-address-4177408-and-u-boot-parameter-storage-space-availability/82113/13?u=danwalkes1).

lfdmn, 2020-11-22:
Hi @om26er 

First you need to modify the partition layout with your custom one. You need to add another partition with name ```APP_b``` and another one for the data (the name doesn't matter)

Then you might need to modify the [init script](https://github.com/OE4T/meta-tegra/blob/master/recipes-core/initrdscripts/tegra-minimal-init/init-boot.sh) to mount the correct APP partition (```APP``` for slot 0 and ```APP_b``` for slot1).

[platform-preboot](https://github.com/OE4T/meta-tegra/blob/master/recipes-core/initrdscripts/tegra-minimal-init/tegra194/platform-preboot.sh) used with Xavier family (line 11) should already be detecting the correct ```rootdev``` to mount from, as long as you comply with the ```APP_b``` naming.

Also, I recall some message from @madisongh stating that only the eMMC version Xavier NX support the dual slot config.

dwalkes, 2020-11-22:
> Also, I recall some message from @madisongh stating that only the eMMC version Xavier NX support the dual slot config.

A fix for this went in at https://github.com/OE4T/meta-tegra/pull/494

om26er, 2020-11-22:
I used the partition layout that was created for mender, the image builds fine, however while flash it throws
`End sector for APP_b, expected at: 33554398, actual: 59729703`. I have attached the complete log below.

```
om26er@HomePC:~/.yocto/cfx-hub/tmp-glibc/deploy/images/jetson-xavier-nx-devkit/output$ ./dosdcard.sh 
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0048 ] tegrasign_v2 --key None --getmode mode.txt
[   0.0058 ] Assuming zero filled SBK key
[   0.0060 ] 
[   0.0063 ] Generating RCM messages
[   0.0072 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
[   0.0081 ] Header already present for mb1_t194_prod.bin
[   0.0110 ] 
[   0.0120 ] tegrasign_v2 --key None --getmode mode.txt
[   0.0127 ] Assuming zero filled SBK key
[   0.0128 ] 
[   0.0138 ] tegrasign_v2 --key None --file mb1_t194_prod_sigheader.bin --offset 2960 --length 1136 --pubkeyhash pub_key.key
[   0.0146 ] Assuming zero filled SBK key
[   0.0151 ] 
[   0.0159 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mb1_t194_prod_sigheader.bin mb1_t194_prod_sigheader.hash zerosbk
[   0.0197 ] 
[   0.0207 ] tegrabct_v2 --chip 0x19 0 --sfuse tegra194-mb1-soft-fuses-l4t.cfg sfuse.bin
[   0.0217 ] 
[   0.0226 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x19 0 --sfuses sfuse.bin --download rcm mb1_t194_prod_sigheader.bin 0 0
[   0.0234 ] RCM 0 is saved as rcm_0.rcm
[   0.0265 ] RCM 1 is saved as rcm_1.rcm
[   0.0266 ] RCM 2 is saved as rcm_2.rcm
[   0.0266 ] List of rcm files are saved in rcm_list.xml
[   0.0266 ] 
[   0.0266 ] Signing RCM messages
[   0.0275 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key --getmontgomeryvalues montgomery.bin
[   0.0284 ] Assuming zero filled SBK key
[   0.0291 ] 
[   0.0291 ] Copying signature to RCM mesages
[   0.0300 ] tegrarcm_v2 --chip 0x19 0 --updatesig rcm_list_signed.xml
[   0.0314 ] 
[   0.0314 ] Parsing partition layout
[   0.0323 ] tegraparser_v2 --pt flash.xml.tmp
[   0.0340 ] 
[   0.0340 ] Creating list of images to be signed
[   0.0349 ] tegrahost_v2 --chip 0x19 0 --partitionlayout flash.xml.bin --list images_list.xml zerosbk
[   0.0356 ] MB1: Nvheader already present is mb1_t194_prod.bin
[   0.0361 ] Header already present for mb1_t194_prod_sigheader.bin
[   0.0366 ] MB1: Nvheader already present is mb1_t194_prod.bin
[   0.0398 ] Header already present for mb1_t194_prod_sigheader.bin
[   0.0402 ] adding BCH for spe_t194.bin
[   0.0433 ] adding BCH for spe_t194.bin
[   0.0450 ] adding BCH for nvtboot_t194.bin
[   0.0461 ] adding BCH for nvtboot_t194.bin
[   0.0479 ] Header already present for preboot_c10_prod_cr.bin
[   0.0497 ] Header already present for preboot_c10_prod_cr.bin
[   0.0500 ] Header already present for mce_c10_prod_cr.bin
[   0.0502 ] Header already present for mce_c10_prod_cr.bin
[   0.0508 ] adding BCH for mts_c10_prod_cr.bin
[   0.0529 ] adding BCH for mts_c10_prod_cr.bin
[   0.0911 ] Header already present for warmboot_t194_prod.bin
[   0.1287 ] Header already present for warmboot_t194_prod.bin
[   0.1291 ] adding BCH for cboot_t194.bin
[   0.1295 ] adding BCH for cboot_t194.bin
[   0.1342 ] adding BCH for tegra194-p3668-all-p3509-0000.dtb
[   0.1389 ] adding BCH for tegra194-p3668-all-p3509-0000.dtb
[   0.1411 ] adding BCH for tos-trusty_t194.img
[   0.1432 ] adding BCH for tos-trusty_t194.img
[   0.1472 ] adding BCH for eks.img
[   0.1511 ] adding BCH for eks.img
[   0.1512 ] adding BCH for adsp-fw.bin
[... 194 lines cut ...]
[   1.3993 ] Packing sdram param for instance[11]
[   1.3993 ] Packing sdram param for instance[12]
[   1.3993 ] Packing sdram param for instance[13]
[   1.3993 ] Packing sdram param for instance[14]
[   1.3993 ] Packing sdram param for instance[15]
[   1.3993 ] 
[   1.4004 ] tegrahost_v2 --chip 0x19 --align mem_rcm.bct
[   1.4011 ] 
[   1.4020 ] tegrahost_v2 --chip 0x19 0 --magicid MEMB --appendsigheader mem_rcm.bct zerosbk
[   1.4027 ] adding BCH for mem_rcm.bct
[   1.4033 ] 
[   1.4044 ] tegrasign_v2 --key None --list mem_rcm_sigheader.bct_list.xml --pubkeyhash pub_key.key
[   1.4052 ] Assuming zero filled SBK key
[   1.4055 ] 
[   1.4066 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mem_rcm_sigheader.bct.encrypt mem_rcm_sigheader.bct.hash zerosbk
[   1.4085 ] 
[   1.4086 ] Copying signatures
[   1.4094 ] tegrahost_v2 --chip 0x19 0 --partitionlayout flash.xml.bin --updatesig images_list_signed.xml
[   2.0581 ] 
[   2.0593 ] tegraparser_v2 --generategpt --pt flash.xml.bin
[   2.0602 ] End sector for APP_b, expected at: 33554398, actual: 59729703
[   2.0605 ] 
Error: Return value 4
Command tegraparser_v2 --generategpt --pt flash.xml.bin
```

madisongh, 2020-11-22:
@om26er See [this wiki page](https://github.com/OE4T/tegra-demo-distro/wiki/Customization-of-Storage-Size-on-Mender-Builds).  Although it specifically mentions Mender, it would apply to your case, too.

om26er, 2020-11-22:
@madisongh Thanks, a lot, after quite a bit of effort, I was able to get this thing to work. However I need to mention that with A/B rootfs partiton map, the `ROOTFSPART_SIZE` needs to be half (a bit less even) for the image to flash. In my case, I went with two 3GB partitions.

The issue that I now face is that while there are two rootfs partitions, `nvbootctrl` doesn't recognize that. How do I "fix" that ?

```
root@jetson-xavier-nx-devkit:~# partx /dev/mmcblk0
NR    START      END  SECTORS SIZE NAME              UUID
 1       40  6291495  6291456   3G APP               497ae9ba-d88b-4968-8fd8-015664267756
 2  6291496 12582951  6291456   3G APP_b             5ea3be53-3d4d-4ce2-9689-78785fc2f16b
 3 12582952 12714023   131072  64M kernel            6a2bc88c-02b9-4e4e-866e-691a6e1fb63b
 4 12714024 12845095   131072  64M kernel_b          69b750e9-b456-43f1-86b6-5074a5e19103
 5 12845096 12845991      896 448K kernel-dtb        1b59868f-26dc-4ffb-9403-0120d98a002f
 6 12845992 12846887      896 448K kernel-dtb_b      3a2aca55-16d1-4c7e-b99e-787e99c4c811
 7 12846888 12975911   129024  63M recovery          23cbc564-8f0e-4b63-937d-aa2b41f62972
 8 12975912 12976935     1024 512K recovery-dtb      2ce61b7c-90f7-4653-a8f6-9d5337056176
 9 12976936 12977447      512 256K kernel-bootctrl   5fbc6982-ceb7-499f-9b2b-d84cd527603e
10 12977448 12977959      512 256K kernel-bootctrl_b 66820c04-b4b1-4550-b4ea-512a91d4ad50
11 12977960 13592359   614400 300M RECROOTFS         339eb76b-58fa-44bb-bff3-630c5408561d
12 13592360 30244830 16652471   8G UDA               18ad0d50-aac6-40b4-bae9-e720df930634
root@jetson-xavier-nx-devkit:~# nvbootctrl get-number-slots
1
```

dwalkes, 2020-11-22:
@om26er 
> The issue that I now face is that while there are two rootfs partitions, nvbootctrl doesn't recognize that. How do I "fix" that ?

See the note from @lfdmn at https://github.com/OE4T/meta-tegra/issues/499#issuecomment-731761294

> Then you might need to modify the [init script](https://github.com/OE4T/meta-tegra/blob/master/recipes-core/initrdscripts/tegra-minimal-init/init-boot.sh) to mount the correct APP partition (APP for slot 0 and APP_b for slot1).

om26er, 2020-11-22:
Ok, I had to run `nv_update_engine --enable-ab`

dwalkes, 2020-11-22:
> Ok, I had to run nv_update_engine --enable-ab

Yes, now I see.  You should be able to use the script at https://github.com/mendersoftware/meta-mender-community/blob/dunfell/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-install-script as a reference.

om26er, 2020-11-23:
@dwalkes Thanks for the link. Looking at the script it seems that we'd have to flash the rootfs update using Nvidia's tool or is that script only flashing the bootloader ? Does Nvidia "block" updating the rootfs using external tools ?

om26er, 2020-11-23:
I am going to close this issue now. We now have a redundant filesystem scheme and got our first OTA update working. Thanks everyone.

closed 2020-11-23 by om26er

dwalkes, 2020-11-24:
> Looking at the script it seems that we'd have to flash the rootfs update using Nvidia's tool or is that script only flashing the bootloader

nv_update_engine only flashes the bootloader BUP ("Bootloader Update Payload"), see  https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/bootloader_update_agx_tx2.html for more details.

windelbouwman, 2021-02-22:
@om26er I'm curious to what you had to do, are the steps to setup RAUC with meta-tegra descibed somewhere?

om26er, 2021-02-22:
@windelbouwman We were able to get RAUC working on the Xavier NX. There are two main things that we did.

* Defined a partitions scheme in OE4T that allowed for redundant rootfs
* Patched RAUC to support CBoot https://github.com/rauc/rauc/pull/685

Our current tooling is unfortunately closed-source for now.

windelbouwman, 2021-02-22:
Thanks for the hints, I'll give it a try. Where would these patches be stored ideally? Is this patchwork in this layer, or in the rauc update tool?

Where did you apply the modification to the partition table? Using a bbappend file?