OE4T/meta-tegra #502: Where is extlinux.conf on Xavier NX

om26er, 2020-11-24:
I am trying to add/update a kernel cmdline option and the Nvidia guys suggested me to edit `/boot/extlinux/extlinux.conf`, however it seems on the image I created, the /boot directory is empty. How do I add custom kernel cmdline options on the NX ?

madisongh, 2020-11-24:
We don't generate `extlinux.conf` files for the Xaviers, for various reasons (mainly buggy cboot implementation of ext4 and the extlinux.conf parser). You can set/append to the `KERNEL_ARGS` variable in your `local.conf` to change the kernel command line at build time.

om26er, 2020-11-24:
> We don't generate `extlinux.conf` files for the Xaviers, for various reasons (mainly buggy cboot implementation of ext4 and the extlinux.conf parser). You can set/append to the `KERNEL_ARGS` variable in your `local.conf` to change the kernel command line at build time.

Being able to set kernel args solves one part of the problem. The other part is being able to set/update kernel parameter from userspace because we use RAUC for OTA updates and it reads either `rauc.slot=<name>` from /proc/cmdline` or `root=<device_location>` from it. Since the default boot parameters don't contain a `root=` entry, we want to be able to update `rauc.slot=<name>` from the userspace.

madisongh, 2020-11-24:
cboot automatically includes `boot.slot_suffix=` (null value for slot 0, `_b` for slot 1) in the kernel command line.  Could you customize RAUC to use that instead?

om26er, 2020-11-24:
> cboot automatically includes `boot.slot_suffix=` (null value for slot 0, `_b` for slot 1) in the kernel command line. Could you customize RAUC to use that instead?

Right, we were kind of looking to avoid patching RAUC, but in the current scenario with no easy way to update cmdline from userspace, I guess we'd go with patching RAUC indeed.

arrow53, 2020-11-24:
@om26er fwiw, I have an overlay that puts a file in `/boot/extlinux/extlinux.conf`. As @madisongh mentioned cboot's handling of ext4 is pretty primitive so if you go this route in your `do_install` you'll want something like the below so cboot can read it ok
```
dd if=/dev/zero of=./extlinux.conf bs=1 count=1 seek=4095 status=none
```

om26er, 2020-11-27:
I guess we can close it now.

closed 2020-11-27 by om26er