OE4T/meta-tegra #506: Reading imx219 camera images in a Docker container with nvarguscamerasrc

roger652, 2020-11-27:
First of all thanks for the great work on the meta-tegra layer!

I'm facing some issues when I try to read camera images inside a Docker container with the GStreamer plugin `nvarguscamerasrc`. 
This GStreamer pipeline works fine on the host without Docker:
```
$ gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=(int)3264, height=(int)2464, format=(string)NV12, framerate=(fraction)21/1' ! nvvidconv ! 'video/x-raw, width=(int)3264, height=(int)2464,format=(string)BGRx' ! videoconvert ! 'video/x-raw, format=(string)BGR' ! jpegenc ! multifilesink location="frame%d.jpg"
```

Inside the Docker container I get:
```
nvbuf_utils: Could not get EGL display connection
WARNING: erroneous pipeline: no element "nvarguscamerasrc"
```
The same issue is also mentioned in https://github.com/OE4T/meta-tegra/issues/230 without a solution.

My Docker command:
```
$ docker run --net=host --runtime nvidia --rm --ipc=host -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --device=/dev/video0:/dev/video0 --cap-add SYS_PTRACE -e DISPLAY=$DISPLAY -it nvcr.io/nvidia/l4t-base:r32.4.3
```

The `nvarguscamerasrc` plugin is only blacklisted in the Docker container:
```
$ gst-inspect-1.0 -b
  libgstnvarguscamerasrc.so
  libgstnveglglessink.so
  libgstnvjpeg.so
  libgstnvv4l2camerasrc.so
  libgstnvvideosinks.so
  libgstomx.so
  libgstnvvideo4linux2.so
```

I'm using the `dunfell-l4t-r32.4.3` branch and a imx219 camera.

Any advice on how to make the camera work inside the Docker container?

madisongh, 2020-11-27:
Bumping up the gstreamer debug logging level should tell you more about why those plugins are blacklisted:
```
$ rm -rf ~/.cache/gstreamer-1.0
$ GST_DEBUG=4 GST_DEBUG_NO_COLOR=1 gst-inspect-1.0 -b
```
I just fired up your sample pipeline in a `demo-image-full` image from our demo distro, and it worked fine, so there's likely something missing in your image.  Could be just a missing package, or it could be a gstreamer version mismatch - binaries in the NVIDIA containers are linked against gstreamer 1.14, so you need to be building that version into your image (recipes are available in [meta-tegra/contrib](https://github.com/OE4T/meta-tegra/tree/master/contrib)).

But see if the logs help in determining the root cause first, if you can.

roger652, 2020-11-27:
Thanks for the fast reply. You are right it seems to be a GStreamer version conflict:
```
0:00:00.859241826    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideo4linux2.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.860748075    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglstreamsrc.so" loaded
0:00:00.864015373    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstomx.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.865893874    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideosinks.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.867670079    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvidconv.so" loaded
0:00:00.870074796    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvv4l2camerasrc.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.871058736    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvtee.so" loaded
0:00:00.874823706    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvjpeg.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.876748407    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvegltransform.so" loaded
0:00:00.878892594    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglglessink.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.883623691    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideocuda.so" loaded
0:00:00.899272032    26   0x55882dc800 WARN      GST_PLUGIN_LOADING gstplugin.c:486:gst_plugin_register_func: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvarguscamerasrc.so" has incompatible version (plugin: 1.16, gst: 1,14), not loading
0:00:00.901513826    26   0x55882dc800 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvivafilter.so" loaded
```
How do I tell the Yocto build to use GStreamer 1.14 plugins instead of 1.16?

madisongh, 2020-11-27:
See the [README.md](https://github.com/OE4T/meta-tegra/blob/master/contrib/README.md) for the contrib layer.

roger652, 2020-11-30:
Retested it with gstreamer 1.14 and it works fine.

closed 2020-11-30 by roger652