OE4T/meta-tegra #507: EGL seems to be broken

om26er, 2020-11-27:
I have been trying to get run Kivy/SDL2 on the Xavier NX for a few days. It seems that anything that tries to use libdrm segfaults.

trying to run `video_dec_cuda` gives
```
root@jetson-xavier-nx-devkit:/opt/tegra-mmapi/bin# ./video_dec_cuda ../data/Video/sample_outdoor_car_1080p_10fps.h264 H264
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Failed to query video capabilities: Inappropriate ioctl for device
Starting decoder capture loop thread
Video Resolution: 1920x1080
[ERROR] (NvEglRenderer.cpp:98) <renderer0> Error in opening display
[ERROR] (NvEglRenderer.cpp:154) <renderer0> Got ERROR closing display
Segmentation fault
```
Kivy/SDL2 also results in a Segmentation fault.
```
root@jetson-xavier-nx-devkit:~# python3 -u main.py 
[INFO   ] [Logger      ] Record log in /root/.kivy/logs/kivy_20-11-27_7.txt
[INFO   ] [Kivy        ] v2.0.0rc4, git-d74461b, 20201015
[INFO   ] [Kivy        ] Installed at "/usr/lib/python3.8/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.8.2 (default, Nov 27 2020, 19:30:25) 
[GCC 8.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Factory     ] 186 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2 (img_pil, img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: sdl2(['text_pango'] ignored)
[INFO   ] [Window      ] Provider: sdl2
Segmentation fault
```

om26er, 2020-11-27:
I had asked on the Nvidia forum about `kmscube` not starting and was asked to verify if `video_dec_drm` works on my system. It seems `video_dec_drm` doesn't work either. Starting it prints `Using VT number 2` but after that the terminal basically hangs. The system keeps working since I can ssh to it fine.

om26er, 2020-11-27:
Do note: that demos provided by `l4t-graphics-demos` continue to work. So the issue is quite likely at the libdrm level.

madisongh, 2020-12-01:
I'm no expert on the graphics stack, but from what I've seen, the DRM/KMS support on the Jetsons is pretty limited (i.e., broken for a lot of use cases).  Code that hasn't been specifically adapted to those limitations, and assumes the usual setup of stock libdrm with mesa for OpenGL/GLES/EGL + DRI support is unlikely to work.

As for the video_dec_drm example not working - I gave it a quick try myself and couldn't get it work, either.  I have another sample program that I use for testing KMS/DRM [here](https://github.com/madisongh/eglstreams-kms-example), and it works fine, though.  Firing up the stock L4T/JetPack and testing the video_dec_drm sample program there would be the best next step in figuring out whether this is an issue with the sample code or not.

kekiefer, 2020-12-01:
Try changing `DRM_DEVICE_NAME` to `tegra-udrm`. You can't load the default driver `drm-nvdc` when this module is loaded (which is the default in meta-tegra). Alternatively, you can unload the tegra-udrm kernel module. If it says that it's in use, most likely you'll have to `systemctl stop nvargus-daemon`.

madisongh, 2020-12-01:
Thanks for the tip, @kekiefer .  I got the sample program to kind-of work after unloading the tegra_udrm driver.  It actually worked a  better with tegra_udrm, once I patched NvDrmRenderer.cpp to open the `/dev/dri/card0` device and fixed it to `mmap()` the DRM dumb buffer.

om26er, 2021-03-11:
I guess we can close the issue now. The issue was likely in my environment. Anyways, we have now settled with Wayland/KioskShell.

closed 2021-03-11 by om26er