OE4T/meta-tegra #509: problem with missing mender file in do_image_complete_append

arrow53, 2020-11-30:
I have versioning information in my own layer. After my .mender image is generated I'd like to copy it to another file that has a name with my version information in it (I could also add this to the .mender file itself would be another option).

When I do this I see almost all the files but the .mender files are missing. Is there some sort of post post command world where the .mender files would exist?

```
do_image_complete_append() {
import os

mylist = os.listdir('/yocto/tegra-demo-distro/build/tmp/deploy/images/jetson-xavier-nx-devkit/')
for f in mylist:
  print(f)
}
```

In the log I see the .mender files getting removed `log.do_image_complete:DEBUG: Removing manifest: /yocto/tegra-demo-distro/build/tmp/deploy/images/jetson-xavier-nx-devkit/core-image-tern-jetson-xavier-nx-devkit-20201130222938.mender`

dwalkes, 2020-11-30:
I've done a similar thing with a custom image type in the past, see https://groups.google.com/a/lists.mender.io/g/mender/c/WjB8QetQlQ0/m/XAIVQweQBgAJ for details.

arrow53, 2020-12-01:
would not have thought of that approach. I had to change `IMAGE_TYPEDEP_${MENDER_ARTIFACT_EXTENSION}_append` to `IMAGE_TYPEDEP_${MENDER_ARTIFACT_EXTENSION}` so not sure about that.

Thanks a ton @dwalkes

arrow53, 2020-12-01:
@dwalkes so I would have thought IMAGE_TYPEDEF would have handled this but creating my own image type seemed to grab the old .mender file. I went with just editing the `mender-artifcatimg` class. I like the image idea better but somehow I'd need it to be done *after* the symlinks are in place. Closing but if you have any ideas on that I'd be open to trying.

closed 2020-12-01 by arrow53