OE4T/meta-tegra #510: jetson-nano-emmc dunfell failed to read env from userspace

nandra, 2020-12-01:
I just flashed emmv nano and I tried fw_printenv and I'm getting:
```
fw_printenv
Cannot read environment, using default
Cannot read default environment from file
```
```cat /etc/fw-env.config``` shows:
```# Jetson-TX1 dev kit puts U-Boot params in eMMC boot1 partition
#/dev/mmcblk0boot1      0x3d8000        0x8000
```

I checked u-boot source and offset if correct. Any ideas? Thanks.

madisongh, 2020-12-01:
Which branch are you on?  Recent OE-Core releases replaced the old `u-boot-fw-tools` package with `libubootenv`, and that package requires the `u-boot-default-env` package to be installed (which should be generated by the u-boot recipe) in the rootfs so it can create the default environment.  (The old tools, which were built as part of U-Boot, used to compile in the default environment settings.)

nandra, 2020-12-01:
@madisongh ok thanks a lot I'll give a try.

nandra, 2020-12-02:
Hmm looks like on dunfell (```origin/dunfell-l4t-r32.4.3```) 

I'm getting ```Nothing RPROVIDES 'u-boot-fw-utils-tegra'```

madisongh, 2020-12-02:
Yes, with L4T R32.4.3, we switched to upstream U-Boot, so `libubootenv-bin` would be the package with the tools and `u-boot-default-env` should provide the default environment.  It's only an RRECOMMENDS dependency, though, so you may need to explicitly add that to your image (e.g., if you're building core-image-minimal, which does not include recommended packages).

nandra, 2020-12-02:
Hmm but strange is that fw_print/setenv was present on board already.

madisongh, 2020-12-02:
In that case, something in your image is probably pulling in `libubootenv-bin`.

nandra, 2020-12-02:
OK that is fine, but if I have fw_* tools and /etc/fw-env.config why I cannot read anything from env? Do I need ```u-boot-default-env``` package also? Thanks.

madisongh, 2020-12-02:
Yes.  If U-Boot hasn't done a `saveenv` to write it's compiled-in environment to storage, the fw_* tools look for the file `/etc/u-boot-initial-env`, which is provided by `u-boot-default-env`.

nandra, 2020-12-02:
OK I added ```u-boot-default-env``` package and then I can see files on filesystem:
```
ls -la /etc/u-boot*
lrwxrwxrwx    1 root     root            72 Mar  9  2018 /etc/u-boot-tegra-initial-env -> u-boot-tegra-initial-env-jets
on-nano-emmc-2020.04+gAUTOINC+5e1f83dc98-r0
lrwxrwxrwx    1 root     root            72 Mar  9  2018 /etc/u-boot-tegra-initial-env-jetson-nano-emmc -> u-boot-tegra
-initial-env-jetson-nano-emmc-2020.04+gAUTOINC+5e1f83dc98-r0
-rw-r--r--    1 root     root          4793 Mar  9  2018 /etc/u-boot-tegra-initial-env-jetson-nano-emmc-2020.04+gAUTOIN
C+5e1f83dc98-r0
```

but still getting original error. In recipes I didn't found any which modity libubootenv to look for ```u-boot-tegra-initial-env```

madisongh, 2020-12-02:
Hmm, right.  I think maybe we're either supposed to add
```
UBOOT_INITIAL_ENV = "u-boot-initial-env"
```
in the recipe, so it gets named in the way that libubootenv expects, or install a symlink `/etc/u-boot-initial-env` to point to the actual file.

madisongh, 2020-12-02:
It looks like you can explicitly specify `-f /etc/u-boot-tegra-initial-env` on the `fw_printenv` command, too.

Looking at the OE-Core change that introduced the UBOOT_INITIAL_ENV setting, I think the right solution to make ours the default would be to set that variable in our u-boot-tegra recipe as I mentioned above.

madisongh, 2020-12-02:
OK, I've merged the change I mentioned and have back-ported it to gatesgarth and dunfell-l4-r32.4.3.

closed 2020-12-02 by madisongh

nandra, 2020-12-02:
@madisongh thanks a lot for quick fix. I'll verify tomorrow.

PR #511 (2020-12-02): u-boot-tegra: set UBOOT_INITIAL_ENV
to 'u-boot-initial-env' so that the libubootenv tools can
find the file automatically.

Fixes #510