OE4T/meta-tegra #512: The postinstall intercept hook 'update_gtk_icon_cache' failed

bekirbostanci, 2020-12-08:
Hi
I tried to create a standard jetson nano image file using the link below. But I got the mistake in your lover. I am very new to Yocto so I don't have much experience.
https://developer.ridgerun.com/wiki/index.php?title=Yocto_Support_for_NVIDIA_Jetson_Platforms-Old

console 
```
ieu@ieu-ABRA-A5-V15-8:~/yocto-nano/poky-dunfell/build$ bitbake core-image-minimal
WARNING: Host distribution "ubuntu-20.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 1462 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano-qspi-sd"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-tegra           = "dunfell-l4t-r32.4.3:5e1fd3ae2244aa6b61819e462e314a1627c9a5de"
meta                 
meta-poky            
meta-yocto-bsp       = "my-yocto-3.1:5d47cdf448b6cff5bb7cc5b0ba0426b8235ec478"

Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 11 Found 0 Missed 11 Current 1160 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: tegra-minimal-initramfs-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gtk_icon_cache' failed, details in /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_rootfs
ERROR: Logfile of failure stored in: /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_rootfs.2013867
ERROR: Task (/home/ieu/yocto-nano/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3097 tasks of which 3096 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ieu/yocto-nano/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

log.do_rootfs
```
Complete!

NOTE: Running /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/ieu/yoct>
Dependencies:
%{requires}
Recommendations:
%{recommends}
DependenciesEndHere:

NOTE: Running ['oe-pkgdata-util', '-p', '/home/ieu/yocto-nano/poky-dunfell/build/tmp/pkgdata/jetson-nano-qspi-sd', 'glob', '/tmp/installed-pkgs9cub7_za', '']
NOTE: Installing complementary packages ...  (skipped already provided packages )
NOTE: Running intercept scripts:
NOTE: > Executing update_gtk_icon_cache intercept ...
NOTE: Exit code 127. Output:
/home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/intercept_scripts-1cf78bc5fac98d09e7a947d83965ff1d2afc96cddf0b15f837e80a4c66bbd395/update_g>

ERROR: The postinstall intercept hook 'update_gtk_icon_cache' failed, details in /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.d>
DEBUG: Python function do_rootfs finished
```

madisongh, 2020-12-08:
Did you add anything to your `local.conf` to try to add packages to the image you're building?  That intercept script should not be getting run at all when building the initramfs, but something must be pulling in a package that's causing it to be run.

You might find it easier to get started with [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro/tree/dunfell-l4t-r32.4.3); see the README file there for more info.

bekirbostanci, 2020-12-08:
@madisongh thank you for fast response 
I added these are
MACHINE ??= "jetson-nano-qspi-sd"

IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"

SSTATE_DIR ?= "/home/${USER}/yocto-nano/poky-dunfell/build/sstate_dir"
DL_DIR ?= "/home/${USER}/yocto-nano/poky-dunfell/build/downloads"

madisongh, 2020-12-08:
Well, that looks OK.  Could you attach the full `log.do_rootfs` file that you excerpted from?  I'd like to see what packages it was trying to install.

bekirbostanci, 2020-12-08:
of course I changed the file type normally it is .do_rootfs
[log.txt](https://github.com/OE4T/meta-tegra/files/5661763/log.txt)

madisongh, 2020-12-08:
I've run a couple of test builds with the same setup and can't reproduce this, unfortunately.  Could you try
```
$ bitbake -c clean -c cleansstate tegra-minimal-initramfs
```
then
```
$ bitbake tegra-minimal-initramfs
```
again and see if it recurs?  If it does, you might want to make sure that the filesystem on your host is in OK using fsck (assuming it's ext4).

bekirbostanci, 2020-12-09:
@madisongh hi again 
I have tried but it was not work. I will try again without meta-tegra. If it's work we can undrestand problem is meta tegra

console output 
```
ieu@ieu-ABRA-A5-V15-8:~/yocto-nano/poky-dunfell$ source oe-init-build-env 

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
ieu@ieu-ABRA-A5-V15-8:~/yocto-nano/poky-dunfell/build$ bitbake -c clean -c cleansstate tegra-minimal-initramfs
WARNING: Host distribution "ubuntu-20.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################| Time: 0:00:00
Loaded 1462 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano-qspi-sd"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-tegra           = "dunfell-l4t-r32.4.3:5e1fd3ae2244aa6b61819e462e314a1627c9a5de"
meta                 
meta-poky            
meta-yocto-bsp       = "my-yocto-3.1:5d47cdf448b6cff5bb7cc5b0ba0426b8235ec478"

Initialising tasks: 100% |###########################################################| Time: 0:00:00
Sstate summary: Wanted 0 Found 0 Missed 0 Current 0 (0% match, 0% complete)
NOTE: No setscene tasks
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 2 tasks of which 0 didn't need to be rerun and all succeeded.

Summary: There was 1 WARNING message shown.
ieu@ieu-ABRA-A5-V15-8:~/yocto-nano/poky-dunfell/build$ bitbake tegra-minimal-initramfs
WARNING: Host distribution "ubuntu-20.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#######################################################################################################################################################################| Time: 0:00:00
Loaded 1462 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano-qspi-sd"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-tegra           = "dunfell-l4t-r32.4.3:5e1fd3ae2244aa6b61819e462e314a1627c9a5de"
meta                 
meta-poky            
[... 24 lines cut ...]
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-nano-qspi-sd"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-tegra           = "dunfell-l4t-r32.4.3:5e1fd3ae2244aa6b61819e462e314a1627c9a5de"
meta                 
meta-poky            
meta-yocto-bsp       = "my-yocto-3.1:5d47cdf448b6cff5bb7cc5b0ba0426b8235ec478"

Initialising tasks: 100% |##################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 11 Found 0 Missed 11 Current 1160 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: tegra-minimal-initramfs-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gtk_icon_cache' failed, details in /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_rootfs
ERROR: Logfile of failure stored in: /home/ieu/yocto-nano/poky-dunfell/build/tmp/work/jetson_nano_qspi_sd-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_rootfs.17139
ERROR: Task (/home/ieu/yocto-nano/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3097 tasks of which 3096 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ieu/yocto-nano/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

```

mozcelikors, 2020-12-09:
@bekirbostanci If you are talking about the filesystem type selection, I think you should use ext4. I've had disk-related problems before, dealing with other filesystems.
Furthermore, you should seriously consider trying out https://github.com/OE4T/tegra-demo-distro as suggested by @madisongh. It seems that your setup is very basic and does not contain many layers. If you are just starting out with Yocto, try to use a basis distro rather than integrating meta-tegra with plain poky.
Best of luck from Izmir ;)

bekirbostanci, 2020-12-09:
@mozcelikors thank you for suggestion I will start with tegra demo distro and I will try again. This a start project but I need add one more layer. Maybe I can need to help. I would like to contact you. Have a nice day :)

bekirbostanci, 2020-12-12:
I re-downloaded and bitbake solved the problem

closed 2020-12-12 by bekirbostanci

WuHuaGuoShu001, 2023-08-31:
@bekirbostanci May I ask you to re-download what content, I have encountered the same problem