OE4T/meta-tegra #516: Task do_populate_sdk in .... rdepends upon non-existent task do_package_write_deb

MehdiJaiem502, 2020-12-10:
Hello 

I am trying to build an Image based on the Yocto Project for my stm32mp157c-DK2 according to [st-wiki].(https://wiki.st.com/stm32mpu/wiki/STM32MP1_Distribution_Package#Building_the_OpenSTLinux_distribution)

When I try to build the image I get the following error.
`ERROR: Task do_populate_sdk in /home/mehdi/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-st-openstlinux/recipes-st/images/st-image-weston.bb rdepends upon non-existent task do_package_write_deb in /home/mehdi/Distribution-Package/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-st-openstlinux/recipes-st/images/st-image-weston.bb
ERROR: Command execution failed: 1`

What did I miss or what did I do wrong in my local.conf file ?

Best regards.

I can share with you the content of the file:

#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# This sets the default machine to be qemux86 if no other machine is selected:
#MACHINE ??= "qemux86"
MACHINE = "stm32mp1-disco"

#
# Default policy config
#
# The distribution setting controls which policy settings are used as defaults.
# The default value is fine for general Yocto project use, at least initially.
# Ultimately when creating custom policy, people will likely end up subclassing
# these defaults.
#
# This sets the default distribution to be nodistro if no other distribution is selected:
#DISTRO ??= "nodistro"
DISTRO = "openstlinux-weston"

#
# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"

[... 219 lines cut ...]
# Enable PR server to avoid version-going-backward issue
PRSERV_HOST = "localhost:0"

# =========================================================================
# Configure STM32MP default version to github
# =========================================================================
#STM32MP_SOURCE_SELECTION_pn-linux-stm32mp = "github"
#STM32MP_SOURCE_SELECTION_pn-optee-os-stm32mp = "github"
#STM32MP_SOURCE_SELECTION_pn-tf-a-stm32mp = "github"
#STM32MP_SOURCE_SELECTION_pn-tf-a-stm32mp-ssp = "github"
#STM32MP_SOURCE_SELECTION_pn-u-boot-stm32mp = "github"

# =========================================================================
# Set EULA acceptance
# =========================================================================
ACCEPT_EULA_stm32mp1-disco = "0"
IMAGE_INSTALL_append = " inotify-tools"
IMAGE_INSTALL_append = " wget zsh zlib zip zbar yelp yelp-tools yajl wpa-supplicant which vlan vim usermode unzip u-boot-tools u-boot-stm32mp u-boot tree tiptop time tar systemd sudo surf subversion st-image-weston sensors-iks01a2 ruby rsync python3 pkgconfig pkgconf php perl perf parted patch packagegroup-core-ssh-openssh p7zip opkg openssh opencv nodejs nano ncurses networkmanager nfs-export-root make lzip luajit lshw lmsensors lmsensors-config linux-firmware-bluetooth-bcm4343 libxml-perl linux-firmware libssh2 librsync libmodule-build-perl iw jq json-c jquery jsoncpp json-glib json-spirit iproute2 iperf3 initscripts init-ifupdown hwloc i2c-tools ifupdown initramfs-boot ethtool dfu-util fio dpkg gzip file gperf gnuplot gnulib cjson gcc fuse connman fping dconf dbus cronie curl curlpp cryptsetup broadcom-bt-firmware bluetooth-suspend binutils bind bc bash autofs autoconf bluez5 "
ENABLE_UART = "1"
#Non compatibel nmapnfs-utils
# Enable systemd
DISTRO_FEATURES_append = " systemd "
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""

dwalkes, 2020-12-10:
> I am trying to build an Image based on the Yocto Project for my stm32mp157c-DK2

I think you may be posting in the wrong place?  This project is for the meta-tegra layer for NVIDIA Jetson devices.

MehdiJaiem502, 2020-12-10:
> > I am trying to build an Image based on the Yocto Project for my stm32mp157c-DK2
> 
> I think you may be posting in the wrong place? This project is for the meta-tegra layer for NVIDIA Jetson devices.

I opened too much tabs sorry! Is it possible to delete the issue ?

madisongh, 2020-12-10:
No sweat, @MehdiJaiem502 .  We can just close it.

closed 2020-12-10 by madisongh