OE4T/meta-tegra #529: CI/CD for building Yocto image for Jetson TX2

chandshiv, 2021-01-07:
Hi,

We have built the Yocto image for Jetson TX2 manually following this link https://github.com/OE4T/tegra-demo-distro in Ubuntu 18.04.
Now we are trying to set up the CI/CD for the same in GitLab, so wanted to know how do we set up the NVIDIA SDK manager inside the Docker container (which is used for building the Yocto image)?
Since we need SDK manager to install Cuda libraries, can we directly install Cuda without SDK manager by following link
https://docs.nvidia.com/cuda/cuda-installation-guide-linux/#ubuntu-installation and in local.conf file set the NVIDIA_DEVNET_MIRROR to the path where the Cuda deb is downloaded from wget?
We tried doing this but we get error as "NVIDIA_DEVNET_MIRROR location not found; some recipes may not be available".
Wanted to know if this is right approach or do we have to have NVIDIA SDK manager to install Cuda?
Since it has to be done in CI, installing SDK manager non interactively is quite challenging.
Looking forward for your  suggestions.

Thanks!

madisongh, 2021-01-07:
What I've done for my automated builds is to copy the contents of the SDK Manager's downloads folder (which is typically `$HOME/Downloads/nvidia/sdkm_downloads`) to a location that is accessible from the build machines and point NVIDIA_DEVNET_MIRROR to that URL. In my case, it's an S3 bucket, since my builders are hosted in AWS, but you could use a local filesystem or an http server.

lfdmn, 2021-01-07:
Hi @chandshiv 

You don't need SDK manage in your docker image. All you need is the .deb files.

We way I do it for our projects is to download all the packages from SDK manager on my development computer, then move the downloads folder to a web server that I use as devnet mirror.

I've seen other projects like "Boot to Qt", put the .deb files simply into a git repo. That is checked out int he work-shared directory.

One thing to speed up your builds is to setup folder mappings between you container and the host machine to keep the sstate and downloads folder between builds so you don't have to fully rebuild the images. I do that for RnD builds. For Release builds I like to build from scratch.

chandshiv, 2021-01-07:
> What I've done for my automated builds is to copy the contents of the SDK Manager's downloads folder (which is typically `$HOME/Downloads/nvidia/sdkm_downloads`) to a location that is accessible [...]

How did you install Cuda from the .deb file in the docker image?

madisongh, 2021-01-07:
> How did you install Cuda from the .deb file in the docker image?

There's no need to have CUDA installed for running bitbake builds.

chandshiv, 2021-01-07:
> Hi @chandshiv
>
> You don't need SDK manage in your docker image. All you need is the .deb files.
>
> We way I do it for our projects is to download all the packages from SDK manager on my development [...]

> > How did you install Cuda from the .deb file in the docker image?
> 
> There's no need to have CUDA installed for running bitbake builds.

Ok, so just this .deb file cuda-repo-ubuntu1804-11-2-local_11.2.0-460.27.04-1_amd64.deb is enough for build to be kept in NVIDIA_DEVNET_MIRROR location ?

madisongh, 2021-01-07:
> Ok, so just this .deb file cuda-repo-ubuntu1804-11-2-local_11.2.0-460.27.04-1_amd64.deb is enough for build to be kept in NVIDIA_DEVNET_MIRROR location ?

You need the deb file for the version of JetPack you're building for - the host tools version has to match the version used on the Jetson targets.  As far as I know, we're still at CUDA 10.2.89 with the latest version of JetPack (4.4.1), not 11.2.0.

chandshiv, 2021-01-07:
> CUDA 10.2.89

Yes right, we have used CUDA 10.2.89 manually when we built.
I tried below steps:
* Added below commands to install gcc-8 in docker file,
  - RUN  apt-get update ; apt-get upgrade -y ; apt-get install -y gcc-8 g++-8 
  - RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 8
  - RUN update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-8 8
 * Have added below for build in build.sh file 
[image]
* Below is part of local.conf file 
[image]

With this configuration when I build I am getting below error :
[image]

Please let me know if I am missing any thing here?

madisongh, 2021-01-07:
As I mentioned before, you *have to* use the CUDA tools downloaded via SDK manager and store them yourself for use during the build.  (If they were generally available via http without having to sign in with your NVIDIA developer account, we'd already be using that in the recipe.)  The version you are `wget`ting is not the right version.  Manually download the package via SDK manager and store it somewhere you control

Since you're using tegra-demo-distro as your base, you should:
* set NVIDIA_DEVNET_MIRROR to point to the location where you store the CUDA package in the environment before sourcing the `setup-env` script - the script will automatically add the setting in `local.conf` for you, if you have that environment variable set.
* Do not try to replace the `local.conf` or `bblayers.conf` files (although it looks like your `cp` commands are failing anyway). The settings in your local.conf file are unnecessary with the demo distro

The setup script will automatically set things up for using gcc-8 and g++-8 if they're installed; you don't need to do the update-alternatives.  From the log snippet, it looks like they didn't actually get installed into the container, though.

chandshiv, 2021-01-11:
> As I mentioned before, you _have to_ use the CUDA tools downloaded via SDK manager and store them yourself for use during the build. (If they were generally available via http without having to sign [...]

Thank you for all your inputs, it was issue with conf files location so gcc error was bit misleading.
 I could sort it out but then build failed as I did not had the correct version of CUDA package as expected.

 I could build Minimalist image without CUDA, but when I tried to add IoTEdge layer I get below issue 

I am cloning it from below and add these layers in bblayers.conf
git clone -b dunfell https://github.com/Azure/meta-iotedge.git
git clone -b common-rust-native git://github.com/meta-rust/meta-rust.git 
[image]

Added this in local.conf 
[image]

Please let us know if we are missing anything here

madisongh, 2021-01-11:
There must be something wrong with your configuration, I'm just not sure what it is.  You'll need to track down why you're getting the ``ERROR: Nothing RPROVIDES 'iotedge-daemon'` message.  Sounds like bitbake is recognizing the recipe/package name, but it's not going to get built for some reason

ichergui, 2021-02-10:
Hey @chandshiv @madisongh 
Agree with @madisongh, bitbake is not able build `iotedge-daemon` and claim `Nothing RPROVIDES 'iotedge-daemon'`
I will try this later and keep you updated.

ichergui, 2021-02-11:
Hey @chandshiv

ichergui, 2021-02-15:
Hello @chandshiv 

Did you try the steps I sent you ? 
Could you please give us updates about your issue ?

ichergui, 2021-02-23:
No feedback from @chandshiv , I will close this issue

closed 2021-02-23 by ichergui