OE4T/meta-tegra #531: License Checksum Error when building jetson nano 2gb.

ahmedalshamary, 2021-01-13:
**Describe the bug**
When trying to build the base meta-tegra for the jetson nano dev kit I am getting a license checksum mismatch error. I am fairly inexperienced with the Yocto Project so I am not sure if this is something I am doing wrong.  I am also using kas for which I have put the file below.

```
 DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function do_populate_lic
DEBUG: We need to use a SPDXLICENSEMAP for GPLv2
DEBUG: Python function do_populate_lic finished
DEBUG: Executing python function populate_lic_qa_checksum
ERROR: QA Issue: u-boot-tegra: The LIC_FILES_CHKSUM does not match for file://Licenses/README;md5=30503fd321432fc713238f582193b78e
u-boot-tegra: The new md5 checksum is 5a7450c57ffe5ae63fd732446b988025
u-boot-tegra: Here is the selected license text:
vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
SPDX-License-Identifier: GPL-2.0

  U-Boot is Free Software.  It is copyrighted by Wolfgang Denk and
many others who contributed code (see the actual source code and the
git commit messages for details).  You can redistribute U-Boot and/or
modify it under the terms of version 2 of the GNU General Public
License as published by the Free Software Foundation.  Most of it can
also be distributed, at your option, under any later version of the
GNU General Public License -- see individual files for exceptions.

...
GNU Lesser General Public License v2.1 or later LGPL-2.1+ Y   lgpl-2.1.txt    http://www.gnu.org/licenses/old-licenses/lgpl-2.1.txt
eCos license version 2.0      eCos-2.0      eCos-2.0.txt    http://www.gnu.org/licenses/ecos-license.html
BSD 2-Clause License        BSD-2-Clause  Y   bsd-2-clause.txt  http://spdx.org/licenses/BSD-2-Clause
BSD 3-clause "New" or "Revised" License   BSD-3-Clause  Y   bsd-3-clause.txt  http://spdx.org/licenses/BSD-3-Clause#licenseText
IBM PIBS (PowerPC Initialization and    IBM-pibs      ibm-pibs.txt
  Boot Software) license
ISC License         ISC   Y   isc.txt     https://spdx.org/licenses/ISC
MIT License         MIT   Y   mit.txt     https://spdx.org/licenses/MIT.html
SIL OPEN FONT LICENSE (OFL-1.1)     OFL-1.1   Y   OFL.txt     https://spdx.org/licenses/OFL-1.1.html
X11 License         X11       x11.txt     https://spdx.org/licenses/X11.html
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
u-boot-tegra: Check if the license information has changed in /home/ahmed/Projects/build/tmp/work/jetson_nano_2gb_devkit-poky-linux/u-boot-tegra/1_2020.10+gAUTOINC+24abfbe6db-r0/git/Licenses/README to verify that the LICENSE value "GPLv2+" remains valid [license-checksum]
ERROR: Fatal QA errors found, failing task.
DEBUG: Python function populate_lic_qa_checksum finished
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch '...' with `MACHINE` based on '...'
local.conf :
```
# diskmon
BB_DISKMON_DIRS = "\
  STOPTASKS,${TMPDIR},1G,100K \
  STOPTASKS,${DL_DIR},1G,100K \
  STOPTASKS,${SSTATE_DIR},1G,100K \
  STOPTASKS,/tmp,100M,100K \
  ABORT,${TMPDIR},100M,1K \
  ABORT,${DL_DIR},100M,1K \
  ABORT,${SSTATE_DIR},100M,1K \
  ABORT,/tmp,10M,1K"

# downloads
DL_DIR = "/home/ahmed/yocto/downloads"

# package_ipk
PACKAGE_CLASSES = "package_ipk"

# pseudo_ignore_append
PSEUDO_IGNORE_PATHS_append = ",/run/" 

# sstate
SSTATE_DIR = "/home/ahmed/yocto/sstate-cache"

# standard
CONF_VERSION = "1"

MACHINE ??= "jetson-nano-2gb-devkit"
DISTRO ??= "poky"
BBMULTICONFIG ?= ""
```

bblayers.conf : 
```
# standard
POKY_BBLAYERS_CONF_VERSION = "2"
BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
    /home/ahmed/Projects/test/meta-tegra \
    /home/ahmed/Projects/test/poky/meta \
    /home/ahmed/Projects/test/poky/meta-poky"
```

2. Build with bitbake argument '...' 
```
bitbake -k -c build core-image-minimal
```

3. Kas File
```
  version: 10
  includes:
    - kas_local.yml

machine: jetson-nano-2gb-devkit
distro: poky

repos:
  poky:
    url: "git://git.yoctoproject.org/poky"
    refspec: 943ef2fad8428f002850e3655a3312e13d0dcb2c
    layers:
      meta:
      meta-poky:

  meta-tegra:
    url: "https://github.com/OE4T/meta-tegra"
    refspec: 47200e8aca41c668a845815c95ea99bfb32ed052

bblayers_conf_header:
  standard: |
    POKY_BBLAYERS_CONF_VERSION = "2"
    BBPATH = "${TOPDIR}"
    BBFILES ?= ""

local_conf_header:
  standard: |
    CONF_VERSION = "1"

  package_ipk: |
    PACKAGE_CLASSES = "package_ipk"

  pseudo_ignore_append: |
    PSEUDO_IGNORE_PATHS_append = ",/run/"

  diskmon: |
    BB_DISKMON_DIRS = "\
      STOPTASKS,${TMPDIR},1G,100K \
      STOPTASKS,${DL_DIR},1G,100K \
      STOPTASKS,${SSTATE_DIR},1G,100K \
      STOPTASKS,/tmp,100M,100K \
      ABORT,${TMPDIR},100M,1K \
      ABORT,${DL_DIR},100M,1K \
      ABORT,${SSTATE_DIR},100M,1K \
      ABORT,/tmp,10M,1K"
```

madisongh, 2021-01-13:
From the refspecs, it looks like you are trying to mix the gatesgarth branch of poky with the master branch of meta-tegra.  The meta-tegra branches correspond to the branches they track in poky/OE-Core, so try either switching meta-tegra to gatesgarth or using the master branch of poky.

You might also consider using our [https://github.com/OE4T/tegra-demo-distro](demo distro) as a starting point for your builds, which is already set up with several layers.

ahmedalshamary, 2021-01-13:
Thanks so much, that fixed my issue. I'll look at the demo distro as a guideline from now on. Thanks for all your help !

closed 2021-01-13 by ahmedalshamary