OE4T/meta-tegra #539: [jetson-xavier-nx-devkit] OEM authentication of kernel header failed!

brgl, 2021-01-25:
I can't find any precise instructions on how to built a yocto image for an Xavier NX development kit but I did the usual thing of building a core-image-full-cmdline with meta-tegra gatesgarth branch and tried to create an SD card image with make-sdcard script.

I modified the flash.xml.in file and set the APP file to the .ext4 partition image generated by yocto to make it work.

Unfortunately the device fails to boot with the [following](https://pastebin.com/0yYmwQ1U) boot log from UART console.

Is there anything else I should do before being able to boot from the SD card?

madisongh, 2021-01-25:
The [wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) could use an update to mention the Xavier NX dev kit, but the instructions are similar to the ones for the Nano dev kit:

1. Use the `dosdcard.sh` script to generate the SD card image
2. If this is the first time you're booting the NX dev kit from a Yocto-based build, make sure you also reprogram the QSPI flash so the bootloader and BCT match up with what you have on the SDcard.

For that second step:
1. Insert the SDcard you wrote into the slot on the device
2. Connect a USB cable from your host to the type B/OTG port
3. Put the device into forced-recovery mode via the jumper
4. Use `./doflash.sh --spi-only` to program just the QSPI flash

brgl, 2021-01-26:
@madisongh thanks, this works. Sorry for the noise, I didn't know Nano is so similar.

closed 2021-01-26 by brgl