OE4T/meta-tegra #541: Cboot (cboot-t18x) compilation problem

nefethael, 2021-01-26:
With Yocto gatesgarth and jetson-tx2 machine, cboot does not compile with following error:

```
aarch64-my-linux-ld: <path>/cboot-t18x/32.4.4-r0/build/build-t186/bootloader/partner/common/drivers/sdmmc/tegrabl_sdmmc_protocol.o:<path>/cboot-t18x/32.4.4-r0/cboot-t18x-32.4.4/bootloader/partner/t18x/cboot/platform/t186/../../../../common/drivers/sdmmc/tegrabl_sdmmc_defs.h:227: multiple definition of `sdmmc_context_t'; <path>/cboot-t18x/32.4.4-r0/build/build-t186/bootloader/partner/common/drivers/sdmmc/tegrabl_sdmmc_bdev.o:<path>/cboot-t18x/32.4.4-r0/cboot-t18x-32.4.4/bootloader/partner/t18x/cboot/platform/t186/../../../../common/drivers/sdmmc/tegrabl_sdmmc_defs.h:227: first defined here
```

Adding missinng `typedef` keyword for struct `sdmmc_context_t` fixes the issue.

brgl, 2021-01-26:
You can also set GCCVERSION to "8.%" and include the meta-tegra/contrib layer to work around this problem.

madisongh, 2021-01-26:
Good point. I don't think I've tested building cboot from sources with gcc 10.

closed 2021-02-02 by madisongh