OE4T/meta-tegra #542: Integrating YOLO model to Yocto

mozcelikors, 2021-01-26:
Hi, previously here in this forum, thanks to @madisongh we have been able to integrate SSD-MobileNet model to Yocto by integrating jetson-inference to Yocto (https://github.com/OE4T/meta-tegra/issues/440).

Now, I would like to integrate the YOLO model by writing a recipe for tensorrt_demos repository. (https://github.com/jkjung-avt/tensorrt_demos)

What I got so far:
```
SUMMARY = "tensorrt-demos"
DESCRIPTION = "tensorrt-demos"
LICENSE = "CLOSED"

FILESEXTRAPATHS_prepend:="${THISDIR}/${PN}:"

SRC_URI = "file://tensorrt_demos"

S = "${WORKDIR}/tensorrt_demos"

RDEPENDS_${PN} = " gstreamer1.0-plugins-nveglgles gstreamer1.0-plugins-nvvideo4linux2 \
                   gstreamer1.0-plugins-bad gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly gstreamer1.0-plugins-tegra gstreamer1.0-libav \
                   gstreamer1.0-omx-tegra ${PYTHON_PN} ${PYTHON_PN}-core ${PYTHON_PN}-cython ${PYTHON_PN}-numpy cuda-nvtx tensorrt"

DEPENDS = "${PYTHON_PN} ${PYTHON_PN}-numpy ${PYTHON_PN}-numpy-native ${PYTHON_PN}-cython-native tensorrt cudnn cuda-nvtx tegra-libraries v4l-utils glew gstreamer1.0 gstreamer1.0-plugins-base \
           glib-2.0 libxml2"

inherit pkgconfig cuda python3native python3-dir

do_compile_prepend(){
    export STAGING_DIR="${STAGING_DIR_NATIVE}"
    export STAGING_DIR_TARGET="${STAGING_DIR_TARGET}"
    export CUFLAGS="${CUFLAGS}"
    export CUDACXX="${CUDACXX}"
}

do_compile_append(){
    cd ${S}/plugins
    oe_runmake -C .
}

do_fetch[nostamp] = "1"

FILES_SOLIBSDEV = ""
FILES_${PN} = "${PYTHON_SITEPACKAGES_DIR} ${bindir}/*.py"
FILES_${PN} += "${libdir}/lib*${SOLIBSDEV}"

```

Two things need to be built. First Makefile > setup.py, then plugins/Makefile.
For this purpose, I have modified setup.py and plugins/Makefile as follows:

**setup.py**
```
from distutils.core import setup
from distutils.extension import Extension
from Cython.Distutils import build_ext
from Cython.Build import cythonize

import numpy
import os

library_dirs = [
    os.getenv('STAGING_DIR')+'/usr/local/cuda-10.2/lib64',
    os.getenv('STAGING_DIR')+'/usr/local/TensorRT-7.1.3.4/lib',  # for my x86_64 PC
    os.getenv('STAGING_DIR')+'/usr/local/lib',
    os.getenv('STAGING_DIR_TARGET')+'/usr/lib',
]

libraries = [
    'nvinfer',
    'cudnn',
    'cublas',
    'cudart_static',
[... 98 lines cut ...]
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN8nvinfer117YoloPluginCreatorE' which may bind externally can not be used when making a shared object; recompile with -fPIC
| yolo_layer.o: in function `nvinfer1::YoloPluginCreator::~YoloPluginCreator()':
| /usr/src/debug/tensorrt-demos/1.0-r0/tensorrt_demos/plugins/yolo_layer.h:116:(.text._ZN8nvinfer117YoloPluginCreatorD0Ev[_ZN8nvinfer117YoloPluginCreatorD5Ev]+0x8): dangerous relocation: unsupported relocation
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN8nvinfer117YoloPluginCreatorE' which may bind externally can not be used when making a shared object; recompile with -fPIC
| /usr/src/debug/tensorrt-demos/1.0-r0/tensorrt_demos/plugins/yolo_layer.h:116:(.text._ZN8nvinfer115PluginRegistrarINS_17YoloPluginCreatorEED2Ev[_ZN8nvinfer115PluginRegistrarINS_17YoloPluginCreatorEED5Ev]+0x4): dangerous relocation: unsupported relocation
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZN8nvinfer122CalDetection_NewCoordsEPKfPfiiiiS1_iiif' which may bind externally can not be used when making a shared object; recompile with -fPIC
| yolo_layer.o: in function `__sti____cudaRegisterAll()':
| /tmp/tmpxft_000118ec_00000000-5_yolo_layer.cudafe1.stub.c:1:(.text.startup+0x48): dangerous relocation: unsupported relocation
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZN8nvinfer112CalDetectionEPKfPfiiiiS1_iiif' which may bind externally can not be used when making a shared object; recompile with -fPIC
| yolo_layer.o: in function `__nv_cudaEntityRegisterCallback':
| /tmp/tmpxft_000118ec_00000000-5_yolo_layer.cudafe1.stub.c:1:(.text.startup+0x7c): dangerous relocation: unsupported relocation
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN8nvinfer117YoloPluginCreatorE' which may bind externally can not be used when making a shared object; recompile with -fPIC
| yolo_layer.o: in function `nvinfer1::YoloPluginCreator::YoloPluginCreator()':
| /usr/src/debug/tensorrt-demos/1.0-r0/tensorrt_demos/plugins/yolo_layer.cu:329:(.text.startup+0xec): dangerous relocation: unsupported relocation
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: yolo_layer.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZNSt6vectorIN8nvinfer111PluginFieldESaIS1_EED1Ev' which may bind externally can not be used when making a shared object; recompile with -fPIC
| yolo_layer.o: in function `_GLOBAL__sub_I__ZN8nvinfer115YoloLayerPluginC2EiiiPfiiifi':
| /usr/src/debug/tensorrt-demos/1.0-r0/tensorrt_demos/plugins/yolo_layer.cu:429:(.text.startup+0x104): dangerous relocation: unsupported relocation
| collect2: error: ld returned 1 exit status
| make: *** [Makefile:22: libyolo_layer.so] Error 1
| make: Leaving directory '/home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a_tegra194-poky-linux/tensorrt-demos/1.0-r0/tensorrt_demos/plugins'
| WARNING: exit code 1 from a shell command.
| 
```

I would appreciate if I could get some pointers here how to proceed. Thank you very much in advance.

madisongh, 2021-01-27:
There may be something else as well, but check to make sure that the right C++ compilation flags for building a shared library (e.g., -fPIC) are being used and being passed through nvcc (via -Xcompiler)

mozcelikors, 2021-01-27:
That immediately solved the problem in question, thanks.

```
	$(NVCC)  ${CUFLAGS} -Xcompiler -fPIC -c -o $@ $<
```
I will still be investigating and updating this thread.

mozcelikors, 2021-02-01:
When I try to copy the build folder to the platform, I'm getting some Q/A and rootfs errors. Any idea how to fix the root issue here? The recipe and Makefile stays almost the same.
I am compiling with native nvcc and using the target libraries.
```

Error: 
 Problem: conflicting requests
  - nothing provides libcublas.so.10.0()(64bit) needed by tensorrt-demos-1.0-r0.armv8a_tegra194
  - nothing provides libcublas.so.10.0(libcublas.so.10.0)(64bit) needed by tensorrt-demos-1.0-r0.armv8a_tegra194
  - nothing provides libnvinfer.so.5()(64bit) needed by tensorrt-demos-1.0-r0.armv8a_tegra194
  - nothing provides libnvinfer.so.6()(64bit) needed by tensorrt-demos-1.0-r0.armv8a_tegra194
```

madisongh, 2021-02-01:
> but for file-rdeps I don't know what to do. Why it is linked to different libraries?

It looks like those libraries are pre-built binaries provided in the source repo. Not much you can do about that (if the sources aren't provided), other than not install them (which may mean not installing that entire subdirectory).

bhargavthriler, 2021-02-01:
> but for file-rdeps I don't know what to do. Why it is linked to different libraries?

Hi, you can try this in your recipe 
`INSANE_SKIP_${PN} = "file-rdeps"`

I know this is not the correct way to do, but it will proceed with the build. 
Recently I had some issues with a precompiled Qt binary and this solved my issue.

mozcelikors, 2021-02-02:
Unfortunately, that does not work for me, then I get do_rootfs errors saying the same thing.
There must be something I'm doing unproperly in the Makefiles with the linker, I have yet to discover what.

mozcelikors, 2021-02-02:
@madisongh I hadn't notice that they were provided with the repo. I won't install them. Thanks

mozcelikors, 2021-02-04:
I was able to compile tensorrt_demos successfuly. Now it comes down to the dependencies. I would like to create a recipe for python3-onnx (onnx pypi package). So far I have:

```
SUMMARY = "python3-onnx"

LICENSE = "CLOSED"

inherit pypi setuptools3

SRC_URI += "https://files.pythonhosted.org/packages/59/09/61f817a530cd9b9bd0e289bc9cda46f662fcda4dc1bd687b36d45365e3ea/onnx-1.4.1.tar.gz"
SRC_URI[sha256sum] = "50f3cc33358685428ff7ab23042c2eef1d73fb58c38c7cc9b0a1b22af9086ee9"

RDEPENDS_${PN} += " protobuf protobuf-c "
DEPENDS += " protobuf protobuf-native protobuf-c-native cmake-native python3 python3-native python3-pytest-runner-native"

EXTRA_OECMAKE += "-DONNX_USE_PROTOBUF_SHARED_LIBS=OFF -DProtobuf_USE_STATIC_LIBS=ON"

do_compile_prepend(){
    ln -s ${STAGING_DIR_NATIVE}/usr/bin/python3-native/python3 ${STAGING_DIR_NATIVE}/usr/bin/python
}

```
I am getting protobuf related issues here, the error is:

```
| /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a-poky-linux/python3-onnx/1.4.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/ld: /home/mozcelikors/projects/Cladis/CLADIS/cladis-yocto/poky/build/tmp/work/armv8a-poky-linux/python3-onnx/1.4.1-r0/recipe-sysroot-native/usr/lib/libprotobuf.so: error adding symbols: file in wrong format
| collect2: error: ld returned 1 exit status

```
I have searched online and could not find any concrete solution. Any help is appreciated here. 
I will be sharing all of the recipes (final versions) with the community for those who want to also integrate Yolo model to the Yocto.

madisongh, 2021-02-04:
Looks like it's using the native protobuf for linking rather than the target protobuf.  You need native protobuf for running any code generators, but it's likely that their makefiles (or cmake files, or whatever) assume that there's just one protobuf package to find, rather than splitting host and target.

mozcelikors, 2021-02-09:
Manually editing CMakeLists.txt to seperate target-native lib/include/bins did the trick.
```
  set (ONNX_PROTOC_EXECUTABLE "$ENV{STAGING_DIR_NATIVE}/usr/bin/protoc")
  set (Protobuf_PROTOC_EXECUTABLE "$ENV{STAGING_DIR_NATIVE}/usr/bin/protoc")
  set (Protobuf_PROTOC_LIBRARY "$ENV{STAGING_DIR_TARGET}/usr/lib/libprotobuf.so")
  set (Protobuf_INCLUDE_DIR "$ENV{STAGING_DIR_TARGET}/usr/include")
  #find_package(Protobuf REQUIRED)
  set (Protobuf_PROTOC_LIBRARY "$ENV{STAGING_DIR_TARGET}/usr/lib/libprotobuf.so")
  set (PROTOBUF_LIBRARIES "$ENV{STAGING_DIR_TARGET}/usr/lib/libprotobuf.so")
```
Now, it gives me runtime error 'tensorrt' python package not found.
I checked online and found out that only way to do this is from tensorrt sources and python3-pip-native installing the wheel file. Apparently current tensorrt recipe does not install the Python library.

So I'm looking to do something like:

```
DEPENDS_append = " python3-pip-native"

do_install_append() {
    pip3 install ${S}/tensorflow......._aarch64.whl
}
```
Do you know how to access this whl file, as the tensorrt source is not complete under /tmp. (as far as i can see)

madisongh, 2021-02-09:
>Apparently current tensorrt recipe does not install the Python library.

We would need to add a recipe for it, to download and repackage the python3-libnvinfer package from JetPack.

Unfortunately, just as for the DeepStream SDK, the shared library in that package which provides the Python bindings was built against Python 3.6, and is unlikely to work with the version of Python we currently have in OE-Core (that was the case with DeepStream).

There is no easy way to address that problem.  You can see the discussion over on #425 for more info.

mozcelikors, 2021-02-15:
Hmm, that is very bad. Is this on the agenda? How can I help speed it up?
If there is anything you can suggest to try out, I'm more than happy to do so.
I need this for my project to go on, should I try porting python 3.6 instead of python 3.9?
If it is sensible to you, where do I find the python3-libnvinfer sources/package for arm64 ?

dwalkes, 2021-02-15:
Hi @mozcelikors 

> I need this for my project to go on

Have you considered/tried using docker containers instead as the solution/workaround for this?  This is how we are currently using deepstream with meta-tegra.  See https://ngc.nvidia.com/catalog/containers/nvidia:deepstream-l4t for instance.

mozcelikors, 2021-02-16:
@dwalkes Hi
Were you able to run tensorrt/yolo stuff in Yocto-based distro this way?
If so, I can give it a shot. As far as I understood, my roadmap would be to first integrate docker to Yocto distro, then run this docker image in it? Is that correct?
Thanks for answers in advance.

dwalkes, 2021-02-16:
> Were you able to run tensorrt/yolo stuff in Yocto-based distro this way?

TensorRT yes.  I haven't tried yolo specifically but I'd expect it should work to dockerize something like the repo at https://github.com/jkjung-avt/tensorrt_demos.  Look at Dusty's repo at https://github.com/dusty-nv/jetson-inference as a model. The [deepstream](https://ngc.nvidia.com/catalog/containers/nvidia:deepstream-l4t) samples I've tried work on meta-tegra `dunfell-l4t-r32.4.3` using the NVIDIA docker runtime.

> As far as I understood, my roadmap would be to first integrate docker to Yocto distro, then run this docker image in it? Is that correct?

Yes, exactly.  You'll need to make sure you are using `--runtime nvidia` to run your containers or [set as the default runtime](http://reader.epubee.com/books/mobile/40/40b7b4f104fa62b2b8fdda2b7c4b0503/text00340.html).  You'll need the default runtime setting if you are building docker containers on target.  You can alternatively build your containers on the devkit and then run them on target.

mozcelikors, 2021-02-18:
Seems quite challenging (to me) to base this on jetson-inference docker, since all the assumptions are made for Ubuntu in the scripts, also Yocto release doesn't provide some tools that are required by the jetson-inference docker.

```
root@jetson-xavier-nx-devkit:/opt/jetson-inference/jetson-inference/jetson-inference# ./docker/run.sh 
head: cannot open '/etc/nv_tegra_release' for reading: No such file or directory
reading L4T version from "dpkg-query --show nvidia-l4t-core"
tools/l4t-version.sh: line 29: dpkg-query: command not found
L4T BSP Version:  L4T R.
docker/tag.sh: line 18: [: -eq: unary operator expected
cannot find compatible jetson-inference docker container for L4T R.
please upgrade to the latest JetPack, or build jetson-inference natively from source
root@jetson-xavier-nx-devkit:/opt/jetson-inference/jetson-inference/jetson-inference# 
```

This link went successful: https://developer.ridgerun.com/wiki/index.php?title=Yocto_Support_for_NVIDIA_Jetson_Platforms-Old#DeepStream
I'm able to run this docker:
`docker run -it --rm --net=host --runtime nvidia -w /opt/nvidia/deepstream/deepstream-5.0 nvcr.io/nvidia/deepstream-l4t:5.0-dp-20.04-samples`
But I'm getting "nvcc" not found, when building tensorrt_demos, so there are somethings missing there for sure.

Being the docker novice that I am, how should I proceed? I just want tensorrt python library integrated so that yolo model works on the Yocto distribution. I am having excellent results with the Yolo model on Ubuntu/Jetson Xavier NX so far that needs to be integrated to Yocto.

So three ways that I see, all can be challenging:
1) deepstream-5.0 docker as base, and install nvcc compiler and all the other stuff on it.
2) jetson-inference docker as base, refactor all the Ubuntu-dependent scripts, and see where it leads.
3) docker-less method: (try to) somehow integrate python 3.6 to the existing Yocto distro

Which one should I prefer and is there an easier method that I overlooked? Maybe a Dockerfile already prepared for this kind of work.

Thanks in advance.

dwalkes, 2021-02-18:
> But I'm getting "nvcc" not found,

We haven't done much with cuda compile on meta-tegra target but we did setup a simple example based on l4tbase at https://gitlab.com/boulderai/bai-edge-sdk/-/tree/master/cuda-compile, this would be based on tegra-demo-distro dunfell-r32.4.3 with `DISTRO_EXTRA_RDEPENDS` including:
```
                           cuda-libraries \
                           cuda-toolkit \
```

> Which one should I prefer and is there an easier method that I overlooked?

I think the one I'd try first is experimenting with NGC containers on a devkit running Ubuntu and seeing if you can get a container built and running there with the dependencies you need for NVIDIA components.  Once you get that working and have a working runtime container you can push up to docker hub, then pull back down and run on target.  That way you don't need to get the build dependencies on the target and the target docker container, only the runtime dependencies, which will probably be much easier to find and manage.

You can use [multistage docker builds](https://docs.docker.com/develop/develop-images/multistage-build/) to use different containers for build/runtime based on what you need in each, in case that makes anything easier.  So rather than install cuda in the deepstream container for instance you might be able to build the cuda portion on something like l4t-base, then take the outputs of the build and use with a deepstream or other runtime container.

Since I haven't done this exact thing before this might not be the best approach, but this is the place I would start, that and maybe some posts on the NVIDIA Jetson developer forum.

mozcelikors, 2021-02-22:
Good news!
I was able to generate tensorrt python 3.9 bindings for TensorRT 7.2.2.3 using the repository at https://github.com/NVIDIA/TensorRT. A bit messy at the moment, but I will clean it up along with the tensorrt-demos repository. All in all, Yolo model seems to work at first look.

```
#
# Recipe for creating and installing TensorRT python 3.9 bindings
#
#  Author:
#    M.Ozcelikors <github.com/mozcelikors>
#

SUMMARY = "TensorRT python 3.9 bindings"
DESCRIPTION = "TensorRT python 3.9 bindings"
LICENSE = "CLOSED"

FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI = "git://github.com/NVIDIA/TensorRT.git;protocol=https"
SRC_URI += "file://git/python/CMakeLists.txt"

S = "${WORKDIR}/git/python"

SRCREV = "13f7441a06be3d86397fe3578d789757fa36b61b"

RDEPENDS_${PN} = " bash "

DEPENDS = " python3 python3-pybind11 tensorrt python3-setuptools-native python3-wheel-native python3-pip-native"

inherit cmake cuda python3native python3-dir

TARGET_ARCHITECTURE = "aarch64"
PYTHON_MAJOR_VERSION = "3"
PYTHON_MINOR_VERSION = "9"
EXT_PATH = "/tmp/external"
CUDA_INCLUDE_DIRS = "${STAGING_DIR_TARGET}/usr/local/cuda-10.2/include"
CUDA_ROOT = "${STAGING_DIR_TARGET}/usr/local/cuda-10.2"
TENSORRT_ROOT_PATH = "${S}/.."
TENSORRT_BUILD = "${B}"
PY_CONFIG_INCLUDE = "${STAGING_DIR_TARGET}/usr/include/python3.9"
PY_INCLUDE = "${STAGING_DIR_TARGET}/usr/include/python3.9"
WHEEL_OUTPUT_DIR = "${B}"

TARGET_CC_ARCH += "${LDFLAGS}"

EXTRA_OECMAKE = " -DCMAKE_BUILD_TYPE=Release \
         -DTARGET_ARCHITECTURE=${TARGET_ARCHITECTURE} \
         -DPYTHON_MAJOR_VERSION=${PYTHON_MAJOR_VERSION} \
         -DPYTHON_MINOR_VERSION=${PYTHON_MINOR_VERSION} \
         -DEXT_PATH=${EXT_PATH} \
         -DCUDA_INCLUDE_DIRS=${CUDA_INCLUDE_DIRS} \
         -DTENSORRT_ROOT=${TENSORRT_ROOT_PATH} \
         -DTENSORRT_BUILD=${TENSORRT_BUILD} \
         -DPY_CONFIG_INCLUDE=${PY_CONFIG_INCLUDE} \
         -DPY_INCLUDE=${PY_INCLUDE} \
         -DCUDA_ROOT=${CUDA_ROOT} \
         "

expand_vars_cp () {
    test -f ${1} || (echo "ERROR: File: ${1} does not exist!" && exit 1); \
    sed -e "s|\#\#TENSORRT_VERSION\#\#|${TRT_VERSION}|g" \
        -e "s|\#\#TENSORRT_MAJMINPATCH\#\#|${TRT_MAJMINPATCH}|g" \
        ${1} > ${2}
}

do_compile_append(){
    # Generate wheel
    TRT_MAJOR=$(awk '/^\#define NV_TENSORRT_MAJOR/ {print $3}' ${TENSORRT_ROOT_PATH}/include/NvInferVersion.h)
    TRT_MINOR=$(awk '/^\#define NV_TENSORRT_MINOR/ {print $3}' ${TENSORRT_ROOT_PATH}/include/NvInferVersion.h)
    TRT_PATCH=$(awk '/^\#define NV_TENSORRT_PATCH/ {print $3}' ${TENSORRT_ROOT_PATH}/include/NvInferVersion.h)
    TRT_BUILD=$(awk '/^\#define NV_TENSORRT_BUILD/ {print $3}' ${TENSORRT_ROOT_PATH}/include/NvInferVersion.h)
    TRT_VERSION=${TRT_MAJOR}.${TRT_MINOR}.${TRT_PATCH}.${TRT_BUILD}
    TRT_MAJMINPATCH=${TRT_MAJOR}.${TRT_MINOR}.${TRT_PATCH}
    
    echo "Generating python ${PYTHON_MAJOR_VERSION}.${PYTHON_MINOR_VERSION} bindings for TensorRT ${TRT_VERSION}"

    cd ${S}/packaging
    
    for dir in $(find . -type d); do mkdir -p ${WHEEL_OUTPUT_DIR}/$dir; done
    for file in $(find . -type f); do expand_vars_cp $file ${WHEEL_OUTPUT_DIR}/${file}; done
    
    cd ${WHEEL_OUTPUT_DIR}

    python3 setup.py -q bdist_wheel --python-tag=cp${PYTHON_MAJOR_VERSION}${PYTHON_MINOR_VERSION} --plat-name=linux_${TARGET_ARCHITECTURE}
}

do_install(){
    # TODO: Wheel-based installation
    #pip3 install ${WHEEL_OUTPUT_DIR}/dist/tensorrt-7.2.2.3-cp${PYTHON_MAJOR_VERSION}${PYTHON_MINOR_VERSION}-none-linux_${TARGET_ARCHITECTURE}.whl
    
    # Instead, we simply install by copying:
    install -d ${D}/usr/lib/python3.9/site-packages/tensorrt
    cp -r ${B}/tensorrt/* ${D}/usr/lib/python3.9/site-packages/tensorrt
}

FILES_${PN} += "/usr/lib/python3.9/site-packages/tensorrt"

```

Thanks for all the help along the way.

nandra, 2021-02-24:
> SRC_URI += "file://git/python/CMakeLists.txt"

@mozcelikors can you please post also above file? I'm trying to make your recipe working in dunfell f (but for dunfell) with python 3.8 and I'm getting issues like:

```
build/tmp/work/armv8a_tegra210-poky-linux/python3-tersorrt/git-r0/recipe-sysroot/usr/include/python3.8/pyconfig.h:44:10: fatal error: python3.8/pyconfig-64.h: No such file or directory
|  #include <python3.8/pyconfig-64.h>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~
```

I checked and such file exists in sysroot.

and also:
```
build/tmp/work/armv8a_tegra210-poky-linux/python3-tersorrt/git-r0/git/python/../include/NvInferRuntimeCommon.h:20:10: fatal error: cstddef: No such file or directory
|  #include <cstddef>
|           ^~~~~~~~~
| compilation terminated.

```

This seems to be bit strange as this file is part of toolchain. Any ideas? Thanks.

mozcelikors, 2021-02-24:
@nandra I believe I only commented CFLAGS in that file. I will double check tomorrow when I am on the PC.

nandra, 2021-02-25:
@mozcelikors was able to make it working with dunfell branch. I have recipe I'll share. @madisongh should it go also to external directory or main? Thanks.

madisongh, 2021-02-25:
@nandra With the dependency on `python3-pybind11`, which is in meta-python, it will have to go under `external`.

closed 2021-03-02 by mozcelikors