OE4T/meta-tegra #551: kernel-bup-payload failes with custom xavier dtb file on Dunfell l4t r32.5.0 rr

hannes09, 2021-01-29:
Hi,

i have tested the latest PR to upgrade to jetpack 4.5 with Dunfell l4t r32.5.0 rr. **[](url)**
Im on a custom carrier board with  a custom dtb file.

With version 32.4  everything was building and I had a bootbale image.
Since 32.5.0 the kernel-bup-payload task fails for some reason.

Some parts of the log:

`| Generating signed binaries for 2888-400-0001-J.0-1-2-syslogic-xavier-mmcblk0p1 ...
| python3 tegraflash.py --chip 0x19 --bl nvtboot_recovery_cpu_t194.bin --sdram_config syslogic-xavier.cfg,syslogic-xavier-override.cfg --odmdata 0x8590000 --applet mb1_t194_prod.bin --soft_fuses tegra194-mb1-soft-fuses-l4t.cfg --cmd "sign" --skipuid --cfg flash.xml --uphy_config tegra194-mb1-uphy-lane-p2888-0000-p2822-0000.cfg --device_config tegra19x-mb1-bct-device-sdmmc.cfg --misc_config tegra194-mb1-bct-misc-flash.cfg --misc_cold_boot_config tegra194-mb1-bct-misc-l4t.cfg --pinmux_config tegra19x-mb1-pinmux-brla3.cfg --gpioint_config tegra194-mb1-bct-gpioint-p2888-0000-p2822-0000.cfg --pmic_config tegra194-mb1-bct-pmic-p2888-0001-a04-E-0-p2822-0000.cfg --pmc_config tegra19x-mb1-padvoltage-brla3.cfg --prod_config tegra19x-mb1-prod-p2888-0000-p2822-0000.cfg --scr_config tegra194-mb1-bct-scr-cbb-mini.cfg --scr_cold_boot_config tegra194-mb1-bct-scr-cbb-mini.cfg --br_cmd_config tegra194-mb1-bct-reset-p2888-0000-p2822-0000.cfg --dev_params tegra194-br-bct-sdmmc.cfg --bins "mb2_bootloader nvtboot_recovery_t194.bin; mts_preboot preboot_c10_prod_cr.bin; mts_mce mce_c10_prod_cr.bin; mts_proper mts_c10_prod_cr.bin; bpmp_fw bpmp_t194.bin; bpmp_fw_dtb tegra194-a02-bpmp-p2888-a04.dtb; spe_fw spe_t194.bin; tlk tos-trusty_t194.img; eks eks.img; bootloader_dtb tegra194-brla3-dp1.dtb"
| Welcome to Tegra Flash
| version 1.0.0
| Type ? or help for help and q or quit to exit
| Use ! to execute system commands
| 
| [   0.0084 ] tegrasign_v2 --key None --getmode mode.txt
| [   0.0095 ] Assuming zero filled SBK key
| [   0.0097 ]
| [   0.0162 ] Generating RCM messages
| [   0.0175 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
| [   0.0187 ] Header already present for mb1_t194_prod.bin
| [   0.0223 ]
| [   0.0236 ] tegrasign_v2 --key None --getmode mode.txt
| [   0.0247 ] Assuming zero filled SBK key
| [   0.0249 ]
| [   0.0261 ] tegrasign_v2 --key None --file mb1_t194_prod_sigheader.bin --offset 2960 --length 1136 --pubkeyhash pub_key.key
| [   0.0271 ] Assuming zero filled SBK key
| [   0.0276 ]
| [   0.0288 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mb1_t194_prod_sigheader.bin mb1_t194_prod_sigheader.hash zerosbk
| [   0.0338 ]
| [   0.0353 ] tegrabct_v2 --chip 0x19 0 --sfuse tegra194-mb1-soft-fuses-l4t.cfg sfuse.bin
| [   0.0367 ]
| [   0.0379 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x19 0 --sfuses sfuse.bin --download rcm mb1_t194_prod_sigheader.bin 0 0
| [   0.0389 ] RCM 0 is saved as rcm_0.rcm
| [   0.0436 ] RCM 1 is saved as rcm_1.rcm
| [   0.0438 ] RCM 2 is saved as rcm_2.rcm
| [   0.0438 ] List of rcm files are saved in rcm_list.xml
| [   0.0438 ]
| [   0.0438 ] Signing RCM messages
| [   0.0450 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key --getmontgomeryvalues montgomery.bin
| [   0.0461 ] Assuming zero filled SBK key
| [   0.0468 ]
| [   0.0468 ] Copying signature to RCM mesages
| [   0.0481 ] tegrarcm_v2 --chip 0x19 0 --updatesig rcm_list_signed.xml
| [   0.0501 ]
| [   0.0501 ] Parsing partition layout
| [   0.0513 ] tegraparser_v2 --pt flash.xml.tmp
| [   0.0534 ]
| [   0.0534 ] Creating list of images to be signed
| [   0.0546 ] tegrahost_v2 --chip 0x19 0 --partitionlayout flash.xml.bin --list images_list.xml zerosbk
| [   0.0561 ] MB1: Nvheader already present is mb1_t194_prod.bin
| [   0.0567 ] Header already present for mb1_t194_prod_sigheader.bin
| [   0.0576 ] MB1: Nvheader already present is mb1_t194_prod.bin
| [   0.0614 ] Header already present for mb1_t194_prod_sigheader.bin
| [   0.0625 ] adding BCH for spe_t194.bin
| [   0.0659 ] adding BCH for spe_t194.bin
| [   0.0687 ] adding BCH for nvtboot_t194.bin
| [   0.0720 ] adding BCH for nvtboot_t194.bin
| [   0.0765 ] Header already present for preboot_c10_prod_cr.bin
| [   0.0813 ] Header already present for preboot_c10_prod_cr.bin
| [   0.0815 ] Header already present for mce_c10_prod_cr.bin
| [   0.0818 ] Header already present for mce_c10_prod_cr.bin
| [   0.0838 ] adding BCH for mts_c10_prod_cr.bin
| [   0.0883 ] adding BCH for mts_c10_prod_cr.bin
| [   0.1895 ] adding BCH for cboot_t194.bin
| [   0.2838 ] adding BCH for cboot_t194.bin
| [   0.2955 ] Stat for tegra194-brla3-dp1.dtb failed
| [   0.3070 ]
[... 110 lines cut ...]
| [   0.0493 ] tegraparser_v2 --pt flash.xml.tmp
| [   0.0514 ]
| [   0.0515 ] Creating list of images to be signed
| [   0.0527 ] tegrahost_v2 --chip 0x19 0 --partitionlayout flash.xml.bin --list images_list.xml zerosbk
| [   0.0553 ] MB1: Nvheader already present is mb1_t194_prod.bin
| [   0.0575 ] Header already present for mb1_t194_prod_sigheader.bin
| [   0.0597 ] MB1: Nvheader already present is mb1_t194_prod.bin
| [   0.0616 ] Header already present for mb1_t194_prod_sigheader.bin
| [   0.0635 ] adding BCH for spe_t194.bin
| [   0.0650 ] adding BCH for spe_t194.bin
| [   0.0676 ] adding BCH for nvtboot_t194.bin
| [   0.0705 ] adding BCH for nvtboot_t194.bin
| [   0.0752 ] Header already present for preboot_c10_prod_cr.bin
| [   0.0803 ] Header already present for preboot_c10_prod_cr.bin
| [   0.0822 ] Header already present for mce_c10_prod_cr.bin
| [   0.0841 ] Header already present for mce_c10_prod_cr.bin
| [   0.0860 ] adding BCH for mts_c10_prod_cr.bin
| [   0.0874 ] adding BCH for mts_c10_prod_cr.bin
| [   0.1856 ] adding BCH for cboot_t194.bin
| [   0.2821 ] adding BCH for cboot_t194.bin
| [   0.2942 ] Stat for tegra194-brla3-dp1.dtb failed
| [   0.3051 ]
| Error: Return value 4`

The  kernel-bup-payload task is far beyond my knowledge and Im totaly lost with this error.

madisongh, 2021-01-29:
I finally got around to trying a test build myself yesterday, and ran into a similar issue (for jetson-tx2). So yeah, we're not quite there yet. Thanks for the report.  When we have R32.5-based branches that are ready for general use, I'll update the wiki and post an announcement in our Slack channel.

madisongh, 2021-01-29:
I tracked down this problem and have fixed it in the [wip-dunfell-l4t-r32.5.0](https://github.com/OE4T/meta-tegra/tree/wip-dunfell-l4t-r32.5.0) branch (3192c23d890e3148b4fb8f766d870ac94554f74d).

closed 2021-01-29 by madisongh