OE4T/meta-tegra #562: Build error of core-image-minimal for Jetson TX2

chandshiv, 2021-02-08:
Hi,

I am trying to build core-image-minimal for Jetson TX2. My build configuration is

Build Configuration:		
BB_VERSION	=	"1.46.0"
BUILD_SYS	=	"x86_64-linux"
NATIVELSBSTRING	=	"ubuntu-18.04"
TARGET_SYS	=	"aarch64-oe4t-linux"
MACHINE	=	"jetson-tx2"
DISTRO	=	"tegrademo"
DISTRO_VERSION	=	"3.1+snapshot-20210208"
TUNE_FEATURES	=	"aarch64 armv8a crc"
TARGET_FPU	=	""
meta	=	"HEAD:6e89d668246fb37b2217aae7ae57390e793696d8"
meta-tegra		
contrib	=	"HEAD:3b4df1ac05e9f96e0363630c036f5445800af435"
meta-tegra-support	=	"dunfell-l4t-r32.4.3:63f3bf90e9c40428ec6ba4b30b57a8596faf5a56"
meta-oe		
meta-python		
meta-networking		
meta-filesystems	=	"HEAD:de37512b25c1f8c6bb6ab2b3782ac0fe01443483"
meta-tegrademo	=	"dunfell-l4t-r32.4.3:63f3bf90e9c40428ec6ba4b30b57a8596faf5a56"
meta-virtualization	=	"HEAD:92cd3467502bd27b98a76862ca6525ce425a8479"
meta-rust	=	"common-rust-native:3b783652cc9e0e0ff95fdf151eab3e178b5f1667"
meta-iotedge	=	"dunfell:6840cf2afb51b4f2c535a05d5635736652f664f7"

But I got this below error related to tegra-minimal-initramfs recipe.
The problem is that I have built the same image before with the same packages and I didn't get this error.
[2021-02-08T13:00:28.598Z] ERROR: tegra-minimal-initramfs-1.0-r0 do_image_complete: Error executing a python function in exec_python_func() autogenerated:
[2021-02-08T13:00:28.598Z] 
[2021-02-08T13:00:28.598Z] The stack trace of python calls that resulted in this exception/failure was:
[2021-02-08T13:00:28.598Z] File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
[2021-02-08T13:00:28.598Z]      0001:
[2021-02-08T13:00:28.598Z]  *** 0002:sstate_report_unihash(d)
[2021-02-08T13:00:28.598Z]      0003:
[2021-02-08T13:00:28.598Z] File: '/home/jenkins/workspace/dges_yoctoazureiot_jetson_master/tegra-demo-distro/layers/meta/classes/sstate.bbclass', lineno: 844, function: sstate_report_unihash
[2021-02-08T13:00:28.598Z]      0840:    report_unihash = getattr(bb.parse.siggen, 'report_unihash', None)
[2021-02-08T13:00:28.598Z]      0841:
[2021-02-08T13:00:28.598Z]      0842:    if report_unihash:
[2021-02-08T13:00:28.598Z]      0843:        ss = sstate_state_fromvars(d)
[2021-02-08T13:00:28.598Z]  *** 0844:        report_unihash(os.getcwd(), ss['task'], d)
[2021-02-08T13:00:28.598Z]      0845:}
[2021-02-08T13:00:28.598Z]      0846:
[2021-02-08T13:00:28.598Z]      0847:#
[2021-02-08T13:00:28.598Z]      0848:# Shell function to decompress and prepare a package for installation
[2021-02-08T13:00:28.598Z] File: '/home/jenkins/workspace/dges_yoctoazureiot_jetson_master/tegra-demo-distro/repos/poky/bitbake/lib/bb/siggen.py', lineno: 555, function: report_unihash
[2021-02-08T13:00:28.598Z]      0551:
[2021-02-08T13:00:28.598Z]      0552:            if "." in self.method:
[2021-02-08T13:00:28.598Z]      0553:                (module, method) = self.method.rsplit('.', 1)
[2021-02-08T13:00:28.598Z]      0554:                locs['method'] = getattr(importlib.import_module(module), method)
[2021-02-08T13:00:28.598Z]  *** 0555:                outhash = bb.utils.better_eval('method(path, sigfile, task, d)', locs)
[2021-02-08T13:00:28.598Z]      0556:            else:
[2021-02-08T13:00:28.598Z]      0557:                outhash = bb.utils.better_eval(self.method + '(path, sigfile, task, d)', locs)
[2021-02-08T13:00:28.598Z]      0558:
[2021-02-08T13:00:28.598Z]      0559:            try:
[2021-02-08T13:00:28.598Z] File: '/home/jenkins/workspace/dges_yoctoazureiot_jetson_master/tegra-demo-distro/repos/poky/bitbake/lib/bb/utils.py', lineno: 420, function: better_eval
[2021-02-08T13:00:28.598Z]      0416:    if extraglobals:
[2021-02-08T13:00:28.598Z]      0417:        ctx = copy.copy(ctx)
[2021-02-08T13:00:28.598Z]      0418:        for g in extraglobals:
[2021-02-08T13:00:28.598Z]      0419:            ctx[g] = extraglobals[g]
[2021-02-08T13:00:28.598Z]  *** 0420:    return eval(source, ctx, locals)
[2021-02-08T13:00:28.598Z]      0421:
[2021-02-08T13:00:28.598Z]      0422:@contextmanager
[2021-02-08T13:00:28.598Z]      0423:def fileslocked(files):
[2021-02-08T13:00:28.598Z]      0424:    """Context manager for locking and unlocking file locks."""
[2021-02-08T13:00:28.598Z] File: '<string>', lineno: 1, function: <module>
[2021-02-08T13:00:28.598Z]   File "<string>", line 1, in <module>
[2021-02-08T13:00:28.598Z] 
[2021-02-08T13:00:28.598Z] File: '/home/jenkins/workspace/dges_yoctoazureiot_jetson_master/tegra-demo-distro/layers/meta/lib/oe/sstatesig.py', lineno: 593, function: OEOuthashBasic
[2021-02-08T13:00:28.598Z]      0589:
[2021-02-08T13:00:28.598Z]      0590:                update_hash("\n")
[2021-02-08T13:00:28.598Z]      0591:
[2021-02-08T13:00:28.598Z]      0592:            # Process this directory and all its child files
[2021-02-08T13:00:28.598Z]  *** 0593:            process(root)
[2021-02-08T13:00:28.598Z]      0594:            for f in files:
[2021-02-08T13:00:28.598Z]      0595:                if f == 'fixmepath':
[2021-02-08T13:00:28.598Z]      0596:                    continue
[2021-02-08T13:00:28.598Z]      0597:                process(os.path.join(root, f))
[2021-02-08T13:00:28.598Z] File: '/home/jenkins/workspace/dges_yoctoazureiot_jetson_master/tegra-demo-distro/layers/meta/lib/oe/sstatesig.py', lineno: 553, function: process
[2021-02-08T13:00:28.598Z]      0549:                    add_perm(stat.S_IXOTH, 'x')
[2021-02-08T13:00:28.598Z]      0550:
[2021-02-08T13:00:28.598Z]      0551:                if include_owners:
[2021-02-08T13:00:28.598Z]      0552:                    try:
[2021-02-08T13:00:28.598Z]  *** 0553:                        update_hash(" %10s" % pwd.getpwuid(s.st_uid).pw_name)
[2021-02-08T13:00:28.598Z]      0554:                        update_hash(" %10s" % grp.getgrgid(s.st_gid).gr_name)
[2021-02-08T13:00:28.598Z]      0555:                    except KeyError:
[2021-02-08T13:00:28.598Z]      0556:                        bb.warn("KeyError in %s" % path)
[2021-02-08T13:00:28.598Z]      0557:                        raise
[2021-02-08T13:00:28.598Z] Exception: KeyError: 'getpwuid(): uid not found: 1000'

Please let me know how do we fix it

andy-git-usage, 2021-02-08:
Hi,
I also have this problem when I would like to try to build Xavier-NX platform and error message like yours.
I have no idea how to solve this issue, if you could give some tips will be appreciated, thanks in advance.

============== build error message ===============
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-xavier-nx-devkit"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.5"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-tegra           = "wip-dunfell-l4t-r32.5.0:3a410d4229bcd13ecf726ac63efa8d76d230ad27"
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:977b7268bf4fd425cb86d4a57500350c9b829162"

Initialising tasks: 100% |####################################################################################################| Time: 0:00:01
Sstate summary: Wanted 61 Found 3 Missed 58 Current 1090 (4% match, 94% complete)
NOTE: Executing Tasks
WARNING: tegra-minimal-initramfs-1.0-r0 do_image_complete: KeyError in .
ERROR: tegra-minimal-initramfs-1.0-r0 do_image_complete: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:sstate_report_unihash(d)
     0003:
File: '/mnt/uSSD/tegra-yocto/poky-dunfell/meta/classes/sstate.bbclass', lineno: 844, function: sstate_report_unihash
     0840:    report_unihash = getattr(bb.parse.siggen, 'report_unihash', None)
     0841:
     0842:    if report_unihash:
     0843:        ss = sstate_state_fromvars(d)
 *** 0844:        report_unihash(os.getcwd(), ss['task'], d)
     0845:}
     0846:
     0847:#
     0848:# Shell function to decompress and prepare a package for installation
File: '/mnt/uSSD/tegra-yocto/poky-dunfell/bitbake/lib/bb/siggen.py', lineno: 555, function: report_unihash
     0551:
     0552:            if "." in self.method:
     0553:                (module, method) = self.method.rsplit('.', 1)
     0554:                locs['method'] = getattr(importlib.import_module(module), method)
 *** 0555:                outhash = bb.utils.better_eval('method(path, sigfile, task, d)', locs)
     0556:            else:
     0557:                outhash = bb.utils.better_eval(self.method + '(path, sigfile, task, d)', locs)
     0558:
     0559:            try:
File: '/mnt/uSSD/tegra-yocto/poky-dunfell/bitbake/lib/bb/utils.py', lineno: 420, function: better_eval
     0416:    if extraglobals:
     0417:        ctx = copy.copy(ctx)
     0418:        for g in extraglobals:
     0419:            ctx[g] = extraglobals[g]
 *** 0420:    return eval(source, ctx, locals)
     0421:
     0422:@contextmanager
     0423:def fileslocked(files):
     0424:    """Context manager for locking and unlocking file locks."""
File: '<string>', lineno: 1, function: <module>
  File "<string>", line 1, in <module>

File: '/mnt/uSSD/tegra-yocto/poky-dunfell/meta/lib/oe/sstatesig.py', lineno: 593, function: OEOuthashBasic
     0589:
     0590:                update_hash("\n")
     0591:
     0592:            # Process this directory and all its child files
 *** 0593:            process(root)
     0594:            for f in files:
     0595:                if f == 'fixmepath':
     0596:                    continue
     0597:                process(os.path.join(root, f))
File: '/mnt/uSSD/tegra-yocto/poky-dunfell/meta/lib/oe/sstatesig.py', lineno: 553, function: process
     0549:                    add_perm(stat.S_IXOTH, 'x')
     0550:
     0551:                if include_owners:
     0552:                    try:
 *** 0553:                        update_hash(" %10s" % pwd.getpwuid(s.st_uid).pw_name)
     0554:                        update_hash(" %10s" % grp.getgrgid(s.st_gid).gr_name)
     0555:                    except KeyError:
     0556:                        bb.warn("KeyError in %s" % path)
     0557:                        raise
Exception: KeyError: 'getpwuid(): uid not found: 1000'

ERROR: Logfile of failure stored in: /mnt/uSSD/tegra-yocto/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/tegra-minimal-initramfs/1.0-r0/temp/log.do_image_complete.335097
ERROR: Task (/mnt/uSSD/tegra-yocto/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_image_complete) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3024 tasks of which 2885 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /mnt/uSSD/tegra-yocto/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_image_complete
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
================== message end ====================

madisongh, 2021-02-09:
I'm not sure what's going on there, but it looks like there's some confusion over file ownership that's causing a problem.  I've seen reports of similar issues happening when trying builds in containers.  Is that what you are doing?

Was this a build from a clean workspace, or were you rebuilding in an existing workspace?  In some of the reports I've seen, cleaning the workspace (or at least doing a `bitbake -c cleanall` for the recipe that triggered the problem) has helped.

chandshiv, 2021-02-10:
> I'm not sure what's going on there, but it looks like there's some confusion over file ownership that's causing a problem. I've seen reports of similar issues happening when trying builds in [...]

Yes, right tried building in docker container.
We tired both with new workspace and existing workspace, getting same error. However did not try with bitbake -c cleanall.
After adding this setting TEGRA_INITRAMFS_INITRD = "0" to local.conf file, it is working fine. I am not sure with the implications of this on build.

madisongh, 2021-02-10:
> After adding this setting TEGRA_INITRAMFS_INITRD = "0" to local.conf file, it is working fine.
OK, interesting.  That leaves the initial ramdisk unbundled from the kernel.  There must be something in the minimal-initramfs recipe that, when it's bundled into the kernel, is triggering the problem.

> We tired both with new workspace and existing workspace, getting same error. However did not try with bitbake -c cleanall.
OK.  Were you reusing the `sstate-cache` between those builds, or using a prior build's shared state as an SSTATE_MIRROR?

When I get a chance, I'll see if I can put together a container-based build environment to run some experiments in, to see if I can reproduce.

pnewman-cti, 2021-02-10:
I have the same error. I am not in a container though, I am using a native Ubuntu 18.04. 

I am not a Yocto expert but my guess is that the pwd function is looking for a /etc/passwd file in one of the sysroots that doesn't have a uid 1000 instead of the hosts /etc/passwd.

pnewman-cti, 2021-02-10:
Attached is the log from running
`bitbake -DDD tegra-minimal-initramfs &> initramfs_error.log` 
[initramfs_error.log](https://github.com/OE4T/meta-tegra/files/5958526/initramfs_error.log)

lfdmn, 2021-02-10:
How do you create the users for the container? Did you use volume mapping?

On our host PCs, we have a ```buildsystem``` group created with gid 1500, a ```jenkins``` user created with uid and gid 150. We create matching user and group in the containers

```
...
# Create jenkins user and add it to sudoers
RUN \
addgroup --system --gid 150 jenkins && \
adduser --disabled-password --system --uid 150 --gid 150 --shell /bin/bash jenkins && \
echo "jenkins ALL = NOPASSWD: /usr/local/bin/nvflash, /bin/dd, /sbin/losetup, /bin/mount, /bin/umount, /sbin/resize2fs, /bin/rm" > /etc/sudoers.d/jenkins-sudo && \
chmod 0440 /etc/sudoers.d/jenkins-sudo && \
addgroup --system --gid 1500 buildsystem && \
usermod -a -G buildsystem jenkins
...
```
That way we can access the files from both host and container, to keep the sstate and download directories between builds without too much overhead, and have an nginx instance on the host serve those directories as local mirrors.

I think for Ubuntu based distro the first uid assigned is 1000 so it's probably this type of configuration issue.

lfdmn, 2021-02-10:
> Attached is the log from running
> `bitbake -DDD tegra-minimal-initramfs &> initramfs_error.log`
> [initramfs_error.log](https://github.com/OE4T/meta-tegra/files/5958526/initramfs_error.log)

Maybe try to print the files to see which one gives you trouble and could have non-existent group/owner

```python3
for f in files:
    print(f)
```

madisongh, 2021-02-10:
@pnewman-cti Thanks, it helps to know that it's not just container-based builds.

I'm guessing that there is some interaction between the unusual setup we have here of enabling shared state for the initramfs image (image recipes typically don't) and the new hashserv feature in OE-Core.  I disable hashserv in my builds, since it doesn't really work well for my CI environment, which would explain why I haven't seen this.

pnewman-cti, 2021-02-10:
[initramfs_error_after_clean.log](https://github.com/OE4T/meta-tegra/files/5958731/initramfs_error_after_clean.log)

here is the log after running cleanall then building. Probably more useful to you. 

How do you disable hashserv?

madisongh, 2021-02-10:
> How do you disable hashserv?

Set
```
BB_SIGNATURE_HANDLER = "OEBasicHash"
```
in your build configuration (e.g. in local.conf).  For poky-derived distros, the default is to use `OEEquivHash`, which enables the generation of output hashes for the hash equivalence server.

pnewman-cti, 2021-02-10:
> 
> 
> > How do you disable hashserv?
> 
> Set
> 
> ```
> BB_SIGNATURE_HANDLER = "OEBasicHash"
> ```
> 
> in your build configuration (e.g. in local.conf). For poky-derived distros, the default is to use `OEEquivHash`, which enables the generation of output hashes for the hash equivalence server.

Adding that did allow for it to build.

madisongh, 2021-02-10:
Turns out I already had a workaround  (7a8cf92) in place for this issue in master and gatesgarth.  I've gone ahead and back-ported it to the dunfell-based branches, and have updated the corresponding branches in tegra-demo-distro to pick up the change.

If you've been seeing this issue, try updating to latest.  It should work better.

closed 2021-02-25 by madisongh