OE4T/meta-tegra #570: kernel-bup-payload.bb:do_deploy fails since  32.5 in tegraflash_internal.py

hannes09, 2021-02-12:
The deploy task of kernel-bup-payload fails with 32.5:

`/meta-tegra/recipes-kernel/kernel-bup-payload/kernel-bup-payload.bb:do_deploy) failed with exit code '1'`

`| [   2.7480 ] Filling MB1 storage info
| [   2.7481 ] Generating br-bct
| [   2.7499 ] Performing cfg overlay
| [   2.7499 ] ['syslogic-xavier.cfg', 'syslogic-xavier-override.cfg']
| [   2.7503 ] sw_memcfg_overlay.pl -c syslogic-xavier.cfg -s syslogic-xavier-override.cfg -o /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/bup-payload/13714/tmpj_taxig31.cfg
| [   2.8179 ]
| [   2.8180 ] Updating dev and MSS params in BR BCT
| Traceback (most recent call last):
|   File "tegraflash.py", line 1296, in <module>
|     tegraflash_run_commands()
|   File "tegraflash.py", line 1155, in tegraflash_run_commands
|     interpreter.onecmd(command)
|   File "/home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/recipe-sysroot-native/usr/lib/python3.8/cmd.py", line 217, in onecmd
|     return func(arg)
|   File "tegraflash.py", line 667, in do_sign
|     tegraflash_sign(exports)
|   File "/home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/bup-payload/tegraflash_internal.py", line 1008, in tegraflash_sign
|     tegraflash_sign_images()
|   File "/home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/bup-payload/tegraflash_internal.py", line 2910, in tegraflash_sign_images
|     tegraflash_fill_mb1_storage_info()
|   File "/home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/bup-payload/tegraflash_internal.py", line 3112, in tegraflash_fill_mb1_storage_info
|     run_command(command)
|   File "/home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/bup-payload/tegraflash_internal.py", line 189, in run_command
|     info_print(' '.join(cmd))
| TypeError: sequence item 2: expected str instance, NoneType found
| WARNING: /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/temp/run.do_deploy.13129:262 exit 1 from './doflash.sh'
| WARNING: Backtrace (BB generated script):
| 	#1: tegraflash_custom_sign_bup, /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/temp/run.do_deploy.13129, line 262
| 	#2: oe_make_bup_payload, /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/temp/run.do_deploy.13129, line 251
| 	#3: do_deploy, /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/temp/run.do_deploy.13129, line 158
| 	#4: main, /home/hannes/ame/yocto/build/tmp/work/syslogic_xavier-oe4t-linux/kernel-bup-payload/1.0-r0/temp/run.do_deploy.13129, line 337
| 
| Backtrace (metadata-relative locations):
| 	#1: tegraflash_custom_sign_bup, /home/hannes/ame/yocto/sources/meta-tegra/classes/image_types_tegra.bbclass, line 126
| 	#2: oe_make_bup_payload, /home/hannes/ame/yocto/sources/meta-tegra/classes/image_types_tegra.bbclass, line 662
| 	#3: do_deploy, /home/hannes/ame/yocto/sources/meta-tegra/recipes-kernel/kernel-bup-payload/kernel-bup-payload.bb, line 25
ERROR: Task (/home/hannes/ame/yocto/sources/meta-tegra/recipes-kernel/kernel-bup-payload/kernel-bup-payload.bb:do_deploy) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6856 tasks of which 6843 didn't need to be rerun and 1 failed.`

madisongh, 2021-02-12:
It looks like you're building for a custom machine.  If you are upgrading from a prior BSP version, did you refresh your `tegra-flashvars` configuration for R32.5.0?  For the Xavier-based machines, there's a new MINRATCHET_CONFIG file that has to be specified there, and the UPHY_CONFIG setting (if it's an AGX Xavier) now has to include the `--uphy_config ` option in front of the cfg file name (that config used to be needed for Xavier NX as well, but was dropped in R32.5.0).

If you've got meta-tegra cloned locally, try:
```
$ git diff origin/dunfell-l4t-r32.4.3 origin/master -- recipes-bsp/tegra-binaries/tegra-flashvars
```
to see the differences in the config files for the development kit platforms, and make sure you've made corresponding changes for your custom machine. (Substitute whatever branch/BSP version corresponds to your pre-upgrade builds in the first parameter).  You might also repeat that for the `conf/machine` subdirectory, to see if there are other machine settings that have changed between BSP versions which you need to account for.

The other thing that you could try is to go to the working directory for `kernel-bup-payload`, modify the `tegraflash_internal.py` script that's there under `recipe-sysroot-native` to dump the contents of `cmd` at line 189 (with a simple `print()` function, for instance) and manually run `temp/run.do_deploy` to try and zero in on what's missing from that command.  Hopefully that will provide enough information to trace back to what might be missing from your machine configuration.

hannes09, 2021-02-12:
Yes, you are right. 
Updating my flashvar with `UPHY_CONFIG`  and `MINRATCHET_CONFIG ` solved the issue.
Thanks for the help.

closed 2021-02-16 by madisongh