OE4T/meta-tegra #573: [jetson-nano] spidev_test not working on L4T 32.4.4

acostach, 2021-02-12:
Hi,

Is there any known issue with spidev on 32.4.4 ? I've been running the [loopback](https://github.com/rm-hull/spidev-test/blob/master/README.md) test on pins 19 and 21 and I'm seeing no data, neither with the oscilloscope:

```
root@jetson-nano-qspi-sd:/home# ./spidev_test -v -D /dev/spidev0.0 
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 kHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  |......@.........................|
RX | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................................|
```

I'm trying on the B01 sd-card Nano module + B01 devkit carrier board, with "demo-image-base" for jetson-nano-qspi-sd, generated using https://github.com/OE4T/tegra-demo-distro on the gatesgarth branch, at head revision f9676a72e0dca075b517b9916d5c61887900f2ea 

The spidev nodes are created with the default dtb from the raw partitions, and I've also tried using a dtb with spi1 enabled, generated by the jetson-io tool on the 32.4.4 sd-card image downloaded from nvidia's website: [spicustom.zip](https://github.com/OE4T/meta-tegra/files/5972570/spicustom.zip)

Looking at this comment https://github.com/OE4T/meta-tegra/issues/454#issuecomment-717532155 I see it was working on 32.4.2 - 32.4.3.

Thank you

madisongh, 2021-02-13:
It's working OK for me, but I patched the device tree to enable the SPI pins, rather than trying to use the overlay.  I suspect the overlay isn't working, and the pins aren't actually being configured for SPI use.

[0001-configure-SPI-pins.patch.txt](https://github.com/OE4T/meta-tegra/files/5975866/0001-configure-SPI-pins.patch.txt) is the patch I applied.

[pinctrl-check.txt](https://github.com/OE4T/meta-tegra/files/5975867/pinctrl-check.txt) is the output from the command `grep -C 4 spi1 /sys/kernel/debug/pinctrl/pinctrl-maps`.  Note the `function spi1` on each of the pins - if you see `rsvd1` instead, they're still configured as GPIOs.

After building the image and unpacking the tegraflash package, I used
```
$ FAB=400 ./dosdcard -y /dev/sdX
```
to write the SDcard image.  The FAB setting ensures that the `-b00` variant of the device tree is used, rather than the default `-a02` one (which I did not patch).

And to make sure the QSPI flash boot partitions were matching, I also put my Nano in recovery mode and used
```
$ ./doflash --spi-only
```
to re-flash those partitions, too.

With all that in place, this is the output I see with pins 19 and 21 looped:
```
root@jetson-nano-qspi-sd:~# spidev_test -v -D /dev/spidev0.0
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....�..................�.
RX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....�..................�.
```

acostach, 2021-02-15:
Thank you very much for the patch and the explanation @madisongh, with your patched and signed dtb in DTB and RP1 spidev works OK. But if it's only passed to u-boot trough extlinux FDT, the pinctrl-maps will report spi1 as expected, but spidev won't work. 

Is my understanding correct that the pinmux configuration is applied by cboot based on the dtb(s) in the raw partitions DTB / RP1?

And that u-boot 2020+ for tegra no longer applies any pinmux config from the dtb specified trough the FDT entry in extlinux.conf, like the uboot 2016.08 from nvidia sd-card image for 32.4.4 does?

madisongh, 2021-02-15:
>And that u-boot 2020+ for tegra no longer applies any pinmux config from the dtb specified trough the FDT entry in extlinux.conf, like the uboot 2016.08 from nvidia sd-card image for 32.4.4 does?

I did port in the DTB-merging code from L4T.  It's possible I got something wrong, though, and not being a big user of stock L4T myself, I'm not exactly sure how it's supposed to work.

acostach, 2021-02-15:
Thanks a lot @madisongh , I'll try to look further trough the upstream patches. Thought this one https://nv-tegra.nvidia.com/gitweb/?p=3rdparty/u-boot.git;a=commitdiff;h=3751aef823129de092caec658632e719c94dd869 seemed relevant. Doesn't solve the problem, but the pinctrl-maps state switched from "state 2" to "state unused" with FDT dtb. In case you happen to stumble on anything relevant, please let me know.

closed 2021-02-15 by acostach