OE4T/meta-tegra #589: Not able to clone the meta-layers

vaishnavils97, 2021-02-18:
Hi,

I have followed the link https://hub.mender.io/t/nvidia-tegra-jetson-tx2/123 to build Yocto image for jetson tx2. But getting errors while cloning the sub-layers. Below is the error log. As per my company security policy we should not add the git port "9418" to the rule book. So while trying to clone from git repos, clone is getting failed. Can you please suggest a solution that we could clone directly using "https" instead of "git".

fatal: unable to connect to git.openembedded.org: 
[2021-01-04T12:12:19.815Z] git.openembedded.org[0: 104.236.212.160]: errno=Connection timed out 
[2021-01-04T12:12:19.815Z] git.openembedded.org[1: 2604:a880:800:10::2127:2001]: errno=Cannot assign requested address 
fatal: clone of 'git://git.openembedded.org/meta-openembedded' into submodule path /home/tegra-demo-distro/repos/meta-openembedded' failed

madisongh, 2021-02-18:
[This StackOverflow article](https://stackoverflow.com/questions/4891527/git-protocol-blocked-by-company-how-can-i-get-around-that) has a pretty good description of how to handle this.  Just note that the repos hosted at yoctoproject.org use a different path for https vs. git protocols, so the rewrite would be `url."https://git.yoctoproject.org/git".insteadOf "git://git.yoctoproject.org"` to account for the difference.

vaishnavils97, 2021-02-18:
@madisongh  I have replaced the git with https but getting the below error /tegra-demo-distro/repos/poky/bitbake/lib/bb/fetch2/__init__.py", line 894, in runfetchcmd(cmd='export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export PATH="/home/tegra-demo-distro/layers/scripts:/home/tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux:/home/jenkins/workspace/IoT_cartridges_jetson_dev_master/tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot/usr/bin/crossscripts:/home/tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/sbin:/home/tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/usr/bin:/home/tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/sbin:/home//tegra-demo-distro/build/tmp/work/armv8a_tegra-oe4t-linux/eglstreams-kms-example/fetcheravoidrecurse-fetcheravoidrecurse/recipe-sysroot-native/bin:/home//tegra-demo-distro/repos/poky/bitbake/bin:/home/tegra-demo-distro/build/tmp/hosttools"; export HOME="/home/jenkins"; git -c core.fsyncobjectfiles=0 ls-remote "git://github.com/madisongh/eglstreams-kms-example" ', d=<bb.data_smart.DataSmart object at 0x7f28a7195eb8>, quiet=True, cleanup=[], log=None, workdir=None):
22:06:11       
22:06:11      >        raise FetchError(error_message)
22:06:11       

And I have manually changed this file git:// to https:// but still the issue remains same.

DESCRIPTION = "Simple example program demonstrating the use of EGLStreams with DRM KMS."
HOMEPAGE = **"https://github.com/aritger**"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://COPYING;md5=2c4ffb74754562207b887a66a37aad6c"

DEPENDS = "libdrm virtual/egl"

SRC_REPO ?= "github.com/madisongh/eglstreams-kms-example"
SRCBRANCH ?= "master"
SRC_URI = **"https://${SRC_REPO};branch=${SRCBRANCH}"**
SRCREV = "${AUTOREV}"
PV = "1.0+git${SRCPV}"

COMPATIBLE_MACHINE = "(tegra)"
PACKAGE_ARCH_tegra = "${TEGRA_PKGARCH}"

S = "${WORKDIR}/git

madisongh, 2021-02-18:
> SRC_URI = "https://${SRC_REPO};branch=${SRCBRANCH}"

No, that's not the right change.  For the SRC_URI, you still need to specify `git://` at the front, so the fetcher knows it's a git repository.  The right change would be to add `;protocol=https` to have the git fetcher use that protocol instead of git.  So you could change the SRC_REPO setting like this:
```
SRC_REPO ?= "github.com/madisongh/eglstreams-kms-example;protocol=https"
```
instead of the SRC_URI change you made.

Again, though, you should be able to use git configuration to do the rewrites under the hood, without having to change any recipes.  Since there are likely to be quite a few recipes that just use the git protocol by default, working out the configuration-based rewrites (or convincing your IT folks that git fetches are a necessity, so they unblock outbound access on the git port) would probably work better.

ichergui, 2021-03-01:
Hey @vaishnavils97 

Could you please give us an update ? Are you able to clone the `Yocto meta layer` without issue ?
Please let me know if I can help.

ichergui, 2021-03-28:
No updates here, I will close this issue.

closed 2021-03-28 by ichergui