OE4T/meta-tegra #596: Error when flashing jetson-xavier-nx-devkit-emmc

TheophileBlard, 2021-02-23:
**Describe the bug**
Hi! I have a custom image based on `meta-tegra` & `tegra-demo-distro` (master branch) which recently stopped working for `jetson-xavier-nx-devkit-emmc`. It is still working for `jetson-agx-xavier-devkit` (formerly `jetson-xavier`). There is no build error, but flashing fails wit the following error:
```shell
...
[   1.6421 ] Packing sdram param for instance[13]
[   1.6421 ] Packing sdram param for instance[14]
[   1.6421 ] Packing sdram param for instance[15]
[   1.6421 ] 
[   1.6422 ] Reading ramcode from backup chip_info.bin file
[   1.6422 ] RAMCODE Read from Device: 0

[   1.6423 ] Generating BPMP dtb for ramcode - 0
Error: Could not find dtc
``` 

[Full trace](https://pastebin.com/U97TdB7E)

madisongh, 2021-02-23:
Yeah, the flashing tools in L4T R32.5.0 require the device tree compiler for some of the platforms (might just be the Xavier NX ones).  If you're on a Ubuntu system, just `sudo apt install device-tree-compiler`.

TheophileBlard, 2021-02-23:
Thank you so much, it works now ! Did I miss it in the wiki or smth ?

madisongh, 2021-02-23:
No, it was something I ran into myself, then forgot to write down.  I've updated the [flashing](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) page on the wiki with a note about it.

closed 2021-02-26 by madisongh