OE4T/meta-tegra #607: Nothing RPROVIDES 'ltrace'

wgthompson, 2021-03-01:
When running
`bitbake core-image-sato`

I receive following errors:
```
ERROR: Nothing RPROVIDES 'ltrace' (but /home/wgthompson/yocto-tegra/poky-dunfell/meta/recipes-sato/images/core-image-sato.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'ltrace' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['ltrace']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'ltrace']
```
Running for "jetson-nano-devkit-emmc"

Can provide my local.conf file if uesful.

madisongh, 2021-03-01:
Do you want the ltrace package in your build? If so, you'll need to make sure you've made the recipe available. From [this layer database query](http://layers.openembedded.org/layerindex/branch/master/recipes/?q=ltrace) it looks like it's available in meta-oe.

If you don't want it in your build, then you'll need to figure out where that dependency is coming from to resolve the problem. There's nothing in OE-Core or meta-tegra that mention ltrace at all, so what other layers are you including in your build?  The `Build Configuration:` output you get when running bitbake, which lists the layers and their git revisions, would help, but you could probably find it yourself more easily by `grep`ping through the recipes in your build workspace to look for any references to it.

ichergui, 2021-03-01:
Hey @wgthompson 

Could you please share the branch are you using ?  are you using [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) ?  if no, please try it by using our stable branch `dunfell-l4t-r32.4.3`  

**Note:** you can use the following command `$ bitbake demo-image-sato`

Please let us know if you need support.

ichergui, 2021-03-13:
Hey @wgthompson 

Could you please give us updates concerning your issue ?

wgthompson, 2021-03-15:
Thanks guys, I am not building the demo distro, I was trying to build the sato image and I was running up against it, but I successfully built the minimal image with no issue. Still not sure where ltrace was coming from.

closed 2021-03-15 by wgthompson