OE4T/meta-tegra #618: RS32.5 nvvidconv patch breaking NVMM->non NVMM conversions

sweng27518, 2021-03-08:
**Describe the bug**
After pulling in dunfell-l4t-r32.5.0, pipelines using nvvidconv that convert NVMM memory to non-NVMM memory are no longer working. After some investigation, removing this patch: 0002-Use-filter-function-for-fixating-caps.patch, fixes the issue and conversions from NVMM to non-NVMM go back to working.

**To Reproduce**
gst-launch-1.0 nvarguscamerasrc  ! 'video/x-raw(memory:NVMM),format=(string)NV12'  ! nvvidconv ! 'video/x-raw,format=(string)NV12' ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed
Setting pipeline to PLAYING ...

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.305: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.306: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.306: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.306: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed
New clock: GstSystemClock

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.995: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_has_field: assertion 'structure != NULL' failed
ERROR: from element /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0: Internal data stream error.

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.996: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed
Additional debug info:
../gstreamer-1.16.2/libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.690824992

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_object_unref: assertion '((GObject *) object)->ref_count > 0' failed
Setting pipeline to PAUSED ...

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_caps_get_structure: assertion 'index < GST_CAPS_LEN (caps)' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_structure_get_value: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_structure_get_string: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_structure_has_field: assertion 'structure != NULL' failed

(gst-launch-1.0:3870): GStreamer-CRITICAL **: 16:27:21.997: gst_structure_fixate_field_string: assertion 'gst_structure_has_field (structure, field_name)' failed
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

**Additional context**
Not seeing this issue on dunfell-l4t-r32.4.3

madisongh, 2021-03-08:
Are you using the latest on the dunfell-l4t-r32.5.0 branch?  I reworked that patch over the weekend (fdac5902) due to a similar report that came in on the Slack channel.

sweng27518, 2021-03-08:
Ah, no, definitely would not have had something from this weekend. We started pulling r32.5.0 a week or so ago. I'll retest with the latest.

sweng27518, 2021-03-08:
Thanks @madisongh! It does look like that fixes my problem as well. I'll do some more testing tomorrow to be sure.