OE4T/meta-tegra #637: SD card image for Jetson Xavier AGX

chhg82, 2021-03-22:
How can I generate an SD card image on dunfell-l4t-r32.4.3 for xavier agx?

madisongh, 2021-03-22:
AGX Xavier doesn't come with an SDcard slot.  If you mean Xavier NX, there's some info on [this wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit).

chhg82, 2021-03-22:
I use agx xavier developer kit. You're right, on the som is no slot but on the evalboard...

madisongh, 2021-03-22:
Well, so there is... I've had mine for a long time, and I never noticed that before.  I'm not sure you can boot from it, though.  I've never seen anything in L4T that supports such a setup on the AGX.

chhg82, 2021-03-23:
With nvidia sdk i got it running with: sudo ./flash.sh jetson-xavier mmcblk1p1. But how can I do it with meta-tegra?

ichergui, 2021-03-23:
Hey @chhg82 @madisongh

I confirm that the `Jetson Xavier AGX` have a SDcard slot. I didn't try to boot by using the SDcard tbh.

[image]

Here is some interesting links:

From Nvidia Dev talk [does-jetson-agx-xavier-support-boot-from-sd](https://forums.developer.nvidia.com/t/does-jetson-agx-xavier-support-boot-from-sd/65872)

[Flashing Jetson Xavier AGX](https://developer.ridgerun.com/wiki/index.php?title=Xavier/Flashing_the_Board)

I will experiment this and I will let you know.

chhg82, 2021-03-26:
Hi @ichergui 

Thanks for experimenting! Any news about it?

ichergui, 2021-03-26:
Sorry @chhg82 
I was quite busy these days. I will try later and give a feedback.
I will double check with @madisongh if script `make-sdcard` provide by tegraflash package make our lifre easy to create the right SDcard to boot with.

koansoftware, 2021-03-26:
I did it on June 2020 and now I don't have the board here, so I hope to remember correctly.
[NVIDIA Jetson AGX Xavier with Yocto Project](https://koansoftware.com/yocto-project-on-nvidia-jetson-agx-xavier/)

**Flashing Linux on the eMMC**
Enable recovery mode, then run
`sudo ./flash.sh jetson-xavier mmcblk0p1`

**Flashing Linux on the microSD**
Enable recovery mode, then run
`sudo ./flash.sh jetson-xavier mmcblk1p1`

ichergui, 2021-03-26:
Thanks @koansoftware

The `meta-tegra` layer provides scripts `doflash.sh` and `dosdcard.sh` for Jetsons platform that have only eMMC (production devices) and  the SDcard (Development kit) respectively.  The Jetson Xavier AGX devkit is an exception which have both (eMMC and SDcard).

The Jetson Xavier AGX tegraflash package include `doflash.sh` and `make-sdcard` scripts, I will double check with @madisongh concerning the SDcard flashing process.

madisongh, 2021-03-26:
I spent some time looking at this today.  Even stock L4T doesn't really support SDcard booting for the Xavier "out of the box", and even the not-very-clear instructions I've seen in various forum posts and on eLinux aren't working for me, at least with recent versions of L4T.

Now as for OE/Yocto builds... you can easily append ` root=/dev/mmcblk1p1` to the KERNEL_ARGS setting to direct the kernel to use the SDcard for the rootfs.  We don't implement the `extlinux.conf` setup that they have in L4T on the cboot-based platforms, but adding support for it would probably make it easier to handle SDcard booting. I'm not sure yet whether we could get it to work without it, but maybe.

In any case, there's some more work involved for this.

madisongh, 2021-03-27:
I have managed to put together some changes to split the kernel/DTB/rootfs partitions out to an SDcard, and it appears to work OK for simple use cases, at least. PR #643 has the changes.  I've tried them with a build off master (R32.5.1), and patched them over to dunfell-l4t-r32.4.3 (they patch cleanly). If folks who need this functionality could give them a try, I'd appreciate any feedback.

ichergui, 2021-03-28:
Hey @madisongh 
Thanks for the changes, I have a quick question:
How `cboot` will deal when SDcard is plugged ? Is the booting sequence automatically try the SDcard then eMMC ? 
Have we to make updates in `KERNEL_ARGS` (root=`/dev/mmcblk1p1`).

I'm trying it without any changes in the `KERNEL_ARGS`, I will let you know.

madisongh, 2021-03-28:
Yes, by default cboot checks the SDcard before the eMMC when loading the kernel.  And the default initramfs we provide looks for the APP partition by name, so you don't need the `root=` in KERNEL_ARGS.

ichergui, 2021-03-28:
Thanks @madisongh to make this available, I tested it and it works fine to me.
Well done

closed 2021-03-28 by madisongh