OE4T/meta-tegra #647: CMake can't find CUDA in the populated sdk (when cross compiling using the sdk)

fei4xu, 2021-04-02:
**Describe the bug**
I'm bitbaking an image and have populated the SDK, install the SDK, everything is ok.
When I am using the SDK to do cross compiling, cmake has an error of can't find cuda.

**To Reproduce**
Steps to reproduce the behavior:
1. in image.bb
```bb
IMAGE_INSTALL_append = "	opencv cuda-toolkit"
```
2. Build with bitbake argument '-c populate_sdk'
3. install the sdk
4. check the default sdk installation folder:
`/opt/poky/3.1.6/sysroots/aarch64-poky-linux/usr/local/cuda-10.2/bin`
`nvcc` is there
5. Create a new CMake file, in it, use `OpenCV` (no direct usage of `CUDA`)
6. cmake the project, there is an error:

```sh
CMake Error at /opt/poky/3.1.6/sysroots/x86_64-pokysdk-linux/usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE
  CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (Required is exact version "10.2")
Call Stack (most recent call first):
  /opt/poky/3.1.6/sysroots/x86_64-pokysdk-linux/usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /opt/poky/3.1.6/sysroots/x86_64-pokysdk-linux/usr/share/cmake-3.16/Modules/FindCUDA.cmake:1104 (find_package_handle_standard_args)
  /opt/poky/3.1.6/sysroots/aarch64-poky-linux/usr/lib/cmake/opencv4/OpenCVConfig.cmake:86 (find_package)
  /opt/poky/3.1.6/sysroots/aarch64-poky-linux/usr/lib/cmake/opencv4/OpenCVConfig.cmake:105 (find_host_package)
  CMakeLists.txt:19 (find_package)

```
**Additional context**

I'm not sure if `/opt/poky/3.1.6/sysroots/aarch64-poky-linux/usr/local/cuda-10.2/bin` should be `/opt/poky/3.1.6/sysroots/aarch64-poky-linux/usr/local/cuda/bin` (without the version number suffix)

ichergui, 2021-04-02:
Hey @fei4xu 
Are you using the [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) ? if so, which branch ?
I think you are using `dunfell` but just to be sure.
I will try to reproduce the issue.

fei4xu, 2021-04-03:
> Hey @fei4xu
> Are you using the [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) ? if so, which branch ?
> I think you are using `dunfell` but just to be sure.
> I will try to reproduce the issue.

I'm not using `tegra-demo-distro` but I'm creating my own image.
The `meta-tegra` branch is `dunfell-l4t-r32.4.3` (for JetPack 4.4)

Inside it, OpenCV is 4.1.0 (but I think in JetPack 4.4's document, OpenCV should be 4.1.1)

madisongh, 2021-04-03:
I have a project that uses OpenCV and builds against the SDK using CMake, but it uses:
```
find_package(PkgConfig REQUIRED)
pkg_check_modules(OPENCV REQUIRED IMPORTED_TARGET opencv4)
```
and then uses the `PkgConfig::OPENCV` target to reference the libraries. I'm not sure I've tested whether OpenCV's find module works in that environment; it might need some patching, or some additional variable settings, to get it to handle the native/target split we have for CUDA in the SDK.

madisongh, 2021-04-04:
OK, I ran a quick test and here is the missing piece:
```
TOOLCHAIN_HOST_TASK_append = " nativesdk-packagegroup-cuda-sdk-host"
```
Including that packagegroup on the host side of the SDK will put the necessary bits in place for `FindCUDA.cmake` to work.

I found I also had to add:
```
SDKIMAGE_INSTALL_COMPLEMENTARY_append = " opencv-staticdev"
```
because the OpenCV CMake module looks for some libraries that are part of that package, but `-staticdev` packages aren't included in my SDK builds by default. You may have to do the same.  That's not tegra-specific.

fei4xu, 2021-04-10:
`nativesdk-packagegroup-cuda-sdk-host` fixed my issue of the generated SDK.

fei4xu, 2021-04-10:
Before I close this issue, I have another very close issue:
if I directly `bitbake my-app` (`my-app` is a recipe I created for my application, it's a common cmake project's recipe, `my-app` will find_package of OpenCV then find_package of CUDA), it still has the same error, how can I fix it?

madisongh, 2021-04-10:
Just `inherit cuda` in the recipe to have everything set up.

closed 2021-04-10 by fei4xu

TheophileBlard, 2021-04-21:
The real question here is why `find_package(OpenCV)` tries to find the `nvcc` host compiler. Correct me if I'm wrong but even if OpenCV is compiled with CUDA code, then you do no need `nvcc` to link to it right ? I agree that OpenCV needs the cross-compiled cuda libraries (libcublas, libcudnn...) on the target-side, but why does it also search for host-side dependencies ?

madisongh, 2021-04-21:
@TheophileBlard OpenCV's Cmake find module (OpenCVConfig.cmake) is just calling on the CUDA find module when it has CUDA support configured into it.  The CUDA find module looks for both the host-side and target-side CUDA support (which includes the nvcc compiler). The CUDA SDK normally bundles them together; for OE/Yocto builds with split host and target sysroots, that complicates things a bit.

As I mentioned earlier, you can use the pkgconfig mechanism for locating OpenCV, and that should eliminate the need for the host-side CUDA tools.

TheophileBlard, 2021-04-22:
We would like to stick with modern CMake best-practices  and avoid `pkg-config` as much as possible. This solution is also not applicable for third party libraries relying on `find_package(OpenCV)`. But yeah this is more a `FindCUDA` issue, so I guess we'll have to stay with the `inherit cuda` workaround. Too bad they didn't make separate Compiler & Development components like in `FindPython3`.