OE4T/meta-tegra #650: libnvidia-container-tools fetch error in branch dunfell-l4t-r32.4.3

lumurillo, 2021-04-03:
**Describe the bug**
I am using the dunfell-l4t-r32.4.3 branch of meta tegra in the revision 387d97e0d66c6e1583f89606aa8f148b46830c4f. I created a Yocto manifest to reproduce the environment. I have been using this branch without errors, but recently I started a build from scratch, but this time I am seeing the following fetch error with the libnvidia-container-tools recipe:

```
WARNING: libnvidia-container-tools-0.9.0-r1 do_fetch: Failed to fetch URL git://github.com/NVIDIA/nvidia-modprobe.git;name=modprobe;destsuffix=git/deps/src/nvidia-modprobe-396.51, attempting MIRRORS if available
ERROR: libnvidia-container-tools-0.9.0-r1 do_fetch: Fetcher failure: Unable to find revision d97c08af5061f1516fb2e3a26508936f69d6d71d in branch master even from upstream
ERROR: libnvidia-container-tools-0.9.0-r1 do_fetch: Fetcher failure for URL: 'git://github.com/NVIDIA/nvidia-modprobe.git;name=modprobe;destsuffix=git/deps/src/nvidia-modprobe-396.51'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /media/lmurillo/258bcdab-a9a0-4390-ac53-f19781ebfb7f/lmurillo/work/zebra/yocto-tegra-08-10-2020/build-08-10-2020/tmp/work/aarch64-poky-linux/libnvidia-container-tools/0.9.0-r1/temp/log.do_fetch.29101
ERROR: Task (/media/lmurillo/258bcdab-a9a0-4390-ac53-f19781ebfb7f/lmurillo/work/zebra/yocto-tegra-08-10-2020/meta-tegra/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2136 tasks of which 2135 didn't need to be rerun and 1 failed.
​
Summary: 1 task failed:
  /media/lmurillo/258bcdab-a9a0-4390-ac53-f19781ebfb7f/lmurillo/work/zebra/yocto-tegra-08-10-2020/meta-tegra/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb:do_fetch
Summary: There was 1 WARNING message shown.
```

To solve it I did the following change to the recipe:

```
diff --git a/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb b/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb
index 7a34b29..a8f1200 100644
--- a/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb
+++ b/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb
@@ -32,7 +32,7 @@ ELF_TOOLCHAIN_VERSION = "0.7.1"
 LIBTIRPC_VERSION = "1.1.4"
 
 SRC_URI = "git://github.com/NVIDIA/libnvidia-container.git;name=libnvidia;branch=jetson \
-           git://github.com/NVIDIA/nvidia-modprobe.git;name=modprobe;destsuffix=git/deps/src/nvidia-modprobe-${NVIDIA_MODPROBE_VERSION} \
+           git://github.com/NVIDIA/nvidia-modprobe.git;name=modprobe;branch=main;destsuffix=git/deps/src/nvidia-modprobe-${NVIDIA_MODPROBE_VERSION} \
            file://0001-Makefile-Fix-RCP-flags-and-change-path-definitions-s.patch \
            file://0002-common.mk-Set-JETSON-variable-if-not-set-before.patch \
            file://0003-Fix-mapping-of-library-paths-for-jetson-mounts.patch \
```

Where I specify that the branch of the nvidia modprobe as **main**. Why is this failing now? Did something change in this repo?

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'dunfell-l4t-r32.4.3' (rev 387d97e0d66c6e1583f89606aa8f148b46830c4f) with `MACHINE` based on 'jetson-nano-emmc'
2. See error mentioned above while fetching nvidia modprobe source from libnvidia-container-tools recipe.

ichergui, 2021-04-04:
Hey @lumurillo 
Thanks for catching this.
I think that Nvidia recently changed the name branch from `master` to `main` because we tested this branch `dunfell-l4t-r32.4.3`. Actually, it is our stable branch.
I will let you know when I push the fix.
Thanks a gain

ichergui, 2021-04-04:
Hey @lumurillo 
I checked the `meta-tegra` (`dunfell-l4t-r32.4.3` branch) and the fix was introduced by @madisongh recently.
Here is the [fix](https://github.com/OE4T/meta-tegra/blob/dunfell-l4t-r32.4.3/external/virtualization-layer/recipes-containers/libnvidia-container-tools/libnvidia-container-tools_0.9.0.bb#L35).
Could you please update the meta-tegra to pick up the recent changes.

madisongh, 2021-04-05:
Yes, the fix for that repo's branch rename went in a couple of weeks ago: 6e2a2c1417e435468b5993bad231bcd60793f129

lumurillo, 2021-04-11:
With fix on 6e2a2c1417e435468b5993bad231bcd60793f129 now the build completes successfully. Due to the manifest that I created, I didn't get the fix, but I just updated it. Thanks!

closed 2021-04-11 by lumurillo

massicotjgab, 2021-04-14:
Hello, the same error is occured on branch dunfell-l4t-r32.4.2 and seems to not have been corrected by this kind of commit.
I switched on the branch dunfell-l4t-r32.4.3 for my work, but i think it could be usefull to update dunfell-l4t-r32.4.2 branch with the same modification.

massicotjgab, 2021-04-14:
I think i will use dunfell-l4t-r32.4.3 now. But i think you can introduce the fix anyway for other people :)
Thank you very much!

ichergui, 2021-04-14:
Hey @massicotjgab 
The fix was introduced to the branch `dunfell-l4t-r32.4.2`: [f0757704](https://github.com/OE4T/meta-tegra/commit/f0757704ce495878d778fc46ae5591de3e52cdb2)
Enjoy

massicotjgab, 2021-04-14:
Nice! Thank you very much! Have a nice day :)