OE4T/meta-tegra #654: docker: Error response from daemon: Unknown runtime specified nvidia.

RomainGsd, 2021-04-08:
Hello,
I am trying to run NVIDIA's l4t containers. I can pull an image and run it with 
`docker run --gpus all --rm -it nvcr.io/nvidia/l4t-base:r32.5.0 bash`

But if I add --runtime nvidia to docker options it result in this error :
`docker: Error response from daemon: Unknown runtime specified nvidia.`

I am running on a Jetson Xavier NX devkit on `dunfell` branch and I added these layers to my local.conf :
`CORE_IMAGE_EXTRA_INSTALL_append = " nvidia-docker cuda-libraries cuda-samples tensorrt cudnn libvisionworks"
CORE_IMAGE_EXTRA_INSTALL_append = " nvidia-container-runtime cudnn-container-csv tensorrt-container-csv libvisionworks-container-csv"
DISTRO_FEATURES_append = " ldconfig virtualization"`

First thought is that I missed something when configuring but I can't find anything in READMEs, closed/open issues or the wiki

madisongh, 2021-04-08:
Which dunfell branch?  The dunfell (with no version suffix) branch doesn't support Xavier NX, so it must be one of the others.  I ask because the L4T version of your built OS has to match the L4T version of the container you're trying to run.  So is it `dunfell-l4t-r32.5.0` ?

Any other layers in your build besides the ones mentioned on the [wiki page](https://github.com/OE4T/meta-tegra/wiki/NVIDIA-Container-Runtime-support)?

The additions you made look OK (although the `-container-csv` packages should get pulled in automatically without explicitly listing them). Is the docker daemon starting OK?  Can you provide its log output?

RomainGsd, 2021-04-09:
I'm on `dunfell-l4t-32.4.3`, therefore I just pulled `nvcr.io/nvidia/l4t-tensorflow:r32.4.3-tf2.2-py3` accordingly.

I also have 
`PREFERRED_VERSION_python3-docker = "4.3.1"`
`CORE_IMAGE_EXTRA_INSTALL_append = " python3-docker"`

docker log output in _/var/log/message.log_:
```
2021-04-09T09:05:57.655114+00:00 (none) root: dockerd has started.
2021-04-09T09:05:59.964011+00:00 (none) connmand[2689]: docker0 {create} index 7 type 1 <ETHER>
2021-04-09T09:05:59.964032+00:00 (none) connmand[2689]: docker0 {update} flags 4098 <DOWN>
2021-04-09T09:05:59.964045+00:00 (none) connmand[2689]: docker0 {newlink} index 7 address 02:42:3B:99:8A:65 mtu 1500
2021-04-09T09:05:59.964055+00:00 (none) connmand[2689]: docker0 {newlink} index 7 operstate 2 <DOWN>
2021-04-09T09:05:59.964361+00:00 (none) connmand[2689]: docker0 {add} address 172.17.0.1/16 label docker0 family 2
2021-04-09T09:06:00.014695+00:00 (none) connmand[2689]: docker0 {update} flags 4163 <UP,RUNNING>
2021-04-09T09:06:00.014721+00:00 (none) connmand[2689]: docker0 {newlink} index 7 address 02:42:3B:99:8A:65 mtu 1500
2021-04-09T09:06:00.014733+00:00 (none) connmand[2689]: docker0 {newlink} index 7 operstate 0 <UNKNOWN>
2021-04-09T09:06:00.015240+00:00 (none) connmand[2689]: docker0 {add} route 172.17.0.0 gw 0.0.0.0 scope 253 <LINK>
2021-04-09T09:06:00.017476+00:00 (none) kernel: [   21.481832] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
2021-04-09T09:06:00.901990+00:00 (none) connmand[2689]: docker0 {update} flags 4099 <UP>
2021-04-09T09:06:00.902021+00:00 (none) connmand[2689]: docker0 {newlink} index 7 address 02:42:3B:99:8A:65 mtu 1500
2021-04-09T09:06:00.902033+00:00 (none) connmand[2689]: docker0 {newlink} index 7 operstate 2 <DOWN>
```

madisongh, 2021-04-09:
All that looks OK.  You mentioned the l4t-base container in your OP, though, and now you're talking about the tensorflow container. Did you get the base container to work? Are you getting the same error as before?

I usually test with the base container and the DeepStream container, using the `demo-image-full` image built out of our [demo distro](https://github.com/OE4T/tegra-demo-distro), to verify that the container support is working. If I get a chance in the next couple of days, I'll try the tensorflow container.

arpit6232, 2021-04-09:
The following seemed to work for me on the following parameters
Board: `NVIDIA Jetson Nano (Developer Kit)`
Branch : `dunfell-l4t-r32.4.3`
Commit Hash : `d5348114b0a9acd42a473aa4542a19f27f3a2a2c`
Yocto Setup : . ./setup-env --machine jetson-nano-qspi-sd
NVIDIA Deepstream SDK Installed as part of the binaries under `/nvidia/downloads`
Command Run : `docker run --gpus all -it --runtime nvidia nvcr.io/nvidia/l4t-base"r32.5.0`

Results on the Jetson Nano 
[image]

ichergui, 2021-04-09:
Thanks @arpit6232  for sharing this

RomainGsd, 2021-04-09:
Is it a layer that is installing the nvidia runtime ? I guess it's `nvidia-container-runtime` but I can't debug why it isn't on my system

madisongh, 2021-04-09:
> Is it a layer that is installing the nvidia runtime ? I guess it's nvidia-container-runtime but I can't debug why it isn't on my system

It should get pulled in automatically if you install `nvidia-docker` in your image.

>I also got a warning during build from tegra-bootfiles_32.4.3.bb
update-boot-partition.sh: file could not be found

No idea where that's coming from.  There's no reference to any `update-boot-partition.sh` script in any of our recipes.  Are you including other layers in your build setup?

RomainGsd, 2021-04-09:
>Are you including other layers in your build setup?

Yes I'm building a system that was doing other things at first and now I try to integrate `meta-tegra`

I can see that for normal usage, nvidia-container-runtime adds /etc/docker/daemon.json with this inside :
```
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}
```
I don't have this file on my system so I created it but it is deleted when I reboot, I might look into that

dwalkes, 2021-04-09:
> I don't have this file on my system 

It should be added by `nvida-docker` I believe, see [this line](https://github.com/OE4T/meta-tegra/blob/dunfell-l4t-r32.4.3/external/virtualization-layer/recipes-containers/nvidia-docker/nvidia-docker_2.2.2.bb#L29), so it seems something is going wrong there.

madisongh, 2021-04-11:
> Yes I'm building a system that was doing other things at first and now I try to integrate meta-tegra

[...]

> I don't have this file on my system so I created it but it is deleted when I reboot, I might look into that

Double-check the `.manifest` file for the image you built (which should end up in`$BUILDDIR/tmp/deploy/images/${MACHINE}`, with the image) to ensure that `nvidia-docker`, `nvidia-container-runtime` and anything else you tried to add to your image actually got included. If they aren't there, then perhaps something in your layers isn't using `CORE_IMAGE_EXTRA_INSTALL`, or is overwriting the additions you've made.

RomainGsd, 2021-04-12:
> Double-check the .manifest file for the image you built to ensure that nvidia-docker, nvidia-container-runtime and anything else you tried to add to your image actually got included 
```
> cat sdtv-image-iot-dev-jetson-xavier-nx-devkit.manifest | grep nvidia
libnvidia-container-tools aarch64 0.9.0-r1
libnvidia-egl-wayland1 aarch64 1.1.5-r0
nvidia-container-runtime aarch64 3.1.0-r0
nvidia-container-toolkit aarch64 1.0.5-r0
nvidia-docker aarch64 2.2.2-r0
```

That's weird, I have it.
If I manually create a daemon.json in /etc/docker with the "runtimes" as above and reload dockerd it works and `docker info` shows nvidia runtime. But it gets deleted when I reboot my device. Problem might come from somewhere else, I will investigate upon that

closed 2021-04-27 by RomainGsd