OE4T/meta-tegra #659: glibc / libc6 version conflict with jetpack 4.4 image

fei4xu, 2021-04-12:
**Describe the bug, or it's more like a problem I have**
I'm using a customized Xavier board which is based on jetpack 4.4 and some private BSP for Canbus ports. 

Since the private BSP is not open sources, I'm using Yocto and `meta-tegra` to build a complete rootfs/image but I don't use it as a whole. On the contrary, I manually copy the deploy `*.deb` files to my board and install them, at least it's my current plan.

I built an `app.deb`, during the `sudo apt install`, I found it depends on libc6. 
```sh
Depends: libc6 (>= 2.31+git0+df31c7ca92) but 2.27-3ubuntu1 is to be installed
```
From this message, it says libc6 2.27 is the current version as in Jetpack 4.4, bud in my Yocto and meta-tegra (`dunfell-r32.4.3` branch), it builds libc6_2.31.

At first, I didn't hesitate but forced to install the deployed `libc_2.31.deb` file, the result were many warnings and actually an error in the end.

Then I read something and thought I'd better not to upgrade libc.

Here is my question:

In my app, I don't write specific code to use libc 2.31 and I think it should work with libc 2.27, how can I remove this version requirement?

I can also copy the recipe of glibc 2.27 (https://layers.openembedded.org/layerindex/recipe/80771/) and set `PREFERRED_VERSION_glibc = 2.27`, then in my meta-tegra image, it use libc 2.27 anywhere.

Which way should I go?

Thanks.

fei4xu, 2021-04-12:
I did some test but failed, it seems that I need to use `gcc 7.3` (which is same as in default `jetpack 4.4`), but gcc 7.3 only exists in yocto `sumo` branch. 

Maybe I have gone too far in the wrong direction?

graugans, 2021-04-12:
Hey @fei4xu ,

related to the GCC part please check the [wiki page](https://github.com/OE4T/meta-tegra/wiki/Using-gcc7-from-the-contrib-layer) am I right you wanna build the app and some tools with the YOCTO layer but run them on the Nvidia L4T Ubuntu based image?

closed 2021-04-20 by fei4xu