OE4T/meta-tegra #673: gstnvtracker fails to dlopen `libnvvpi.so.0`

TheophileBlard, 2021-04-21:
**Describe the bug**
This simple gstreamer (+deepstream) pipeline:
```shell
gst-launch-1.0 videotestsrc is-live=true ! "video/x-raw,width=1280,height=720,framerate=30/1" ! \
nvvideoconvert ! "video/x-raw(memory:NVMM), format=NV12" ! \
m.sink_0 nvstreammux name=m width=1280 height=720 batch-size=1 ! \
nvtracker ll-lib-file=/opt/nvidia/deepstream/deepstream-5.0/lib/libnvds_nvdcf.so ! \
nvegltransform ! nveglglessink sync=false
```
fails with the following error: 
```shell
gstnvtracker: Loading low-level lib at /opt/nvidia/deepstream/deepstream-5.0/lib/libnvds_nvdcf.so
gstnvtracker: Failed to open low-level lib at /opt/nvidia/deepstream/deepstream-5.0/lib/libnvds_nvdcf.so
 dlopen error: libnvvpi.so.0: cannot open shared object file: No such file or directory
gstnvtracker: Failed to initilaize low level lib.
```
In order to fix this I had to do:
```shell
locate libnvvpi.so.0 # /opt/nvidia/deepstream/deepstream-5.0/lib/libnvvpi.so.0.3.0
ln -s /opt/nvidia/deepstream/deepstream-5.0/lib/libnvvpi.so.0.3.0 /opt/nvidia/deepstream/deepstream-5.0/lib/libnvvpi.so.0
```
I believe this soft link should already be there.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` tag 'f5fc11f941cda7778a49f5e6808966d6c7da9e0b' with `MACHINE` based on 'jetson-agx-xavier-devkit'

madisongh, 2021-04-21:
> My understanding is that this soft link should already be there.

It's not in the deepstream-5.0 deb package, so the recipe would have to be updated to add it.  I'm not sure how this works in stock L4T - the postinstall script in the deb appears to be calling on `ldconfig`, which might resolve the problem, too.  I'd be a bit concerned about doing the same here, if any of the shared libraries bundled into DS-5 have SONAMEs that conflict with other packages.

TheophileBlard, 2021-04-21:
Eventually, we should try to update the Deepstream recipe to 5.1, which was introduced with [Jetpack 4.5.1](https://docs.nvidia.com/jetson/jetpack/release-notes/index.html). Maybe the changed the `dlopen` path, or the deb package was updated.

PR #719 (2021-05-28): deepstream-5.0: add missing .so symlinks
and a runtime dependency on libvisionworks-devso-symlink, since
that is also needed by some of the gstreamer plugins.

Fixes:  #673 #713
closed 2021-05-28 by madisongh