OE4T/meta-tegra #674: backport l4t-32.4.3 from `dunfell` to `thud`

fei4xu, 2021-04-21:
Hi,

We are using a 3rd party hardware which has `systemd` 237 and `glibc` 2.27, both of which are default versions as in yocto `sumo`.
I have tested that if I copy the recipes from sumo to thud, they can still be compiled, but they will fail if compiled in `warrior` or later yocto, so I was basically stuck to sumo or thud.

I can verify that current `thud-l4t-r32.3.1` branch can compile just fine, as long as I follow the readme files of that branch. But r32.3.1 is from JetPack 4.3 and I would like to use JetPack 4.4 as that 3rd party hardware's default jetpack version.

I'd like to check how hard it will be if I decided to backport from `dunfell-l4t-r32.4.3` to current `thud-l4t-r32.3.1`? I have some yocto experience mainly as a user to call bitbake but not a developer to write .bb files.

I have checked this PR: **Backporting Linux4Tegra R32.3.1, JetPack 4.3 to Yocto/OE release Thud #263**
I think in theory another backport of JetPack 4.4 from `dunfell` to `thud` should be possible.

What's the suggested first step? 

1) Copy all the files directly from `meta-tegra` `dunfell` to `thud`? I have tested and it did not work.

2) To compare what's changed between `dunfell-l4t-r32.3.1` to `dunfell-l4t-r32.4.3` and do the same thing on `thud-l4t-r32.3.1` line by line? Unfortunately there is no branch named `dunfell-l4t-r32.3.1` so I can't do the comparison.

3) to check the history of `dunfell` to compare current `r32.4.3` branch head to the version when it had `r32.3.1`? which commit should I begin with?

I really want to heart expert's suggestions here.

Thanks,
Fei

madisongh, 2021-04-21:
The `dunfell` branch (with no version suffix) is [based off L4T R32.3.1](https://github.com/OE4T/meta-tegra/wiki/Which-branch-should-I-use%3F), so you could use that for your comparison option (2).

You may also need to update `thud-l4t-r32.3.1` with any applicable fixes/updates from `dunfell`... that backport happened a year ago, so it may be missing some changes that would simplify the upgrade.

fei4xu, 2021-04-25:
Hi @madisongh , I think I have a candidate for `thud--l4t-r32.4.3` now. What I have done is actually ugly: just copy recipes from `dunfell-l4t-r32.4.3` to `thud--l4t-r32.3.1` but not all of them (at least I left OpenCV because OpenCV 4.1 will not compile in `thud` anyway).

If I create a PR with those **hard copied** recipes, will it be accepted? 
(The ideal way is to to use cherry picks as much as possible, but there are too many commits depend on each other and I can't figure them out.)

madisongh, 2021-04-25:
@fei4xu I've created a `thud-l4t-r32.4.3` branch based off `thud-l4t-r32.3.1`.  Go ahead and submit your PR for the new branch, and I'll look it over.  The copy-over approach is fine.

closed 2021-04-29 by fei4xu