OE4T/meta-tegra #694: Can't get spidev to work on Jetson Nano (production module)

Buanderie, 2021-04-27:
Hello,

I'm making a custom Yocto distro, targeting Jetson Nano, using the Leopard Imaging carrier board.
I can't get the spi driver (spidev) to work, despite all my effort.
Nothing mentioning SPI or spidev either in dmesg.

Steps to reproduce the behavior:
1. Use meta-tegra **dunfell-l4t-r32.4.3** 
2. I also tried to patch pinmux to enable spi1 (see attached files)
[a02.txt](https://github.com/OE4T/meta-tegra/files/6384910/a02.txt)
[b00.txt](https://github.com/OE4T/meta-tegra/files/6384911/b00.txt)

I've seen this as a potential bug/misconfiguration here an there (#454, #573 ?), but I don't know if it's relatable or not.

I also made sure SPI was enabled in the kernel.

What could I be doing wrong ?

Buanderie, 2021-04-27:
Addendum:
Using the following command **dtc -I fs -O dts -o ./extracted.dts /proc/device-tree**, I get this result:

> dtc -I fs -O dts -o ./extracted.dts /proc/device-tree |& grep spi
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d600/spi@1: incorrect #address-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d600/spi@1: incorrect #size-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d600/spi@0: incorrect #address-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d600/spi@0: incorrect #size-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d400/spi@1: incorrect #address-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d400/spi@1: incorrect #size-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d400/spi@0: incorrect #address-cells for SPI bus
./extracted.dts: Warning (spi_bus_bridge): /spi@7000d400/spi@0: incorrect #size-cells for SPI bus
./extracted.dts: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'
./extracted.dts: Warning (avoid_unnecessary_addr_size): /spi@70410000/spiflash@0: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
./extracted.dts: Warning (avoid_unnecessary_addr_size): /spi@7000d800: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
./extracted.dts: Warning (unique_unit_address): /spi@7000da00/spi-touch14x8@0: duplicate unit-address (also used in node /spi@7000da00/spi-touch25x16@0)
./extracted.dts: Warning (unique_unit_address): /spi@7000da00/spi-touch14x8@0: duplicate unit-address (also used in node /spi@7000da00/spi-touch19x12@0)
./extracted.dts: Warning (unique_unit_address): /spi@7000da00/spi-touch25x16@0: duplicate unit-address (also used in node /spi@7000da00/spi-touch19x12@0)
./extracted.dts: Warning (clocks_property): /spi@70410000:clocks: cell 0 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@70410000:clocks: cell 2 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d600:clocks: cell 0 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d600:clocks: cell 2 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d600:clocks: cell 4 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000da00:clocks: cell 0 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000da00:clocks: cell 2 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000da00:clocks: cell 4 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d800:clocks: cell 0 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d800:clocks: cell 2 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d800:clocks: cell 4 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d400:clocks: cell 0 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d400:clocks: cell 2 is not a phandle reference
./extracted.dts: Warning (clocks_property): /spi@7000d400:clocks: cell 4 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@70410000:dmas: cell 0 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@70410000:dmas: cell 2 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d600:dmas: cell 0 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d600:dmas: cell 2 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000da00:dmas: cell 0 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000da00:dmas: cell 2 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d800:dmas: cell 0 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d800:dmas: cell 2 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d400:dmas: cell 0 is not a phandle reference
./extracted.dts: Warning (dmas_property): /spi@7000d400:dmas: cell 2 is not a phandle reference
./extracted.dts: Warning (iommus_property): /spi@70410000:iommus: cell 0 is not a phandle reference
./extracted.dts: Warning (iommus_property): /spi@7000d600:iommus: cell 0 is not a phandle reference
./extracted.dts: Warning (iommus_property): /spi@7000da00:iommus: cell 0 is not a phandle reference
./extracted.dts: Warning (iommus_property): /spi@7000d800:iommus: cell 0 is not a phandle reference
./extracted.dts: Warning (iommus_property): /spi@7000d400:iommus: cell 0 is not a phandle reference
./extracted.dts: Warning (resets_property): /spi@70410000:resets: cell 0 is not a phandle reference
./extracted.dts: Warning (resets_property): /spi@7000d600:resets: cell 0 is not a phandle reference
./extracted.dts: Warning (resets_property): /spi@7000da00:resets: cell 0 is not a phandle reference
./extracted.dts: Warning (resets_property): /spi@7000d800:resets: cell 0 is not a phandle reference
./extracted.dts: Warning (resets_property): /spi@7000d400:resets: cell 0 is not a phandle reference
./extracted.dts: Warning (gpios_property): /spi@7000da00/spi-touch14x8@0:reset-gpio: cell 0 is not a phandle reference
./extracted.dts: Warning (gpios_property): /spi@7000da00/spi-touch25x16@0:reset-gpio: cell 0 is not a phandle reference
./extracted.dts: Warning (gpios_property): /spi@7000da00/spi-touch19x12@0:reset-gpio: cell 0 is not a phandle reference

madisongh, 2021-04-27:
I see a couple of things.  One is that your patches aren't really complete, since they don't set all of the needed attributes.  See the patch I attached [here](https://github.com/OE4T/meta-tegra/issues/573#issuecomment-778630092) that sets the pin attributes correctly for the SPI functions.

Another is to check whether you have the `spidev` module loaded on your device via `lsmod`.  If not, try `modprobe spidev` to see if you can load it. Make sure you've included `kernel-module-spidev` in your image build so it is present.  (This is with our default kernel config, which sets `CONFIG_SPIDEV=m` to have it built as a module.)

The warnings you're seeing from `dtc` are, unfortunately, normal for the NVIDIA device trees, and can be ignored.

Buanderie, 2021-04-28:
Thanks for replying !
Hmmm, okay I added your patch. I'm rebuilding as we speak... I also made sure spidev was included is in the image. It was indeed loaded ans listed in lsmod. I also tried to manually reload it, but no improvement.
Following [this](https://github.com/OE4T/meta-tegra/issues/573#issuecomment-778630092) post, I also checked the output of `grep -C 4 spi1 /sys/kernel/debug/pinctrl/pinctrl-maps` - see attached file.
[pinctrl_diff.txt](https://github.com/OE4T/meta-tegra/files/6389898/pinctrl_diff.txt)
The only difference is that I get "state 0" everywhere...

Buanderie, 2021-04-28:
Update: Okay I got the spidev devices listed in /dev/ now...
Thank you :)

closed 2021-04-28 by Buanderie