OE4T/meta-tegra #707: Jetson-nano emmc spidev issue

nandra, 2021-05-10:
This is similar as #694 but I'm trying to use emm version of jetson nano. I've added patch for pinmux:
```
diff --git a/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0002-b00.dtsi b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0002-b00.dtsi
index 60e5bb40a826..4892f201ba51 100644
--- a/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0002-b00.dtsi
+++ b/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-pinmux-p3448-0002-b00.dtsi
@@ -721,7 +721,7 @@
 
                        spi1_mosi_pc0 {
                                nvidia,pins = "spi1_mosi_pc0";
-                               nvidia,function = "rsvd1";
+                               nvidia,function = "spi1";^M
                                nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                                nvidia,tristate = <TEGRA_PIN_DISABLE>;
                                nvidia,enable-input = <TEGRA_PIN_ENABLE>;
@@ -729,7 +729,7 @@
 
                        spi1_miso_pc1 {
                                nvidia,pins = "spi1_miso_pc1";
-                               nvidia,function = "rsvd1";
+                               nvidia,function = "spi1";^M
                                nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                                nvidia,tristate = <TEGRA_PIN_DISABLE>;
                                nvidia,enable-input = <TEGRA_PIN_ENABLE>;
@@ -737,7 +737,7 @@
 
                        spi1_sck_pc2 {
                                nvidia,pins = "spi1_sck_pc2";
-                               nvidia,function = "rsvd1";
+                               nvidia,function = "spi1";^M
                                nvidia,pull = <TEGRA_PIN_PULL_DOWN>;
                                nvidia,tristate = <TEGRA_PIN_DISABLE>;
                                nvidia,enable-input = <TEGRA_PIN_ENABLE>;
@@ -745,7 +745,7 @@
 
                        spi1_cs0_pc3 {
                                nvidia,pins = "spi1_cs0_pc3";
-                               nvidia,function = "rsvd1";
+                               nvidia,function = "spi1";^M
                                nvidia,pull = <TEGRA_PIN_PULL_UP>;
                                nvidia,tristate = <TEGRA_PIN_DISABLE>;
                                nvidia,enable-input = <TEGRA_PIN_ENABLE>;
@@ -753,7 +753,7 @@
 
                        spi1_cs1_pc4 {
                                nvidia,pins = "spi1_cs1_pc4";
-                               nvidia,function = "rsvd1";
+                               nvidia,function = "spi1";^M
                                nvidia,pull = <TEGRA_PIN_PULL_UP>;
                                nvidia,tristate = <TEGRA_PIN_DISABLE>;
                                nvidia,enable-input = <TEGRA_PIN_ENABLE>;
```

check pinmux shows spi1:
```
device 700008d4.pinmux
state 0
type MUX_GROUP (2)
controlling device 700008d4.pinmux
group spi1_mosi_pc0
function spi1

device 700008d4.pinmux
state 0
type CONFIGS_GROUP (4)
controlling device 700008d4.pinmux
group spi1_mosi_pc0
config pull=1
config tristate=0
config enable-input=1

device 700008d4.pinmux
state 0
type MUX_GROUP (2)
controlling device 700008d4.pinmux
group spi1_miso_pc1
function spi1

device 700008d4.pinmux
state 0
type CONFIGS_GROUP (4)
controlling device 700008d4.pinmux
group spi1_miso_pc1
config pull=1
config tristate=0
config enable-input=1

device 700008d4.pinmux
state 0
type MUX_GROUP (2)
controlling device 700008d4.pinmux
group spi1_sck_pc2
function spi1

device 700008d4.pinmux
state 0
type CONFIGS_GROUP (4)
controlling device 700008d4.pinmux
group spi1_sck_pc2
config pull=1
config tristate=0
config enable-input=1

device 700008d4.pinmux
state 0
type MUX_GROUP (2)
controlling device 700008d4.pinmux
group spi1_cs0_pc3
function spi1

device 700008d4.pinmux
state 0
type CONFIGS_GROUP (4)
controlling device 700008d4.pinmux
group spi1_cs0_pc3
config pull=2
config tristate=0
config enable-input=1

device 700008d4.pinmux
state 0
type MUX_GROUP (2)
controlling device 700008d4.pinmux
group spi1_cs1_pc4
function spi1

device 700008d4.pinmux
state 0
type CONFIGS_GROUP (4)
controlling device 700008d4.pinmux
group spi1_cs1_pc4
config pull=2
config tristate=0
config enable-input=1
```

and shortcut 19 and pins with jumper and run spidev_test and getting:
```
./spidev_test -v -D /dev/spidev0.0
spi mode: 0x0
bits per word: 8
max speed: 500000 Hz (500 KHz)
TX | FF FF FF FF FF FF 40 00 00 00 00 95 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F0 0D  | ......@....𕕕...
RX | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ...............
```

Any idea what I'm doing wrong? Thanks.

madisongh, 2021-05-10:
Looks like the same as in #694 - you haven't changed the the other attributes of the pins.  See the patch attached to [this earlier comment](https://github.com/OE4T/meta-tegra/issues/573#issuecomment-778630092) for reference.

nandra, 2021-05-10:
Are you referring to this patch: https://github.com/OE4T/meta-tegra/files/5975866/0001-configure-SPI-pins.patch.txt ? on 0002-b00 seems pins was properly setup (I check one one by with your patch and seems to be same). Thanks.

madisongh, 2021-05-10:
Heh, that's what I get for just looking at the changed lines.  It figures that the defaults in the existing DTS are different between the two platforms.  The settings do look OK... is there anything in the kernel log that might point to a problem? Can you probe the pins to see if there's any signal?

nandra, 2021-05-10:
Hmm searching nvidia forum I hit this: https://forums.developer.nvidia.com/t/check-pinmux-on-spi0-1-on-jetson-nano/115534/7 which says about updating u-boot and also this one: https://forums.developer.nvidia.com/t/spidev-of-jetson-nano-emmc-core-board-cannot-be-used/157946/23?u=marek.belisko I'm bit confused ;)

nandra, 2021-05-10:
No signal on pins -> probed with scope.

madisongh, 2021-05-10:
I ran through the [pinmux configurator spreadsheet](https://developer.nvidia.com/jetson-nano-pinmux) to configure the pins and realized that unlike the Nano devkit, the default GPIO settings in the device tree for the Nano module were allocating those pins to the GPIO controller.

The attached patch covers the device tree changes for both p3448-0000 and p3448-0002, both rev a02 and b00.  This is of the R32.5 branch, but should apply to other recent versions as well.
[0001-nvidia-platform-t210-enable-SPI0-pins-on-40-pin-head.patch.txt](https://github.com/OE4T/meta-tegra/files/6452741/0001-nvidia-platform-t210-enable-SPI0-pins-on-40-pin-head.patch.txt)

nandra, 2021-05-10:
@madisongh thanks a lot. I'll give a try and report back.

nandra, 2021-05-11:
Looks like it's some driver issue -> added few debug messages. I'll report back when find something.

ichergui, 2021-05-17:
Hey @nandra 

Did you have the chance to fix the issue or have traces to share with us ?

nandra, 2021-05-17:
It seems that TPM driver fails when first read state from TPM chip. SPI read pass but then values are wrong. As suggested here and aslo checked with HW developer in team it seems that level shifters can be a problem. HW engineer is trying to solder SPI bus directly to CPU and not on 40 pin header. Can be closed with big thanks to @madisongh for resolving nano spi issue on 40 pin header. Maybe we would need to extend wiki with this patch and description because many can potentially people just hit the same issue. Thanks.

ichergui, 2021-05-17:
Thanks @nandra for the feedback

nandra, 2021-05-17:
@ichergui should I extend wiki? Thanks. Also link to nvidia forum about level shifters: https://forums.developer.nvidia.com/t/infineon-slb9670-tpm2-0-module-with-jetson-nano/163588/11

ichergui, 2021-05-17:
Please check if you are able to use this wiki page  [SPI support](https://github.com/OE4T/meta-tegra/wiki/SPI-support) 
Thanks a lot

nandra, 2021-05-17:
Ok added some notes, please feel free to edit it. I'm not native English speaking ;)

ichergui, 2021-05-17:
Thanks @nandra a lot, I'm not a native english speaker as well.
I will add a minor updates but it looks good to me.

ichergui, 2021-05-17:
Hey @nandra 
I pushed a minor updates, fix a typo in the wiki page [SPI-support](https://github.com/OE4T/meta-tegra/wiki/SPI-support) 
Please let me know what do you think.

closed 2021-05-18 by ichergui