OE4T/meta-tegra #726: package manager for linking nvds_ libraries

jlucke86, 2021-06-04:
Hi Matt,

Firstly great work on the L4T 32.5 / Deepstream 5.1 sources. The project is getting very stable.

It appears there is no way to link the nvds_meta.so, nvdsgst_meta.so libraries that are required to build and link a production quality C deepstream application within yocto. I believe this is mostly an upstream NVIDIA problem which they are aware of but appear to have done nothing about, ie provide a deepstream.pc file so that users of CMake or pkg-config.

I'm plan to submit a PR which works, but needs advice / quality improvement before merge.

https://forums.developer.nvidia.com/t/deepstream-pkg-config-file/120889/13

jlucke86, 2021-06-05:
I see you provided a solution for this packaging in /etc/ld.so.conf.d/deepstream.conf.

The conf file doesn't seem to be adding the "sysroot"/opt/nvidia/deepstream search path for $LD however. At least its not working in my cross toochain. I'm still working on my package to put in a deepstream.pc file

`==================================================
aarch64-poky-linux-ld: mode aarch64linux
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/poky/3.1.7/sysroots/x86_64-pokysdk-linux/usr/aarch64-poky-linux/lib64/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/poky/3.1.7/sysroots/x86_64-pokysdk-linux/usr/aarch64-poky-linux/lib64/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/local/lib64/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/local/lib64/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/lib64/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/lib64/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/lib64/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/lib64/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/poky/3.1.7/sysroots/x86_64-pokysdk-linux/usr/aarch64-poky-linux/lib/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/poky/3.1.7/sysroots/x86_64-pokysdk-linux/usr/aarch64-poky-linux/lib/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/local/lib/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/local/lib/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/lib/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/lib/libnvds_meta.a failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/lib/libnvds_meta.so failed
attempt to open /opt/poky/3.1.7/sysroots/aarch64-poky-linux/usr/lib/libnvds_meta.a failed
aarch64-poky-linux-ld: cannot find -lnvds_meta
joshua@joshua-ThinkCentre-M83:~$ ls /opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/nvidia/deepstream/deepstream-5.1/lib/libnvds_meta.so 
/opt/poky/3.1.7/sysroots/aarch64-poky-linux/opt/nvidia/deepstream/deepstream-5.1/lib/libnvds_meta.so
joshua@joshua-ThinkCentre-M83:~$ cat /opt/poky/3.1.7/sysroots/aarch64-poky-linux/etc/ld.so.conf.d/deepstream.conf 
/opt/nvidia/deepstream/deepstream-5.1/lib
j`

madisongh, 2021-06-05:
> The conf file doesn't seem to be adding the "sysroot"/opt/nvidia/deepstream search path for $LD however. 

That's correct.  It only applies to the search path the dynamic linker uses at runtime.  Doesn't apply to anything that happens at build time.

All you need is to add a `-L` option to pass the path to the location of the deepstream libraries to the linker, just as you would need a `-I` option for the compiler to point to its header files.  You could do this in your makefiles (or cmake files, or whatever) by using the `=` prefix to tell gcc/g++ that the path is relative to the `--sysroot` path, i.e. `-L=/opt/nvidia/deepstream/deepstream-5.1/lib`. Alternatively, you could do this in your recipe by appending either `-L=/opt/nvidia/deepstream/deepstream-5.1/lib` or `-L${STAGING_DIR_HOST}/opt/nvidia/deepstream/deepstream-5.1/lib` to `LDFLAGS` - assuming your makefiles honor the LDFLAGS setting passed down by the recipe, of course.

madisongh, 2021-06-09:
@jlucke86 Did you get this working?

jlucke86, 2021-06-10:
Yes thanks, my app is building. I understand your solution to doing it with no BSP changes. I also have the deepstream-5.1.pc package config working now. Only reason its not up for PR is because I hit another unrelated hitch with another sub library to this same application.

closed 2021-07-20 by madisongh